debian-pi/raspbian-ua-netinst #176: Insecure validation of archive.raspberrypi.org signing key

mologie, 2015-01-08:
Hello,

the marked lines allows for malicious keys to be imported into the temporary keychain:
https://github.com/debian-pi/raspbian-ua-netinst/blob/master/update.sh#L115-L118

Using `gpg -k`, one merely checks for the presence of the good key in the previously imported file. It does not check for the absence of malicious keys. Specifically, `gpg --import` allows for importing multiple keys from one file. Therefore, an attacker must only append his malicious key to the file downloaded over HTTP (and leave the good key in it too) in order to inject a malicious key.

As fix, I recommend embedding a verified copy of the correct key into update.sh.

mologie, 2015-01-08:
This may probably count as duplicate of #64. I opened a new issue because the changes from #64 were already merged and the issue is very old (yet not closed).

diederikdehaas, 2015-01-08:
@mologie You make a valid and correct point and I specifically left issue #64 open since the raspberrypi.org key wasn't (and **still isn't**) verified. 
Consequently your recommended fix can not be applied. 
And it would also not [fetch the latest version of the key with `update.sh`](https://github.com/debian-pi/raspbian-ua-netinst/issues/64#issuecomment-46363867).
Plugwash has [provided proof](https://github.com/debian-pi/raspbian-ua-netinst/issues/64#issuecomment-46373413) that the raspbian.org key is the correct one though :smile: 

Apparently there is only 1 way to check a fingerprint before importing it, but that would require manually parsing gpg's output ... and I don't think that's a good way to go:
`gpg --with-fingerprint --with-colons FILE`
(source: http://lists.gnupg.org/pipermail/gnupg-users/2012-April/044078.html)

But ... I think I have found a proper solution by adding an extra check through counting the number of keys in the keyring: 
`gpg --homedir gnupg --with-colons --list-keys | grep "^pub:" | wc -l`
(source: http://lists.gnupg.org/pipermail/gnupg-users/2012-July/045044.html)
Thus after the first import, the count must be 1 and after the second it must be 2.

Note that I'm not very familiar with gpg stuff, so I can be wrong.

diederikdehaas, 2015-05-05:
Now that the raspberrypi.org signing key is downloaded over HTTPS, I think this issue is solved.
@mologie, what do you think?
See also commit https://github.com/debian-pi/raspbian-ua-netinst/commit/f4510092885d2c703309a45320347a4b153ad910

mologie, 2015-05-06:
Hello @diederikdehaas! I've read through your commit and the pull request #222, and I think that you are missing the point of what a key fingerprint actually is. Various of your comments mention that you do not wish to include the whole key (~1KiB file) with your distribution, but instead download it from the internet, in order to keep it up-to-date. You however also check the key's fingerprint in order to ensure the key has not been tampered with... but these are two mutually exclusive things.

You cannot actually replace the key (apart from changing some irrelevant metadata) without changing the fingerprint. The fingerprint is a value mathematically derived from the key contents. Therefore, you have to update your script if the key/fingerprint changes either way. Therefore, you could just download the key once on your machine, ensure it is the correct one (by asking the developer for the key's fingerprint), and embed the whole key into this repository. I believe it would also be easier for the user to update the key afterwards (just replace the key file).

Either trust the server and trust that the HTTPS connection is not compromised, or trust the key contents (identified through a fingerprint provided by the developer). Not both, that is redundant and potentially insecure because GPG does not allow limiting what is imported through a key file. The current method for attempting to limit which keys are imported may work, but are an ugly hack. Using HTTPS only may be more convenient, but is also less secure than using a verified key directly.

jturner314, 2015-05-06:
Both approaches feel like a hack, and I'm not a huge fan of adding keys to version control. After thinking about it for a little while, though, adding the keys directly to the repository as @mologie suggests seems cleaner and simpler than the current approach of downloading and verifying the keys on each update. I implemented this in #223. @diederikdehaas, what do you think?

goranche, 2015-05-06:
I'd prefer not having the keys in the repository... I do agree with some of @mologie points, but still prefer fetching the key over HTTPS :thought_balloon:

diederikdehaas, 2015-05-06:
@goranche why? 
What's the benefit of fetching the keys over HTTPS vs having a verified key in the repository?

(I'm just trying to get a better understanding on this, because I do know something about GPG but are not exactly an expert)

diederikdehaas, 2015-05-06:
@mologie I'm inclined to agree with you ...

Now I need to figure out where/when my thought process was/went wrong ...

diederikdehaas, 2015-05-06:
I went back over previous issues/PR/etc to figure out why we're doing things we're doing now
I [found the following item](https://github.com/debian-pi/raspbian-ua-netinst/issues/64#issuecomment-46363867):
- By fetching the keys during update.sh, the user gets the latest versions of those keys.

By including the keys in the repo, we wouldn't be notified if a key gets revoked, an expiration date (is set and) is passed and maybe other things, right?
Or am I missing/misinterpreting things (again :stuck_out_tongue: )?

jturner314, 2015-05-06:
The following things can happen to a key (while still keeping the same fingerprint) (see the man page for `gpg2(1)` under `edit-key`):

1. Add/remove/revoke a signature. (A signature from another GPG user on a user ID of the key signifies that he/she believes the the ID to be accurate.)
2. Add/remove/reorder user IDs, notations, or photos.
3. Change the preferences for using the key (e.g. preferred keyserver, algorithm preferences, etc.).
4. Add/remove/revoke a subkey.
5. Change the expiration date.
6. Add a designated revoker to the key.
7. Revoke the primary key.

The only ones that really matter for this use case are 4, 5, and 7. The keys are currently set to never expire, but the expiration date could be shortened or the keys could be revoked.

`update.sh` does not currently check these possibilities, which it probably should (but this appears to require parsing GPG output, not just return codes). However, if a key is revoked/expires, I would expect the Raspberry Pi/Raspbian people to replace the key on their server with a new key (which would have a different fingerprint and be detected by `update.sh`).

If we include the keys in the repository, someone (maybe the person who tags the releases) should periodically check that the keys are still valid.

It's a tradeoff between code complexity and vulnerabilities that we haven't thought of with downloading the keys vs. someone having to check the validity of the keys every once in a while. Both options are not ideal. If we add additional code to check for expired/revoked keys in `update.sh`, I think we're better off downloading the keys; otherwise, I think we're better off adding the keys to the repository and checking them periodically.

diederikdehaas, 2015-05-06:
Thanks for the extensive info :+1: 
I'm currently doing some reading on GPG to get a better understanding of things.

jturner314, 2015-05-07:
You're welcome. The [GNU Privacy Handbook](https://gnupg.org/gph/en/manual.html) is the best introduction that I've found to the concepts of GPG (how public key encryption works, key management, the web of trust, etc.).

diederikdehaas, 2015-05-08:
Status update from my side. 
I hope you will all review it and point out errors and/or omissions in it and/or suggests improvements.

**Embedding keys in repo vs downloading the keys each time**
Embedding the keys in the repo is like adding the keys to your keyring. When you have keys in your keyring [you should refresh your keyring regularly](https://help.riseup.net/en/security/message-security/openpgp/best-practices#selecting-a-keyserver-and-configuring-your-machine-to-refresh-your-keyring).
For the keys in your (actual) keyring you can set up a program which does that for you automatically. 
When you embed the keys in the repo, you don't have that 'luxury'. Instead, you have to remind yourself periodically to check the validity of the key, which is a manual and therefor error-prone procedure.
As it is implemented now in `update.sh`, it is checked automatically and it's done often (I build the installer a **LOT**). That means when something isn't right, I (and possibly others) get informed about it immediately and can take actions accordingly.

An actual example: during some maintenance work on the raspbian.org archive the gpg key was accidentally deleted. During one of my builds I noticed that, informed plugwash (the maintainer) about it and the issue was resolved almost immediately.
While this particular case isn't the responsibility of the installer, when for example the fingerprint of the key changes, I'd also immediately notice it and can take appropriate action.

Therefor I prefer the current method of downloading and verifying the keys each time (in `update.sh`).

**Downloading the keys over HTTPS and/or checking the fingerprint**
I kind of agree with @mologie that doing both is/seems redundant. Downloading it over HTTPS does indeed ensure that it hasn't been tampered with during transit and checking the fingerprint does a similar thing.
However I don't see any harm in doing both, especially since it's an automated procedure (which can be improved, see below for my suggestion).
Doing both has the advantage that if the key is changed, we/I get notified of it immediately.

**Improving the checks of the procedure**
AFAICT you can do various checks **before** you import a key into a keyring.
So I think the following would be an improvement:
```sh
RASPBIAN_ARCHIVE_KEY_LOCATION="https://archive.raspbian.org/"
RASPBIAN_ARCHIVE_KEY_NAME="raspbian.public.key"
RASPBIAN_ARCHIVE_KEY_URL=${RASPBIAN_ARCHIVE_KEY_LOCATION}${RASPBIAN_ARCHIVE_KEY_NAME}
RASPBIAN_ARCHIVE_KEY_FINGERPRINT="A0DA38D0D76E8B5D638872819165938D90FDDD2E"

RASPBERRYPI_ARCHIVE_KEY_LOCATION="https://www.raspberrypi.org/"
RASPBERRYPI_ARCHIVE_KEY_NAME="raspberrypi.gpg.key"
RASPBERRYPI_ARCHIVE_KEY_URL=${RASPBERRYPI_ARCHIVE_KEY_LOCATION}${RASPBERRYPI_ARCHIVE_KEY_NAME}
RASPBERRYPI_ARCHIVE_KEY_FINGERPRINT="CF8A1AF502A2AA2D763BAE7E82B129927FA3303E"

mkdir -p gnupg
chmod 0700 gnupg

echo -n "Downloading ${RASPBIAN_ARCHIVE_KEY_NAME}... "
curl -# -O ${RASPBIAN_ARCHIVE_KEY_URL}
if [ -f ${RASPBIAN_ARCHIVE_KEY_NAME} ] ; then
    echo "OK"
else
    echo "FAILED!"
    cd ..
    exit 1
fi

echo "Checking ${RASPBIAN_ARCHIVE_KEY_NAME}... "
# check that there is only 1 public key in the key file
if [ $(gpg --with-colons ${RASPBIAN_ARCHIVE_KEY_NAME} | grep ^pub: | wc -l) -ne 1 ] ; then
    echo -e "\n  ERROR  \nThere is more then one key in the ${RASPBIAN_ARCHIVE_KEY_NAME} key file!"
    cd ..
    exit 1
fi
# check that the key file's fingerprint is correct
if [ "$(gpg --with-colons ${RASPBIAN_ARCHIVE_KEY_NAME} | grep ^fpr: | awk -F: '{print $10}')" != "${RASPBIAN_ARCHIVE_KEY_FINGERPRINT}" ] ; then
    echo -e "\n  ERROR  \nThe fingerprint of ${RASPBIAN_ARCHIVE_KEY_NAME} key file does NOT match the expected value!"
    cd ..
    exit 1
fi
# do other checks if needed ... (maybe with gpgv?)
echo "OK"

# GPG key checks out, thus import it into our own keyring
echo -n "Importing ${RASPBIAN_ARCHIVE_KEY_NAME} into keyring..."
gpg -q --homedir gnupg --import ${RASPBIAN_ARCHIVE_KEY_NAME}
if [ $? -eq 0 ] then
    echo "OK"
else
    echo "FAILED!"
    cd ..
    exit 1
fi

# and do the same with the raspberrypi.org key
# probably best to refactor the above code into a function

```

goranche, 2015-05-08:
> Downloading it over HTTPS does indeed ensure that it hasn't been tampered with during transit and checking the fingerprint does a similar thing.

not really... HTTPS ensures (well, to a reasonable level) that no funny stuff happens during transmission, while the fingerprint ensures no funny stuff happened on the server...

I agree with leaning towards doing both (as it is now)... will go through the code later today

jturner314, 2015-05-09:
Downloading each time sounds good to me. I'm kind of busy right now since I'm graduating from my university, but I should have time on Sunday or Monday to take a look at the code.

diederikdehaas, 2015-05-09:
np and good luck :+1:

Mausy5043, 2015-05-09:
You might want to consider using the `-m` switch on `mkdir..` like this: 
`mkdir -m 0700 -p gnupg`

It saves on having to use an extra `chmod`.

diederikdehaas, 2015-05-09:
Good point @Mausy5043 :+1:

jturner314, 2015-05-12:
#### Comments on the code

Overall, the code looks good and basically does the same things as the current version with the addition of checking the keys separately and not importing them first. Here are a couple of minor suggestions:

* You need to add the `--with-fingerprint` option to the call to GPG when checking the key's fingerprint. (The fingerprint is not printed otherwise, at least on my machine.)

* It's good practice to quote expansions even if you know the expanded form won't have spaces, e.g.:
  ```bash
  RASPBIAN_ARCHIVE_KEY_URL="${RASPBIAN_ARCHIVE_KEY_LOCATION}${RASPBIAN_ARCHIVE_KEY_NAME}"
  RASPBERRYPI_ARCHIVE_KEY_URL="${RASPBERRYPI_ARCHIVE_KEY_LOCATION}${RASPBERRYPI_ARCHIVE_KEY_NAME}"
  curl -# -O "${RASPBIAN_ARCHIVE_KEY_URL}"
```
and so on.

* You can simplify the last command/if sequence like this:
  ```bash
  if gpg -q --homedir gnupg --import "${RASPBIAN_ARCHIVE_KEY_NAME}" ; then
      echo "OK"
  else
      echo "FAILED!"
      cd ..
      exit 1
  fi
  ```

#### Other possible improvements

Neither the current version of `update.sh` nor your proposal check the expiration/revocation/etc. of the key. It would be nice to check field 2 of the `pub` and `sub` records to see if the primary key/subkeys are invalid, disabled, revoked or expired. It would also be nice to check field 12 to see if the key/subkeys are deprecated. See [this page](http://git.gnupg.org/cgi-bin/gitweb.cgi?p=gnupg.git;a=blob_plain;f=doc/DETAILS;hb=refs/heads/STABLE-BRANCH-2-0) for details of the `--with-colons` listing format.

diederikdehaas, 2015-05-12:
Good points, thanks for your review :+1: 

I'll implement it and submit a PR for it.

diederikdehaas, 2015-05-14:
> You need to add the --with-fingerprint option to the call to GPG when checking the key's fingerprint. (The fingerprint is not printed otherwise, at least on my machine.)

I've been doing some testing/playing and on my machine there's quite some difference in the output when you use 
`gpg --with-colons "${RASPBIAN_ARCHIVE_KEY_NAME}"` vs 
`gpg --with-colons --homedir <homedir>`
With the former construct, the fingerprint is printed for the primairy key (but not subkeys). With the latter you indeed need to use `--with-fingerprint` to see the fingerprint.

Also, information like field 12 is present in one output but not the other ...
I have tried it yet, but I think it's likely that the expiration value is present on both, but I still suggest against validating for it because the value can have all kind of forms (seconds since epoch, date in YYYY-MM-DD format and date in ISO 8601 format).

With that in mind, I'm inclined to only check for fingerprint, that only 1 key is present in the file and check the return value on the `--import` to see if anything else is wrong (thus relying on gpg's checking).

Opinions welcome :-)

diederikdehaas, 2015-06-17:
This issue should now be fixed with the merge of https://github.com/debian-pi/raspbian-ua-netinst/pull/247

closed 2015-06-17 by diederikdehaas