debian-pi/raspbian-ua-netinst #177: BananaPI Support?

andriej, 2015-01-13:
In short words: Is there any chance of getting the netinst to support BananaPi (which is RPi compatible)?

I can't even describe how awesome it would be.

goranche, 2015-01-13:
have you actually tried it already?
as far as I can tell (from documents, don't have an Orange Pi to test for myself), it should be able to run the same images, so the installer might work as is...

diederikdehaas, 2015-01-13:
What's the difference between an OrangePi and a BananaPi?
Like @goranche said, it should work ... but why not use the Debian armhf port?
I have no objections to supporting BananaPi/OrangePi, but since I don't have one, I can't test it ...

diederikdehaas, 2015-03-04:
What's the status on this issue?

diederikdehaas, 2015-03-21:
Closing this issue since the reporter seems to have lost interest. 
If someone does want this and is willing to provide feedback, feel free to reopen it.

closed 2015-03-21 by diederikdehaas