debian-pi/raspbian-ua-netinst #178: incorrect kernel in config.txt

tvandera, 2015-01-22:
Hello Diederik (or someone else),

After my recent apt-get upgrade, I bumped into this issue:

https://bugs.launchpad.net/raspbian/+bug/1222486/comments/5

In short: I was unable to load any more modules, because of a mismatch between the kernel loaded and the modules installed. Cause seemed to be that my pi was booting kernel.img instead of vmlinuz-3.12.-1.rpi. The above mentioned launchpad bug seemed to indicate that this would need to be fixed here.

Has this been fixed since?

Cheers,
Tom

diederikdehaas, 2015-01-22:
Yes, version 1.0.6 of the installer contains a kernel-post-install script which sets both kernel and initramfs lines in `/boot/config.txt`

diederikdehaas, 2015-03-04:
I think this issue is fixed and thus can be closed. If you disagree, feel free to reopen it.

closed 2015-03-04 by diederikdehaas