debian-pi/raspbian-ua-netinst #179: Unable to boot: Superblock last mount time is in the future.

deafmalice, 2015-01-26:
After installing with netinst, making an image and flashing that image on another SD card a few days later I get something like this (is copied from somewhere else since I can't copy directly from the Pi here)

    Checking filesystems
    IMAGE2: Superblock last mount time (Tue Jan 16  1 17:32:48 2015,
          now = Thu Jan 24 17:34:29 2015) is in the future.

    IMAGE2: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
        (i.e., without -a or -p options)    

The same happens after leaving the Pi off for a few days, inserting the SD in any computer that has the correct time and booting the Pi with it. 

Why is this happening and how it can be resolved? I mean, the official Raspbian does it somehow.

diederikdehaas, 2015-01-26:
Hmm, I've never seen this error on my Pi's, but it's usually due to an incorrect system clock, which is understandable since the Pi doesn't have a RTC (by default).
Can you describe the exact commands you used? 
Does the issue also occur without transferring the image to another SD card?
I just started up my Pi, after it was off for a few days, and I didn't get the issue.

A solution/workaround is changing `/etc/default/rcS` from
```
# automatically repair filesystems with inconsistencies during boot
#FSCKFIX=no
```
to 
```
# automatically repair filesystems with inconsistencies during boot
FSCKFIX=yes
```
This is an actual difference between a raspberrypi.org image and a netinstaller installation.

deafmalice, 2015-01-26:
I'm both making and flashing the image on Xubuntu. I've used both `dd` and `dcfldd` to flash it.

The issue manifests itself without transferring the image only if I mount the SD card on another computer which has the correct time before booting up the RasPi. You can (hopefully) duplicate this issue if you set your system time at least 24 hours in the future compared to the one on the Pi (or the last time the Pi was shutdown), mount the SD card on it, which will update the last mount time, and then boot the Pi with it.

Another issue that I have noticed is that on a flashed system it can't find eth0 and I have to manually configure it to use eth1, but this is a totally different issue. Just letting you know.

diederikdehaas, 2015-01-26:
Ok, wanted to make sure that Windows defaulting to local time (as opposed to UTC) for the hardware clock wasn't at play here.

I'm pretty sure that I would be able to duplicate the issue with the procedure described, since it would make total sense in that case. But I consider that appropriate behaviour and not an issue that needs to be fixed (in the installer). If it's a common issue for a user he can implement the aforementioned fix/workaround (optionally in `post-install.txt`)

Regarding the eth0 issue, see https://github.com/debian-pi/raspbian-ua-netinst/issues/103

goranche, 2015-01-26:
this is correct behaviour, and the error message actually explains everything... when shutting down, [fake-hwclock](https://packages.debian.org/unstable/main/fake-hwclock) stores the current time...
when you mount the image on another system, it's last mount time get's updated...

now you boot the image on the RPi, the stored time gets set (fake-hwclock), obviously before the image itself is mounted (not sure about this one, would have to check) and suddenly the mount time is in the future, which the system doesn't like  :wink: 

you could disable fake-hwclock, if you're sure you don't need it... if you regularly leave your RPi off for several days, this might be useful anyway

deafmalice, 2015-01-26:
Thanks a lot for your help. 

@goranche I don't think I want to ditch fake-hwclock. The problem was because the image was flashed on another SD a few days later. Thanks for pointing out that this is expected behaviour.

One more thing I'd like to ask - I have set up the system so it has a read-only root (it's supposed to be a highly reliable setup in case of power failure). Will any of what you have proposed have problems with that? 

My setup:

    /dev/mmcblk0p2 on / type ext4 (ro,noatime,data=ordered)
    /dev/mmcblk0p3 on /var type ext4 (rw,noatime)
    /dev/mmcblk0p3 on /home type ext4 (rw,noatime)

Precautions I have taken:

- linked  `/var/local/system-links/fake-hwclock.data` to `/etc/fake-hwclock.data`
- added `/sbin/fake-hwlocal load` to `rc.local` to get a reasonably correct time in case of reboot without NTP access

goranche, 2015-01-26:
if your root is readonly, then fake-hwclock doesn't make sense anyway, as it will not be able to write the current time on shutdown anyway...
also, you'll have to have at least some parts of the system writeable, probably best in a ramdisk...

I've done quite some work on a readonly (well, basically, although I could do upgrades and all) system, and it's quite doable... still, YMMV

diederikdehaas, 2015-01-26:
> I've done quite some work on a readonly (well, basically, although I could do upgrades and all) system, and it's quite doable... still, YMMV

Feel free to share :wink: (wiki?)

goranche, 2015-01-27:
> Feel free to share

I'm trying to convince that particular client to let me open source everything... :angel:

deafmalice, 2015-01-27:
You might be right about mounting /var as rw not being stable, but that's the only option I could find at the time. In worst case scenario a ro / should allow for basic recovery in case of failure (should keep at least the integrity of files there). In my case /tmp is tmpfs, so that's that. 

Would a system even work with read-only /var?

goranche, 2015-01-27:
you can use overlaid filesystems, where you have one part thats readonly all the time, and an overlay that's readwrite...
in that case the system will boot, and you can write a script that checks if the readwrite part is valid, and recover if it isn't (that, in a nutshell, was my solution)

diederikdehaas, 2015-01-27:
> I'm trying to convince that particular client to let me open source everything... 

Thanks fot trying :+1:

deafmalice, 2015-01-27:
Thanks a lot for helping. And thanks for the tips (though I still have a lot till I get to overlaying fs).

closed 2015-01-27 by deafmalice