debian-pi/raspbian-ua-netinst #181: First boot after Install fails (mount error)

aspeteRakete, 2015-02-10:
Hi there,
I just tried to do a fresh install. When the Pi reboots it stops and says:
> Begin: Running /scripts/init-bottom ... mount: No such file or directory
> done.
> Could not copy file: No such file or directory
> Target filesystem doesn't have requested /sbin/init
> No init found. Try passing init=bootarg.
> (initramfs)

aspeteRakete, 2015-02-10:
installer-config.txt is standard with extra
> packages=python,aptitude # comma separated list of extra packages

diederikdehaas, 2015-02-10:
Which installer version did you use? And on which Pi type?

aspeteRakete, 2015-02-10:
1.0.6
on Raspberry Pi B, tried several units and SD-Cards.

aspeteRakete, 2015-02-10:
During install I also get a lot of 
> Warning: cannot read table of mounted Filesystem: No such File or directory

diederikdehaas, 2015-02-10:
I have not seen this issue before, so maybe it's a change in the archive. Will have to do a new install to see whether I can replicate it.

diederikdehaas, 2015-02-10:
I've just done a fresh install on a B with 1.0.6 and it finished just fine here.
Do you have a proper power supply? And the SD cards you used, are they very cheap ones or reputable ones?
Does an official image from the Raspberry Pi Foundation work properly?

diederikdehaas, 2015-02-10:
> During install I also get a lot of
>>   Warning: cannot read table of mounted Filesystem: No such File or directory

I only get that twice during my install and that's both when it tries to generate an initramfs ...

aspeteRakete, 2015-02-10:
Ok I finally nailed it down. I copied the defaults from README.md to my installer_config.txt and changed the line i wrote in the first comment. The complete file looks like:
````
preset=server
packages=python,aptitude # comma separated list of extra packages
mirror=http://mirrordirector.raspbian.org/raspbian/
release=wheezy
hostname=pi
domainname=
rootpw=raspbian
cdebootstrap_cmdline=
bootsize=+50M # /boot partition size in megabytes, provide it in the form '+<number>M' (without quotes)
rootsize=     # / partition size in megabytes, provide it in the form '+<number>M' (without quotes), leave empty to use all free space
timeserver=time.nist.gov
ip_addr=dhcp
ip_netmask=0.0.0.0
ip_broadcast=0.0.0.0
ip_gateway=0.0.0.0
ip_nameservers=
online_config= # URL to extra config that will be executed after installer-config.txt
usbroot= # set to 1 to install to first USB disk
cmdline="dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1
elevator=deadline"
rootfstype=ext4
rootfs_mkfs_options=
rootfs_install_mount_options='noatime,data=writeback,nobarrier,noinit_itable'
rootfs_mount_options='errors=remount-ro,noatime'
```
Is one of these settings not up to date anymore?

diederikdehaas, 2015-02-11:
Nope, they all look fine to me. Note however that you only have to provide the parameters which you want different from the defaults. Hence the following would do the same:
```
packages=python,aptitude
```

But the installer configuration wouldn't explain the issue you're having.
Can you get the `/var/log/raspbian-ua-netinst.log` from the sd card and post it somewhere? Remove any sensitive info if needed.

diederikdehaas, 2015-03-04:
What's the status on this issue?

diederikdehaas, 2015-03-21:
No response in more then 2 weeks, thus closing.

closed 2015-03-21 by diederikdehaas