debian-pi/raspbian-ua-netinst #185: Syntax error when running update.sh

ChrisMagellan, 2015-03-03:
When I'm running update.sh I'm getting,

update.sh: 61: update.sh Syntax error: "(" unexpected (expecting "}")

Any ideas? I confirmed the required packages were installed. I'm running Ubuntu 64-bit Server 14.10.

goranche, 2015-03-03:
what version are you using, or did you use the "latest and greatest" from github?

ChrisMagellan, 2015-03-03:
1.0.6

goranche, 2015-03-03:
@ChrisMagellan could you also check your bash version, please :thought_balloon:

ChrisMagellan, 2015-03-03:
GNU bash, version 4.3.30(1)-release (x86_64-pc-linux-gnu)

goranche, 2015-03-03:
:smile: ok, even newer than mine... weird

ChrisMagellan, 2015-03-03:
Yeah, it's a pretty new install of Ubuntu Server.

goranche, 2015-03-03:
@ChrisMagellan could you try running the script directly using bash:

    bash update.sh

the line you're seeing the error in creates an array, which is a bash extension (that's why I asked about the version)... I have a suspicion that your bash isn't really bash (not sure how that would come to be, though)

ChrisMagellan, 2015-03-03:
Ah, yep, that's working now. Thanks!

goranche, 2015-03-03:
@diederikdehaas you think might want to change the shebang lines to something like this:

    #!/usr/bin/env bash

also, reopening this issue, as it might lead to an improvement of the scripts :wink:

goranche, 2015-03-03:
> yep, that's working now

@ChrisMagellan you're welcome... but I'm still reopening the issue, because this is something that could probably be handled better in the scripts... thanks for the issue :+1:

diederikdehaas, 2015-03-03:
It's working fine on my  (Debian Sid) machine and my bash version is 
`GNU bash, version 4.3.33(1)-release (x86_64-pc-linux-gnu)`

@ChrisMagellan What does `ls -l /bin/bash` output with you? I get
`-rwxr-xr-x 1 root root 1033720 Jan 28 17:59 /bin/bash`

ChrisMagellan, 2015-03-03:
`-rwxr-xr-x 1 root root 1029720 Oct  7 11:15 /bin/bash`

diederikdehaas, 2015-03-03:
Thanks Chris, I see no important difference there.
Just tried it on my server (`GNU bash, version 4.3.30(1)-release (x86_64-pc-linux-gnu)` and
`-rwxr-xr-x 1 root root 1029624 Nov 13 00:08 /bin/bash`
and I had no problems there either ... weird

@goranche why do you think using `/usr/bin/env bash` is an improvement?
After doing some [research](https://duckduckgo.com/?q=%2Fbin%2Fbash+vs+%2Fusr%2Fbin%2Fenv+bash) I've found it is different, but not necessarily an improvement.

ChrisMagellan, 2015-03-03:
Ah, I think I found the issue. I was running it as `sh update.sh` but my `/bin/sh` is symlinked to  **dash**, not bash. Looks like an Ubuntu thing.

diederikdehaas, 2015-03-03:
ok, thanks Chris :+1: 

I always run it as `./update.sh`

goranche, 2015-03-03:
>  I was running it as `sh update.sh`

that explains a lot :smile: 

> why do you think using `/usr/bin/env bash` is an improvement?

@diederikdehaas because it will find any bash in the path... some systems (LFS pops to mind, although not sure about the versions last couple of years) might install some "sh" (dash) into bin, and bash into /usr... :thought_balloon:

diederikdehaas, 2015-03-03:
> I was running it as sh update.sh

The behaviour actually surprises me since I thought the shebang forced the use of bash.

> because it will find any bash in the path

And thereby it get dependent on the (correct) setting of PATH, not sure if that's in improvement.
I have experienced that the value of PATH can be different based on what you're doing (when trying to replace busybox-static with busybox; don't recall the specifics on the top of my head). 
And from [wikipedia](http://en.wikipedia.org/wiki/Env):
> This makes the script more portable, but also increases the risk that the wrong interpreter is selected because it searches for a match in every directory on the executable search path. It also suffers from the same problem in that the path to the env binary may also be different on a per-machine basis.

But if you feel that using `/usr/bin/env bash` is better, I'll merge your PR.

kpfleming, 2015-03-03:
The shebang line is only used when the script is executed directly. When the script's name is passed as input to a program, that program can decide what it will do, and I don't think any shells will parse the shebang line and replace themselves with the listed program. If you run 'sh python_program.py', the 'sh' shell will try to execute the Python code in that file, and probably fail :-)

diederikdehaas, 2015-03-03:
Ok, thanks for the explanation :+1:

goranche, 2015-03-04:
> also increases the risk that the wrong interpreter is selected because it searches for a match in every directory on the executable search path

@diederikdehaas or the right interpreter, if I (for example) install a newer version of bash in my ${HOME}/bin and add that to the path :wink: 
I do however agree with the path to env part, though  :thought_balloon: maybe removing the path might be a better option, but will have to try that out first :smile:

diederikdehaas, 2015-03-04:
> maybe removing the path might be a better option

Apparently that won't even [work](http://stackoverflow.com/questions/5549044/whats-the-difference-of-using-usr-bin-env-or-bin-env-in-shebang#5549091):
>Unix systems will try to run <interpreter> using execve, which is why it must be a full path, and #!env without a path will not work.

goranche, 2015-03-04:
ok, so I don't need to test that :smile:

sheim-dev, 2015-03-04:
The <code>#!/usr/bin/env bash</code> shebang supports host systems with bash somewhere in <code>$PATH</code> out of the box, whereas the <code>#!/bin/bash</code> shebang carries an additional requirement of the precise location of bash. For a more elaborate discussion on the preferred shebang see http://stackoverflow.com/questions/21612980/why-is-usr-bin-env-bash-superior-to-bin-bash. IMHO the pull request from @goranche should be merged.

There's no way to protect somebody (like OP @ChrisMagellan) from calling <code>update.sh</code> as <code>sh update.sh</code> directly though; this will always result in an error, given the bash-specific constructs in the <code>update.sh</code>.

diederikdehaas, 2015-03-04:
Done.

closed 2015-03-04 by diederikdehaas