debian-pi/raspbian-ua-netinst #190: No space left on device (1.0.7-beta)

Mausy5043, 2015-03-07:
As promised.
Tried to build a Raspberry Pi with the 1.0.7-beta:
```
gzip: 
stdout: No space left on device
update-initramfs: failed for /boot/initrd.img-3.18.0-trunk-rpi with 1.
dpkg: error processing initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E
: 
Sub-process /usr/bin/dpkg returned an error code (1)

E: Internal error: install

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```
(I could've sworn there was already an issue logged for this, but I can't find it anymore.)

Rebooting as I type with `bootsize=+64M` in `...\init.d\rcS`

diederikdehaas, 2015-03-07:
A bit more info is needed here.
What were you trying to do? When did it fail? Do you have a log file you could share? What configuration files did you use?

tais, 2015-03-07:
I have had the same issue, raspi B2, following config

preset=server
mirror=http://mirrordirector.raspbian.org/raspbian/
release=jessie
hostname=xxxxx
domainname=xxxx.local
rootpw=xxxxxxxxx
cdebootstrap_cmdline=
timeserver=nl.pool.ntp.org
ip_addr=xxx.xxx.xxx.xxx
ip_netmask=xxx.xxx.xxx.xxx
ip_gateway=xxx.xxx.xxx.xxx
ip_nameservers=xxx.xxx.xxx.xxx

Mausy5043, 2015-03-07:
Relevant part of the logfile is up there. I'm afraid I destroyed the rest. If you really need it I could re-create  another one later. But I don't see the point right now.
Some background: R-PI B-type, 4GB SD card. I built the SD-card image as usual. Then inserted it into the Pi.

installer-config.txt:
```
packages="apt-utils,avahi-daemon,bc,git,htop,libnss-mdns,logrotate,lsof,nano,nfs-common"
packages="$packages,python,raspi-copies-and-fills,rsync,screen,sudo"
hostname=raspberrypi
cmdline="dwc_otg.lpm_enable=0 console=tty1 noatime loglevel=2 cgroup_enable=memory elevator=noop"
rootfstype=f2fs
label=$hostname
```
I'm watching a new boot right now with the `bootsize=+64M`. That seems to have done the trick.

diederikdehaas, 2015-03-07:
So it fails during the installation?

tais, 2015-03-07:
yup, runs out of space as it seems

Mausy5043, 2015-03-07:
Yes.
Mine just past the `post-install` flawlessly.
I'm calling this one solved.
Shall I pull the solution?

tais, 2015-03-07:
yup, it's rebooting now, bootsize 50M seems to be too small for the B2 install, needs atleast 64M I suppose if it works with Mausy5043, I had it at 70M and it works well

diederikdehaas, 2015-03-07:
@Mausy5043 that's fine (propose it to the v1.0.x branch)

I/we aren't sure yet what the best/full solution is though. I already figured it probably needed to be expanded to 64M by default, but @goranche suggested that doubling it would probably be better (since you'd have very little space left). Another solution would be to delete the RPi1 installer.cpio and installer kernel on a RPi 2 (and vice versa) to reclaim some space.

diederikdehaas, 2015-03-07:
@Mausy5043 could you modify your PR? And while you're at it, also update the README.md to show the new default?

Mausy5043, 2015-03-07:
@diederikdehaas I agree. 64M is a bit tight. Works for now.
What would you like changed?

diederikdehaas, 2015-03-07:
> What would you like changed?

you proposed it to be merged with master and I'd like that to be the v1.0.x branch

Mausy5043, 2015-03-07:
Oh right. Sorry about that.

diederikdehaas, 2015-03-07:
np

diederikdehaas, 2015-03-07:
@Mausy5043 your install was on a Pi 1, right?

Mausy5043, 2015-03-07:
Yes Pi 1 (B).

diederikdehaas, 2015-03-07:
Ok, wanted to make sure the right kernel was getting installed.

This issue should be resolved by the merge of @Mausy5043's PR.
I'll leave it open for now so we don't get others issues about this same problem.

Mausy5043, 2015-03-07:
Up and running:
```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  3.7G  1.1G  2.6G  30% /
[...]
/dev/mmcblk0p1   61M   41M   21M  66% /boot
```

On 1.0.6:
```
/dev/mmcblk0p1   48M   37M   12M  77% /boot
```
Seems we need about >=30% free space on the boot partition to allow for unzipping of packages during installation.

pgrandin, 2015-04-19:
I just faced this issue today. Can you guys please release a new image? This might impact quite a few people. Thanks!

goranche, 2015-04-20:
oh, that's fun... :unamused:

diederikdehaas, 2015-04-22:
This issue should now be fixed with the release of [v1.0.7-RC](https://github.com/debian-pi/raspbian-ua-netinst/releases/tag/v1.0.7-RC)

Mausy5043, 2015-04-22:
I'll test this weekend. It's someones birthday next monday and he gave me the day off to celebrate it. :wink:
How better to celebrate it than by testing a new release. :smile:

diederikdehaas, 2015-04-22:
haha :+1:

Mausy5043, 2015-04-25:
It's working on v1.0.7-RC

closed 2015-04-25 by Mausy5043

aykevl93, 2015-04-26:
I hit this issue while upgrading the `initramfs-tools` package.
I worked around it by moving `/boot/initrd.img-3.12-1-rpi` to `/home/initrd.img-3.12-1-rpi`, installing the update, and moving it back. I haven't tried rebooting yet.

Mausy5043, 2015-04-27:
I think that any installations created with the (old) 3.12 kernels will have this problem when trying to upgrade. As the newer kernels/initrd.img require more space on the /boot partition than is currently provided.
If you want to upgrade you'll need to rebuild with the 1.0.7-RC release or find a way to grow the /boot partition to 128MB.