debian-pi/raspbian-ua-netinst #193: rcS script assumes /boot will be the same partition before and after install

hadess, 2015-03-08:
If I use the "Windows" way of creating an install image (all the files from bootfs/ in a VFAT fs directly on the block device, no partition), the image will fail to continue after having created the new partitions.

We'd get to here, with bootpartition == /dev/sda.

```
echo -n "Initializing /boot as vfat... "
mkfs.vfat $bootpartition &>/dev/null || fail
echo "OK"
```

Is it possible to split out the "bootpartition" as used early to copy the installation files, and the one after fdisk which will be the "installed" /boot?

(Note that I tested this in qemu, but I would think that the same would apply to installation on a physical machine)

goranche, 2015-03-09:
you shouldn't use the block device directly to create a filesystem on it...

hadess, 2015-03-09:
> you shouldn't use the block device directly to create a filesystem on it...

That's the way most removable storage ends up being formatted by default, and that's the easy way to format it under Windows. Furthermore, it allows me to create an image that I can boot in Qemu, as a normal user (using mtools).

goranche, 2015-03-09:
> that's the easy way to format it under Windows

which doesn't make it the right way...
also, I think you're wrong here, but can't check at the moment (don't have a windows machine handy)...

@diederikdehaas I'm leaning towards closing this issue... any additional thoughts?

hadess, 2015-03-09:
You'll want to clarify the installation instructions then.

diederikdehaas, 2015-03-09:
What are you trying to accomplish? Iow what's the use case?

hadess, 2015-03-09:
The final use case is automating the creation of Pi MusicBox' install images. To do this, I'm using QEmu to do the installation. I don't want the build script ("buildroot.sh") to run as root, especially when it's so brittle (if I have an ISO image loop-back mounted, that script will break). I use mtools to copy the file to the original image.

The layout of the target image is unmodified.

diederikdehaas, 2015-03-09:
I still have no idea what you're trying to accomplish ...

> I don't want the build script ("buildroot.sh") to run as root

The things `buildroot.sh` does, need super user privileges, so I don't see how that can be changed.

> especially when it's so brittle (if I have an ISO image loop-back mounted, that script will break)

Patches to make it more stable/resilliant are of course welcome

hadess, 2015-03-09:
With the changes mentioned above, I can run this as a normal user:
```
#!/bin/sh -e

if [ ! -f kernel-qemu ] ; then
	# See http://xecdesign.com/qemu-emulating-raspberry-pi-the-easy-way/
	wget http://xecdesign.com/downloads/linux-qemu/kernel-qemu
fi

./build.sh
# See https://github.com/woutervanwijk/Pi-MusicBox/issues/236 for default size
# dd if=/dev/zero of=Pi-MusicBox-installer.img count=1 bs=1002438656
truncate -s +1002438656 Pi-MusicBox-installer.img
mkfs.vfat -n "Pi-MusicBox-installer" Pi-MusicBox-installer.img

IMAGE="`pwd`/Pi-MusicBox-installer.img"
pushd bootfs/
mcopy -i $IMAGE * ::/
popd

# FIXME: Use -nographic ?
qemu-system-arm -kernel kernel-qemu -initrd bootfs/installer-rpi1.cpio -cpu arm1176 -m 256 -M versatilepb -no-reboot -serial stdio -append "root=/dev/sda1 panic=0 ro" -hda Pi-MusicBox-installer.img
```

And it creates a bootable image without needing root.

hadess, 2015-03-09:
There's a way to fix this:
* For copying the files from /boot, loop through the vfat filesystems (blkid -p -n vfat) to find the one with the boot files.
* For finding the device to format, use the parent block device of the block device we found for /boot
* For booting, use filesystem UUIDs in /etc/fstab

That means that we should be able to solve both #193 and #34 in one go, keeping the image bootable on both QEmu and bare metal. What do you think?

diederikdehaas, 2015-03-11:
I've been trying to understand what it is you're trying to accomplish, but either I failed or I don't see how it would improve the installer for everyone.
If the goal is to have better/proper support for use with QEMU, I agree. But as I said in #34, I think QEMU support should be handled at build time.
And from what I've read the firmware looks for the first **partition** with a vfat filesystem. 

So I don't see this as an improvement (for everyone), but I do see several potentials problems with this. So I'm closing this issue for now. 
Feel free to convince me of my error and I'll happily reopen it.

closed 2015-03-11 by diederikdehaas

hadess, 2015-03-11:
I want to use the same image with QEmu and bare metal. I want to use the same image so I can do development in QEmu and test that same image on the Raspberry Pi.

> And from what I've read the firmware looks for the first partition with a vfat filesystem.

So? That doesn't change. It's the first partition, it's just that it takes the whole disk.

In the comment above, I mentioned 3 fixes. The first one is the only one that deals with how I create the image (and I would have thought you'd be interested in removing the need to be root to generate the images), the other 2 deal with /dev/mmcblk0 vs. /dev/sda depending whether you use a VM or the physical machine.

goranche, 2015-03-11:
> It's the first partition, it's just that it takes the whole disk.

in that case it's not a partition anymore, is it now?

> I would have thought you'd be interested in removing the need to be root to generate the images

if it means introducing additional requirements and dependencies (which it does) on the build machine, then I'm against it... using sudo, the ability to build the image (and not much else) can be given to normal users as well, should you have a situation, where you're developing on a system without root access (which I kind of doubt is the case these days)

> I want to use the same image with QEmu and bare metal

"QEMU support should be handled at build time"

hadess, 2015-03-11:
> if it means introducing additional requirements and dependencies (which it does)

Pray tell which ones? It uses mtools.

> using sudo, the ability to build the image (and not much else) can be given to normal users as well, should you have a situation, where you're developing on a system without root access (which I kind of doubt is the case these days)

Using sudo or root is bad practice, if it can be avoided. It can be avoided. So it really shouldn't be necessary.

> "QEMU support should be handled at build time"

QEmu support doesn't need to be handled at build time, and it kind of defeats the purpose of using a VM for development. If I make changes to the VM and it works, being able to dd the image to the SD card and test it directly on the machine is a nice touch.

I'm fine with the image supporting either RPi1 or RPi2, as long as it's actually RPi1/QEmu and RPi2/QEMu.

I don't understand what the problem is here. I'm offering to write the code for that feature, and then you can make an informed decision based on that code.

diederikdehaas, 2015-03-11:
> * For copying the files from /boot, loop through the vfat filesystems (blkid -p -n vfat) to find the one with the boot files.

We try to move away from making assumptions and instead (try to) make things configurable and this change would move in the opposite direction. And that's assuming that the described heuristic would indeed do the right thing each time.
Furthermore `blkid` is not present in busybox.

>  * For finding the device to format, use the parent block device of the block device we found for /boot

See previous response regarding assumptions -> configurable.
If `usbroot=1` is present in the configuration, the parent block device of `/boot/` is different from the one for `/` (root). 

> * For booting, use filesystem UUIDs in /etc/fstab

Having UUIDs in /etc/fstab makes the [use of initramfs a requirement](http://www.raspberrypi.org/forums/viewtopic.php?p=685676&sid=b889c46e1ecc83907a79da90c8090e84#p685676) as opposed to optional as it is now.

hadess, 2015-03-11:
> We try to move away from making assumptions and instead (try to) make things configurable and this change would move in the opposite direction. And that's assuming that the described heuristic would indeed do the right thing each time.

You're installing libblkid1 in the installer's initrd for nothing to use it? I would imagine that adding the util-linux package wouldn't make that much of a difference. You could probably loop through the usual suspects (/dev/mmcblk0, /dev/mmcblk0p1, /dev/sda, /dev/sda1) because finding the /boot you booted from isn't a configuration option.

> See previous response regarding assumptions -> configurable.
> If usbroot=1 is present in the configuration, the parent block device of /boot/ is different from the one for / (root). 

But the new /boot will be on the same block device as the old /boot, might not be the same /root though, granted.

> Having UUIDs in /etc/fstab makes the use of initramfs a requirement as opposed to optional as it is now.

Not if we use PARTUUID, which GPT supports. And because you can have both GPT and MBR partition tables synchronised, this makes it easy not to break boot if the person adds another USB drive later on.