debian-pi/raspbian-ua-netinst #194: Allow overriding the RPi version to install for

hadess, 2015-03-08:
After installing the system through QEmu:

```
$ cat /boot/config.txt 
[piUNKNOWN !!]
kernel=vmlinuz-
initramfs initrd.img- followkernel
```

It should be possible to override the detected Pi version through a command-line argument, so that the config.txt and image created is usable on a real Raspberry Pi.

```
raspbian-ua-netinst.hw=1
```

for example. Let me know if that's acceptable, I'll create a patch.

goranche, 2015-03-09:
this might be a good idea :thought_balloon: 
although probably not through a command-line argument, but an entry in the config file...

diederikdehaas, 2015-10-08:
How about defaulting to Pi 1?

diederikdehaas, 2015-12-15:
With the merging of #309 I think this issue is resolved.

closed 2015-12-15 by diederikdehaas