debian-pi/raspbian-ua-netinst #199: Stuck at rainbow

Mausy5043, 2015-03-15:
Using the 1.0.7-beta on a Raspberry Pi type B (rpi1). I now get stuck at the rainbow screen.
I've eliminated hardware issues by formatting the SD and installing [the raspbian.org installer](http://archive.raspbian.org/installer/rpi_installer_08-19-12.zip) which boots without a hitch.

I've tried adding `boot_delay=1` in `/boot/config.txt`, but to no avail. :-(

There seems to be a problem with the kernel.img. Anybody else have this problem?
Suggestions are welcome.

Mausy5043, 2015-03-15:
As the kernel seems to be the root of the problem. Is there an easy way to force a specific kernel?

goranche, 2015-03-16:
I had the same thing happen on a B+, but didn't have time to investigate it at the time, so just switched to a B, where the installer (same SD card) worked without a hitch... :thought_balloon:

diederikdehaas, 2015-03-16:
Does the installer fail to start at all or is the installed system the problem?
Do you have a log file you can share and configuration files used?

goranche, 2015-03-16:
in my case the installer failed

diederikdehaas, 2015-03-16:
then it would be useful to test whether a newly build image would fix that since there has been a new kernel version released since v1.0.7-beta was released/build.

Mausy5043, 2015-03-16:
The rainbow image is the last thing the GPU does before loading the kernel and handing over to the ARM. That fact and the fact that a different kernel doesn't have theis problem, has me thinking there is something wrong with the current 3.18.0 kernel up for download (used by the installer). That's why I was wondering if there's an easy way to select a different kernel so I can eliminate that as a possible cause.

goranche, 2015-03-16:
you could build your own installer, just replace the appropriate deb files in the packages dir before doing a ./build.sh (after the update.sh)
also, replacing the kernel file on the SD card should work :thought_balloon: 
or copying additional ones and picking the right one in config.txt

diederikdehaas, 2015-03-16:
can you follow the advice I wrote earlier (about 6 hours ago) first? 
That would use a newer version of the same kernel.

diederikdehaas, 2015-03-16:
I've build a new installer and with it I got a kernel panic.
But when I tried it with v1.0.7-beta everything went fine, including the boot with the new kernel ...

Mausy5043, 2015-03-17:
@diederikdehaas : 
1. No joy on the homemade installer off of the 1.0.7-beta branch :disappointed: .
2. Downloaded the raspbian-ua-netinst-v1.0.7-beta.zip, that one boots into the installer kernel :smile: Unfortunately, once the installer rebooted the RPi got stuck at the rainbow screen again. :disappointed: 

Really wondering what's going on here... :question: :question:

@goranche : You're talking to fast :confused: I'm really hoping this problem will go away by itself. I think it's a kernel problem, so maybe with a next release of the kernel it'll be fixed :thought_balloon: At the moment I have very little time for experiments due to my daytime job requiring a lot of attention and an Arduino project keeping me and the RPi on the testbench busy.

goranche, 2015-03-17:
LOL ok, I'll type slowly... better? :wink: 

I tried building the installer and copied the resulting image to a microSD card...
* after plugging it into a RPi B (using an adapter), it hung at the rainbow screen
* took the same micro SD card out and plugged it into a B+, same result
* tried it once more on a RPi 2, worked without any problem

not sure what's happening, also, would try some more, but have a fever, so not leaving my bed for now

Mausy5043, 2015-03-17:
@goranche sorry to hear you're not feeling well. Get well soon! But, good to hear you were able to reproduce my problem.
I'll see what I can test in the limited time I have available in the coming days. :sweat_drops:

diederikdehaas, 2015-03-18:
As I see it now is that v1.0.7-beta sometimes works, but not always and it looks like the newer kernel makes the problem bigger.

If you want to experiment with different kernels, I think changing the value of `KERNEL_VERSION_RPI1` in both `update.sh` and `build.sh` should change the kernel for the installer itself.
If you want to change the kernel for the installed system, you should change the value of `kernel_meta_package` in `scripts/etc/init.d/rcS` around line 122.
(I haven't verified whether this works, but I think it will)

diederikdehaas, 2015-03-18:
I have just tried a new install with v1.0.7-beta and the install itself went fine, but on reboot I got a kernel panic again.
I've posted [2 logs of the kernel panic](https://gist.github.com/diederikdehaas/11741b09eefc7f1eb827) with a newly build installer, thus having the 3.18.5-1~exp1+rpi15 kernel. Others I've seen are essentially the same. :-/

Mausy5043, 2015-03-18:
Well, at least you get it to panic :wink: 

I'll try your suggestion with a different kernel. BTW, any hints where I might find a list of available kernel-image filenames?

diederikdehaas, 2015-03-18:
if you have a working Pi, do `aptitude search ^linux-image`

Mausy5043, 2015-03-18:
OK. Installer booted with `3.12-1-rpi` kernel succesfully. The `3.12-1-rpi` kernel was also installed and that didn't boot right away because (from the log):
```
Configuring bootloader to start the installed system...
Kernel version: ''
Copying kernel to /boot/kernel.img... cp: can't stat '/rootfs/boot/vmlinuz-': No such file or directory
FAILED !
```
I was able to repair this by editing the `config.txt` file on my Mac. Near the end of `scripts/etc/init.d/rcS` it says:
```
# find kernel/initramfs version
KERNEL_VERSION=$(sed -n -e "/^Package: ${kernel_meta_package}$/,/Package: /{s/^Depends: .*linux-image-\(.*-rpi2\{0,1\}\).*$/\1/p}" /rootfs/var/lib/dpkg/status)
echo "Kernel version: '${KERNEL_VERSION}'"
```
As per instructions I made this change: `kernel_meta_package="linux-image-3.12-1-rpi"`
This results in a `config.txt` that looks like this:
```
[rpi1]
kernel=vmlinuz-
initramfs initrd.img- followkernel
```
Easily repaired.
R-Pi is now up and running. :smile:

diederikdehaas, 2015-03-18:
Ah yeah of course. It's trying to find the dependency of the kernel_meta_package, but in this case there is none. Good to know :+1:

diederikdehaas, 2015-03-19:
Have you seen [my last post on issue #201](https://github.com/debian-pi/raspbian-ua-netinst/issues/201#issuecomment-83263437) ?

I did another test with the v1.0.7-beta installer ... and the installation was successful and it rebooted into my wheezy system without problems! Rebooted again and again I did NOT have any problems (kernel panic/rainbow). 
So that seems to correspond with my earlier remark that the failure is inconsistent.

Then I did `/etc/kernel/postinst.d/initramfs-tools $(uname -r)` to trigger the (re-)generation of the initramfs and rebooted my system ... and got a kernel panic/rainbow.
And then I did the same thing as I described in the other issue (disable initramfs by commenting out that line in `/boot/config.txt` ... and it booted successfully again :triumph: 

So it looks like the problem is indeed in (the generation of) the initramfs ... dunno why though.

dragetd, 2015-03-21:
Okay, here are my findings so far:

I have two RPi (Version 1, built 2011) and two different SD-cards I shuffled around, tho so far there is no sign of a hardware issue with either the SD-card or Pi.

Using the current 1.0.6 image, the installation for wheezy works up until the point the system reboots, then I get the rainbow screen. As far as I remember, booting without the initramfs did not make any difference. Kernel 3.12 (for the installer) seems to always work without issues for me, but the 3.18 does not. Did not try manually replacing the kernel yet.

Then I tried 'jessie' with the 1.0.7-beta image (with the 3.18 kernel). The installer loads, but at some point creating the initramfs fails with no space left on device - tho choosing +64M for boot fixed this. When rebooting, I get the rainbow screen again. Booting without the initramfs does load the kernel tho! It just fails to mount boot obviously.

The confusing question now is: Why would the initramfs be able to break the kernel on the Pi in such a way that it even fails to init. the framebuffer, thus only showing the test-screen of the videocore?

diederikdehaas, 2015-03-21:
Here are various kernel dumps when the kernel panics:
[kernel panic wheezy Pi B](https://gist.github.com/diederikdehaas/11741b09eefc7f1eb827#file-kernel-panic-raspi-b-txt)
[kernel panic wheezy Pi B+](https://gist.github.com/diederikdehaas/11741b09eefc7f1eb827#file-kernel-panic-raspi-bplus-txt)
[jessie-3.18-withinitramfs](https://gist.github.com/dragetd/8a5844016a553854867b)
[jessie-3.18-noinitramfs](https://gist.github.com/dragetd/d09e41c7957c5af211e4)

Hopefully this will help (someone) to figure out where the issue is

diederikdehaas, 2015-03-30:
Based on a hint from plugwash on IRC, I installed the `raspberrypi-bootloader` package which installs the raspberry pi foundation kernel. Rebooted into it and generated an initramfs by doing `/etc/kernel/postinst.d/initramfs-tools $(uname -r)` and specified that initramfs file in `/boot/config.txt`. 
Rebooted ... and got a kernel panic! This means that the problem is also present in the raspberry pi foundation kernel!
Log file: [kernel panic wheezy Pi B+ rpf kernel](https://gist.github.com/diederikdehaas/9a67b6afd902ce75788e)

diederikdehaas, 2015-03-31:
@Mausy5043 and @dragetd, can you do the following for me?
- install and run `rpi-update`, that'll get you the 3.18.10+ kernel
- make sure there are no kernel/initramfs lines enabled in `/boot/config.txt`
- reboot into the 3.18.10+ kernel and then run `/etc/kernel/postinst.d/initramfs-tools $(uname -r)` to generate an initramfs file for that kernel
- specify the following in `/boot/config.txt`: `initramfs initrd.img-3.18.10+ followkernel`
- reboot and tell me whether you still get a rainbow/kernel panic

Mausy5043, 2015-04-01:
@diederikdehaas : Sure I'll try this over the coming weekend.

diederikdehaas, 2015-04-02:
plugwash made a new kernel (not available in the archives yet) and during the 5+ tests on each of my Pi's I have not seen a kernel panic :smile: 

If you would like to test it (preferably today) and would report back, that would be great.
To do so, add the following code to `post-install.txt`
```sh
new_kernel=""
if [ ! -d /rootfs/root/ ] ; then
  mkdir /rootfs/root/
fi

if [ "$rpi_hardware" = "$RPI1_HARDWARE" ] ; then
  echo -n "Downloading new new kernel for RPi 1... "
  wget -q http://plugwash.raspbian.org/linux_3.18/linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1%2brpi16_armhf.deb -O /rootfs/root/linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16_armhf.deb
  if [ $? -eq 0 ]; then
    echo "OK"
  else
    echo "FAILED !"
  fi
  new_kernel=linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16_armhf.deb
elif [ "$rpi_hardware" = "$RPI2_HARDWARE" ] ; then
  echo -n "Downloading new new kernel for RPi 2... "
  wget -q http://plugwash.raspbian.org/linux_3.18/linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1%2brpi16_armhf.deb -O /rootfs/root/linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1+rpi16_armhf.deb
  if [ $? -eq 0 ]; then
    echo "OK"
  else
    echo "FAILED !"
  fi
  new_kernel=linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1+rpi16_armhf.deb
else
  echo "Unable to determine RPi hardware version."
  echo "rpi_hardware value: '$rpi_hardware'"
fi

if [ "$new_kernel" != "" ] ; then
  echo "Installing new kernel '$new_kernel'"
  chroot /rootfs /usr/bin/dpkg -i /root/$new_kernel
  if [ $? -eq 0 ]; then
    echo "Kernel installation was successfully."
  else
    echo "Kernel installation FAILED !"
  fi
fi
```

Mausy5043, 2015-04-02:
@diederikdehaas : Tested that on the hardware that caused this issue. I renamed the config.txt, so no new installer was built (so, installer was still on 3.12). It just rebooted:
```
17:19:04 up 5 min,  0 users,  load average: 0.19, 0.22, 0.11
Linux rbups 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv6l GNU/Linux
```
Seems to work :+1: 

Relevant part of the log:
```
Removing cdebootstrap-helper-rc.d... OK
Configuring apt... OK
Creating preferences file for libraspberrypi-bin... OK
Updating package lists... OK
Installing bootloader package (=firmware)... OK
Installing libraspberrypi-bin package (=vcgencmd)... OK
Configuring bootloader to start the installed system...
Kernel version: ''
Copying kernel to /boot/kernel.img... cp: can't stat '/rootfs/boot/vmlinuz-': No such file or directory
FAILED !
Creating default cmdline.txt... OK
=================================================
=== Start executing post-install.txt. ===
Downloading new new kernel for RPi 1... OK
Installing new kernel 'linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16_armhf.deb'
Selecting previously unselected package linux-image-3.18.0-trunk-rpi.
(Reading database ... 15080 files and directories currently installed.)
Unpacking linux-image-3.18.0-trunk-rpi (from .../linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16_armhf.deb) ...
Setting up linux-image-3.18.0-trunk-rpi (3.18.5-1~exp1+rpi16) ...
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-3.18.0-trunk-rpi
df: Warning: cannot read table of mounted file systems: No such file or directory
warning: failed to read mtab
Kernel installation was successfully.
```

Mausy5043, 2015-04-02:
I assume I no longer need to test the `rpi-update` ?

diederikdehaas, 2015-04-02:
@Mausy5043 Thanks for testing :+1:  and no, there's no need for `rpi-update` any more

Mausy5043, 2015-04-02:
What to do with my easter weekend now :grey_question: 
:stuck_out_tongue_winking_eye:

diederikdehaas, 2015-04-02:
There are still a number of [issue open](https://github.com/debian-pi/raspbian-ua-netinst/issues) ... :stuck_out_tongue_winking_eye:

Mausy5043, 2015-04-25:
Let's close this one then. v1.0.7-RC fixes this issue

closed 2015-04-25 by Mausy5043