debian-pi/raspbian-ua-netinst #200: Feature Request: LUKS Whole Disk Encryption

varialus, 2015-03-16:
The Debian Installer offers the option to set up LUKS whole disk encryption and I'd like for this option to also be available when installing from raspbian-ua-netinst.

I'd like this feature to be added in a modular way, but I probably won't manage to implement it in a module way myself. I'm not very good at modifying shell scripts and I don't know how long it will be before I'll have time to work on the feature myself. I've found three Raspberry Pi specific tutorials that explain how to set it up, but I haven't run through them yet. They all use a small SSH server to unlock the disk and I think they all set everything up directly onto the SD card rather than onto a USB hard drive. Because I'm not very good at modifying shell scripts, once I get around to working on this feature, I'll probably only work toward getting it to work for my specific use case of installing onto a USB hard drive and not including the SSH server. If I'm able to easily make it modular and well designed, I will.

Here are the addresses of those three tutorials.
http://paxswill.com/blog/2013/11/04/encrypted-raspberry-pi/
https://www.offensive-security.com/kali-linux/raspberry-pi-luks-disk-encryption/
https://www.ofthedeed.org/posts/Encrypted_Raspberry_Pi/

goranche, 2015-03-16:
this is something I might be interested in as well, what's your use case for this?
(just so I can understand your requirements / expectation)

varialus, 2015-03-16:
I'm using my Raspberry Pi 2 for occasional development and testing. Encryption isn't an absolute requirement, but I always just use encryption on physical hardware as a matter of course.

I'm doing cross-platform development in the Go programming language. I don't yet have continuous integration set up, so for the time being I just rotate my development and testing between 5 or 6 environments that are representative of the many environments on which the Go tools can be built and run. I don't mind spending a bit of time improving my development environment each time I switch environments, but I've already spent way too much time trying to get the Debian Installer to run and when that failed, just trying to get Debian installed and configured the same as if I had used the Debian Installer. I will continue to try to improve my setup, but I'd prefer to only occasionally spend a small amount of time on it.

varialus, 2015-03-16:
I've got my Raspberry Pi 2 hooked up to a decent monitor, a keyboard and mouse, a large USB SSD, and an ethernet connection.

I've also got a CD/DVD burner that works out of the box which I got to try to boot from the official Debian installation media, but I never managed to boot from it. I figure that if I could just get the SD card to boot from the CD, then I could install to my USB SSD and then configure my SD card to boot to the USB SSD, but I don't know how to configure the SD card to boot from the installation CD.

diederikdehaas, 2015-03-16:
I like the idea :+1: but as I have no expierence with disk encryption it will probably take a while for me to implement it.

varialus, 2015-03-16:
That's no problem. I'll appreciate any help at any time. And a big thanks to everybody who has helped make the installer what it is today.

I'm also not all that familiar with setting up encryption since all I usually have to do is type in my password. I'm not familiar with the manual OS installation either since installers usually shield me from all the details. It can all be learned, but I understand as well as anybody, that it takes time.

goranche, 2015-03-16:
I'm trying to figure out a way how to do this for headless machines... :thought_balloon:

diederikdehaas, 2015-03-16:
By having an SSH server (dropbear was mentioned in one of the articles) in initramfs?

goranche, 2015-03-16:
that would require me to login to the machine, which is kind of useless when you have over 300 of these deployed :angel:
(and to top it off, they get turned on around the same time)

I'm thinking of turning it around, have a script in initramfs "call home", but still have to think this through...

varialus, 2015-03-22:
I've figured out how to set this up for my particular use case and I've published the step by step process in the hopes that it might perhaps help you improve the raspbian-ua-netinst installer.

http://varialus.wikidot.com/rpi2-linux

Let me know if you happen to notice any mistakes or if anything doesn't make sense.

Mausy5043, 2017-12-09:
Closing this issue for now, since a working and tested PR seems not to be forthcoming. If you feel the closure is in error, please feel free to re-open and add new information.

closed 2017-12-09 by Mausy5043