debian-pi/raspbian-ua-netinst #201: Installation of Debian Jessie fails

ondrej1024, 2015-03-18:
Hi,

I tried running the installer with a custom installer-config.txt were I specified the Debian jessie release by adding the following line:
   release=jessie
But after the installation the RPi doesn't reboot anymore. I see just the coloured image in the screen, no kernel messages.

Same thing happens when upgrading to jessie after successfully installing wheezy.

Did I do something wrong?

Ondrej

diederikdehaas, 2015-03-18:
Which version of the installer did you use and on what Pi system (B/B+/2)?

ondrej1024, 2015-03-18:
I am using this installer version: raspbian-ua-netinst-v1.0.6.img.xz on a RPi B.
Installation of wheezy works fine, but both upgrade and direct install of jessie gives me the rainbow screen after reboot.

diederikdehaas, 2015-03-18:
It looks like you're running into the same problem as reported on #199.
But it's symptoms are not entirely consistent. Earlier today I tried an installation (of wheezy) and that succeeded, but I got a kernel crash (=rainbow) when it rebooted.

ondrej1024, 2015-03-18:
I installed wheezy twice today and everything went fine. The problem is only with installing or upgrading to jessie. Here the procedure finishes successful but the RPi gets stuck (rainbow) after reboot. Too bad.

diederikdehaas, 2015-03-18:
Ok, thanks. I'll look into it since it may be a different issue.

Mausy5043, 2015-03-18:
This makes me suspicious of the current batch of kernels.

diederikdehaas, 2015-03-19:
I was able to confirm your finding @ondrej1024 
This is what I did:
- `xzcat raspbian-ua-netinst-v1.0.6.img.xz > /dev/sdd` (sdd = sd card reader/writer)
- put the sd card in my Pi B and booted up
- after the install it automatically rebooted into my wheezy system ... without any problems
- changed `wheezy` to `jessie` in `/etc/apt/sources.list` and did `apt-get update` and `apt-get dist-upgrade` and noticed that there was a newer version of the meta-package `linux-image-rpi-rpfv` but NO new kernel version
- after the upgrade was done I rebooted ... and got a kernel panic/rainbow

And then I did something to try out:
- pulled the power plug of my Pi B to power it off
- put the sd card in my sd card reader/writer and opened `/boot/config.txt` and commented out the `initramfs ...` line (deleting has the same effect)
- removed the sd card from my reader/writer, put it back into my Pi B and powered it up
- watched how the boot process completed SUCCESFULLY :triumph: 

So I have a VERY strong feeling that the initramfs is the problem and likely also with issue #199 
So @Mausy5043 and @ondrej1024 can you commented out (place # in front) the initramfs line in `/boot/config.txt` to see whether that fixes the problem?

Mausy5043, 2015-03-19:
@diederikdehaas : tried your suggestion (removing initramfs line from config.txt).
The installer boots and panics.

diederikdehaas, 2015-03-19:
Damn, I thought I had found the problem. 
Thanks for testing and reporting back :+1:

Mausy5043, 2015-03-19:
I have three RPis. One B+ and two Bees. Only one of the Bees is presenting this problem. The other Pies are on 3.18-0. Could this be a hardware issue?

diederikdehaas, 2015-03-19:
It could be related to hardware, but that would still make it a kernel problem since that's what interacts with it (or at least should do)

ondrej1024, 2015-03-19:
Sorry but I didn't have time to test this myself. Will try tomorrow and report back.

ondrej1024, 2015-03-21:
Hi, just did the test as proposed by diederikdehaas. After commenting out the initramfs line in /boot/config.txt the Pi boots into a fresh jessie system. So for me the trick worked. Great workaround but we still need a proper fix, I guess.

diederikdehaas, 2015-03-21:
We sure do need a proper fix, but we first need to figure out exactly what is going wrong.
A solution/workaround that work for some ppl/systems seem to not work on others.

Can you do another (short) test?
In a terminal do `/etc/kernel/postinst.d/initramfs-tools $(uname -r)` to trigger the (re-)generation of the initramfs, enable the initramfs line in /boot/config.txt again and reboot.

Mausy5043, 2015-03-21:
@diederikdehaas : not sure what you mean. A terminal on which system?

diederikdehaas, 2015-03-21:
on the Pi. And by terminal I meant a terminal window/application (CLI, like xterm/LXTerminal/Konsole/etc).

ondrej1024, 2015-03-21:
Well, that didn't work. I get the rainbow screen again after reboot.

diederikdehaas, 2015-03-21:
Ok, thanks for trying :+1:

diederikdehaas, 2015-04-02:
@ondrej1024 if you have time, could you try [my comment on issue #199](https://github.com/debian-pi/raspbian-ua-netinst/issues/199#issuecomment-88896087)?
If you don't want to install a fresh (Pi 1 B/B+) system, but want to try the new kernel, do this (as root or with sudo):
```sh
wget -q http://plugwash.raspbian.org/linux_3.18/linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1%2brpi16_armhf.deb -O linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16_armhf.deb
dpkg -i linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16_armhf.deb
```
and enable the initramfs line in `/boot/config.txt`

For the Pi 2, do this:
```sh
wget -q http://plugwash.raspbian.org/linux_3.18/linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1%2brpi16_armhf.deb -O linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1+rpi16_armhf.deb
dpkg -i linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16_armhf.deb
```

I'll also test this myself tomorrow, but if you can do it too, that would be great. If not, no worries.

diederikdehaas, 2015-04-13:
The maintainer of raspbian (plugwash) has released a new kernel version (the rpi16 one) and I'm not able to reproduce the issue any more. 
@ondrej1024 Can you try whether it does work correctly now?

ondrej1024, 2015-04-13:
Hi, sorry for not getting back earlier. I was quite busy the last days. I will try tomorrow and report back here.

diederikdehaas, 2015-04-13:
np :-)

diederikdehaas, 2015-04-13:
Damn, I just did another test and now it is crashing on a Pi 1B, but not on a Pi 1B+ :sob:

ondrej1024, 2015-04-14:
I just tested again installing directly Debian Jessie on a RPi1 B and now it works. No more rainbow screen after reboot. So something definitely has changed.

diederikdehaas, 2015-04-14:
Thanks :+1: 
The new kernel is surely an improvement, but the last 2 days I've been trying to create a new installer and now it always fails on a Pi 1B :-/
It seems to be an 'upstream' kernel issue: https://github.com/raspberrypi/linux/issues/914#issuecomment-92817668, but that also impacts the kernel used by the installer.

m4rcu5, 2015-04-19:
I am currently testing v1.0.7-beta with Jessie. One thing I noticed was the installer failing due to a to small default boot size. Except for that, it installed fine =)
```
Raspbian GNU/Linux 8 pi ttyAMA0

pi login: root
Password: 
Linux pi 3.18.0-trunk-rpi2 #1 SMP PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv7l
```

diederikdehaas, 2015-04-22:
With the [release of v1.0.7-RC](https://github.com/debian-pi/raspbian-ua-netinst/releases/tag/v1.0.7-RC) that should now (also) be fixed.

eLvErDe, 2015-05-04:
Hello,

It doesn't seem to work for me, I got a rainbow screen on B+ after installation.

diederikdehaas, 2015-05-04:
do you have DeviceTree (DT) enabled?
To disable DT, the following line should be present in `/boot/config.txt`:
```
device_tree=
```
With my testing, disabling DT was all that was needed to make it boot properly.

diederikdehaas, 2015-05-05:
This issue is now fixed with the release of v1.0.7 of the Raspbian unattended netinstaller, which disables DeviceTree on Pi 1B/1B+ by default..

closed 2015-05-05 by diederikdehaas