debian-pi/raspbian-ua-netinst #207: Power LED is off after boot

Phiber2000, 2015-03-25:
I have a B+ v1.2 here. Everything's working fine.
But while booting, the power led gets dark and keeps this state.

I added "echo 1>/sys/class/leds/led1/brightness" to /etc/rc.local but I think that's a __very dirty__ fix.

Tested with v1.0.6 & 1.0.7 beta.

diederikdehaas, 2015-03-26:
I just did a fresh install on a Pi 1 B+ with 1.0.7-beta and my power led is on. 
(I did get a kernel panic on reboot, but that's reported in #199 and #201)
Do you have a proper power supply?

Apart from that, it doesn't seem like it's a problem with the installer and you're more likely to find a solution to your issue on the [raspberry pi forums](http://www.raspberrypi.org/forums/)

diederikdehaas, 2015-03-26:
Hmm ... I just disabled the initramfs line in `/boot/config.txt` so I could boot up again ... and the power led is indeed off!

Phiber2000, 2015-03-26:
Thanks for the hint #119 & #201! That explains why I had problems after the first reboot and had to went back to v1.0.6...
But I'm pretty sure, that the led was off, too. I can't ensure this, so forget it.
Anyway - an rpi-update didn't solve the problem with v1.0.6. I thought, that replaces/updates the kernel, too!?

Are there any workarounds or fixes?

diederikdehaas, 2015-03-26:
There are multiple 'funny' issues at play here. 
You said you did have success with v1.0.6, but that installs the same kernel, but this time it was successful. 
I had somethings similar when I installed the 3.12 kernel because that triggers a re-generation of the initramfs, including the one for the 3.18 kernel. Thus it is essentially the same 'fix' as I described [in issue 199](https://github.com/debian-pi/raspbian-ua-netinst/issues/199#issuecomment-83274697).
Rebooted it and then I did not get the kernel panic. Another work-around that sometimes helps (with the kernel panic) is disabling/removing the initramfs line in `/boot/config.txt`.
Booting with 3.18 kernel consistently makes the power led off, while it works with the 3.12 kernel.

`rpi-update` does indeed install a new/different kernel, but in order to use it, you need to disable the kernel and initramfs lines in `/boot/config.txt`. That way it takes the default kernel (kernel.img for RPi 1 and kernel7.img for RPi 2) and that's what rpi-update installs/updates.

Phiber2000, 2015-03-27:
So I hope, they deploy a fixed kernel soon.

By the way:
I killed a RPi model B five minutes before your last answer by executing rpi-update - just to see what happens... As expected, the so called "rainbow screen" happened.
Commenting out the two lines of the config.txt, as you suggested, did the trick and saved me a lot of time.

So I have to say thanks a lot twice: First for the detailed answers, second for the raspbian-ua-netinst project.
I don't want to miss it! For the reason alone that "cleaning" a Raspian image can't & isn't an option and will never be sufficient.

Phiber2000, 2015-03-27:
(Closed the issue accidently)

diederikdehaas, 2015-04-02:
I can now consistently reproduce this issue on both Pi 1 B+ and Pi 2 B, but not on a Pi 1 B ...

The rainbow issue may be fixed fairly soon. 
If you'd like to test a pre-release version, see https://github.com/debian-pi/raspbian-ua-netinst/issues/199#issuecomment-88896087
That new kernel still has the power LED issue though.

diederikdehaas, 2015-04-02:
@Phiber2000 if you add the following line to `/boot/config.txt` does the LED work then with you?
```
# disable device tree
device_tree=
```

Phiber2000, 2015-04-02:
I did some "research", too. First: It's definitely fixed with kernel version 3.18.10 ([Kernel-Image](https://github.com/Hexxeh/rpi-firmware/blob/master/kernel.img)).
Second: On Pi 1 & 1B up to revision 2.1 (all), the Power LED is directly connected to ground, so it's not software controlled there. ([Schematics](http://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md))
Since Pi 1B+ the power LED is controlled by a transistor and the old kernel doesn't set it to ground (switch on) at boot time.

@diederikdehaas Because my Pis are in production use, I can't test your config suggestion this week. But I think this is an issue, which is not "caused" to your script and If you want, you could close it.
My proposal is as following: Version 1.0.7, could include the new kernel (I will test plugwashs kernel with the device_tree line as soon as possible) and things are done...
If the config change has no effect, we have to use Hexxehs v3.18.10 maybe.

diederikdehaas, 2015-04-02:
I'm 99.9% sure it's related to device tree. Enabling/disabling device tree is like a switch to turn off/on the power led. Device Tree support is also the main difference between the kernel used in the installer (3.18.5-1~exp1+rpi9) which doesn't have this issue and the current kernel (3.18.5-1~exp1+rpi15/16) which has.
I'll talk to plugwash tonight to discuss the rainbow issue and I'll likely throw this issue in too and see whether a fix can be found.

You're right that this issue isn't caused by the installer (script) itself, but the installer has been more then once the cause for finding issues with packages in the raspbian archive, just like #199 (and there are more). 
So while not strictly the fault/responsibility of the installer, I do consider these kind of issues relevant and thus leave it open until it's fixed :smile:

Phiber2000, 2015-04-02:
Add another 0.1%. I catched a B+, updated to kernel 3.18.5-1~exp1+rpi16 from #199. LED was off.
Then I disabled device tree usage and the power LED is on.

fg2it, 2015-09-28:
@diederikdehaas  Is this issue supposed to be closed ? As far as I understand, the release 1.0.7 was supposed to fix that. Unfortunately, I am facing this problem. I used release 1.0.7 to install jessie on a rpi2 and have problem with the power led. It is on a few second at power up, then is switch off to go on again at shutdown. Disabling device tree allows a normal situation.

Phiber2000, 2015-09-28:
@fg2it Because it's a kernel bug you have to follow the steps desribed here: ["Clean" Raspbian Kernel Update](https://github.com/debian-pi/raspbian-ua-netinst/issues/227#issuecomment-125665524)
That the raspbian kernel to the latest developed by raspberrypi.org.
With the updated kernel, the issue is gone. This step is necessary until raspbian updates his repo with a patched version.

diederikdehaas, 2015-09-28:
@fg2it This issue is deliberately open because, as you noticed, the issue is not fixed.
I don't know when this issue will be fixed, so I suggest to follow @Phiber2000's procedure to install the raspberrypi.org kernel.

fg2it, 2015-09-28:
@Phiber2000 @diederikdehaas thanks for the responsiveness ! Will try the suggestion

diederikdehaas, 2015-10-15:
I just came across a forum post which _MAY_ be the cause of the Power LED issue: https://www.raspberrypi.org/forums/viewtopic.php?p=815647#p815647:
```
At a guess I'd say CONFIG_LEDS_TRIGGER_INPUT=y is missing.
```
And that setting is indeed missing in the raspbian.org kernels ...

diederikdehaas, 2015-10-15:
Apparently I hadn't filed a raspbian bug for it, but I have now: https://bugs.launchpad.net/raspbian/+bug/1506663

I found some more things (described in that bug report) which makes me quite sure that that setting is indeed the cause of this issue and the fix seems quite trivial :relieved:

Phiber2000, 2015-10-15:
That's too simple to be the solution. :)
I look forward to the answers!

diederikdehaas, 2016-08-09:
When my latest PR is merged, this issue should be fixed as well :smile:

diederikdehaas, 2016-08-09:
As it has just been merged, I'm closing this issue.
If someone still experiences it, based on the latest v1.1.x branch, feel free to reopen it.

closed 2016-08-09 by diederikdehaas