debian-pi/raspbian-ua-netinst #210: RAMTMP ignored in jessie + systemd

martinml, 2015-04-07:
In 6422a16012b302c6ca8f1e3666361ad52dade9b6, `RAMTMP` was enabled so `/tmp` were mounted as `tmpfs`. In **jessie**, which runs `systemd` by default, this variable is ignored.

A solution would be modifying directly the `fstab` file (more compatible) or running `systemctl enable tmp.mount` (more simple). For the latter, since the filesystem is not available yet, I guess a symlink from `/etc/systemd/system/local-fs.target.wants/tmp.mount` to `/lib/systemd/system/tmp.mount` would be enough.

More info: https://lists.debian.org/debian-user/2014/07/msg01314.html

Mausy5043, 2015-04-09:
One of the first things I do when I modify an installed system is to modify `/etc/fstab` to put `/tmp` on tmpfs. There's nothing there that needs to survive a reboot and it always works.

diederikdehaas, 2015-05-05:
Thanks for the report :+1: (and sorry for the late reply)

Personally I'm in favor of the `fstab` solution since it works regardless of init system used and I think it's the most clear one. It also makes it rather easy to shrink/enlarge the size of `/tmp`.

martinml, 2015-05-05:
For reference, this is the script that generates the corresponding mount command in wheezy and jessie (`/lib/init/tmpfs.sh`), as it might be interesting to use those defaults: http://pastebin.com/raw.php?i=V35Gbih8

diederikdehaas, 2015-06-23:
@Mausy5043 can you show the exact `/etc/fstab` line you use for putting `/tmp` on `tmpfs`?

@martinml did you install jessie by using `release=jessie`?
I've just upgraded a wheezy system to jessie and `/tmp` was mounted on `tmpfs`. Which surprised me since I expected the behaviour as reported in this issue.
I will also install jessie directly soon to see how that behaves, but still like know/re-create the exact situation as you got it.

diederikdehaas, 2015-06-23:
```
tmpfs           /tmp    tmpfs   nodev,nosuid    0       0
```
in `/etc/fstab` in jessie seems to produce
```
$ grep '/tmp' /proc/mounts 
tmpfs /tmp tmpfs rw,nosuid,nodev,relatime 0 0

$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   15G  943M   13G   7% /
udev             10M     0   10M   0% /dev
tmpfs           186M  4.3M  182M   3% /run
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
tmpfs           464M     0  464M   0% /tmp
/dev/mmcblk0p1  122M   35M   88M  29% /boot
tmpfs            93M     0   93M   0% /run/user/1000
```
which doesn't look too bad imo :smile:

martinml, 2015-06-24:
@diederikdehaas Yes, this was a fresh installation using `release=jessie`, and IIRC I used the last stable installer by the date that I opened the issue, so that would be 1.0.6.

I probably changed the `bootsize` option too, to `+512M`. I'm sorry I don't remember any other details of the config I used.

diederikdehaas, 2015-06-24:
Thanks :+1: 

I'm pretty sure the `release` parameter would be the only one relevant here. 
Another explanation could be that one of the Debian packages got 'patched' so TMPFS would keep working even under systemd. But in the Debian bug report a DD was quite clear that it was a sysv-init specific thing. 
Besides that, the `fstab` solution is more visible and therefor more easily adjustable.

Mausy5043, 2015-06-24:
@diederikdehaas : I use these almost all the time:
```
tmpfs /tmp     tmpfs nodev,nosuid,mode=1777,size=30M              0 0
tmpfs /var/log tmpfs defaults,noatime,nosuid,mode=0755,size=100M  0 0
```

diederikdehaas, 2015-06-24:
Thanks @Mausy5043 :+1: 

What's the reason for `mode=1777` in your `/tmp` line?
I have seen it before, but never could find the reason for it.

Mausy5043, 2015-06-25:
@diederikdehaas : `mode=1777` is a mount option which gives `rwx` permissions to everybody.

When I started putting `/tmp` in `tmpfs` I researched it shortly and found that everybody seems to mount it like this.

diederikdehaas, 2015-06-25:
Looks like `mode=1777` is the default, so you don't have to specify it.
The `stat <dir>` command is rather interesting/useful.

Mausy5043, 2015-06-26:
Yeah well, all examples I find on the line include the `mode` option. 

Better safe than sorry I guess. I don't normally check if I'm using options that are part of the `default`. I tend to include options that I *really* need despite them being part of the `default` option. Its just easier to read later and doesn't require you to look up what are the default options for a specific fs and prevents breakage when the fs-maintainers decide to change the default options.

N.B.: The 1 in 1777 sets the sticky bit. (see here: https://askubuntu.com/questions/432699/what-is-the-t-letter-in-the-output-of-ls-ld-tmp). But, you probably already knew that.

diederikdehaas, 2015-06-29:
This issue is now fixed with the merge of https://github.com/debian-pi/raspbian-ua-netinst/pull/264

closed 2015-06-29 by diederikdehaas