debian-pi/raspbian-ua-netinst #212: NFS-root not possible because NFS not compiled in kernel

olmari, 2015-04-23:
I've using netinst to install raspberry minimal setup, and then making raspberry boot from NFS share as root-folder, instructions from here: http://blogs.wcode.org/2013/09/howto-netboot-a-raspberry-pi/

Problem is that kernel that is provided with netinst isn't capable of NFS in boot time: 

"nfsmount: bad option 'vers' "

When with NOOBS raspbian kernel same setup works... I've hunted it down to this, (I think):

"In order to use nfsroot, NFS client support (in kernel) needs to be selected as
built-in during configuration. Once this has been selected, the nfsroot
option will become available, which should also be selected.

In the networking options, kernel level autoconfiguration can be selected,
along with the types of autoconfiguration to support. Selecting all of
DHCP, BOOTP and RARP is safe."

So could you please include that in upcomin releases? I'm currently using 1.0.7-RC

diederikdehaas, 2015-04-23:
I'm not familiar with NFS myself, but I think that you need to enable it in the initramfs.
Do this by editing `/etc/initramfs-tools/initramfs.conf` and specify `BOOT=nfs`.
After that you should generate a new initramfs by executing `mkinitramfs -o /boot/initrd.img.netboot` and then specify `initramfs initrd.img.netboot` in `/boot/config.txt`.

I don't know whether it then works or some other things need to be set as well.
Hopefully someone with knowledge/experience with NFS knows exactly what to do and chimes in.

Note that I/we don't make the kernel that is used, but use the one provided in the raspbian archives.
If you want it changed in the archives, you should report it on https://bugs.launchpad.net/raspbian

diederikdehaas, 2015-06-23:
What's the status of this issue?

https://help.ubuntu.com/community/DisklessUbuntuHowto#line-155 seems to specify what you need to do wrt the initramfs and NFS.

diederikdehaas, 2015-06-30:
No response in a week, thus I'm closing the issue.
If you feel the issue has not been dealt with, feel free to reopen it again.

closed 2015-06-30 by diederikdehaas