debian-pi/raspbian-ua-netinst #214: SPI devices don't show up unless Device Tree is used

CRImier, 2015-04-27:
Hello!
First of all, thank you for this beautiful piece of software =) I'm really happy to see there's something that allows me to cleanly install Raspbian on Raspberry Pi.

The problem I see is that if Device Tree is disabled and spi modules (spi_bcm2708 automatically, later also spidev manualy) are loaded, /dev/spidev0.0 and 0.1 don't show up. If I enable DT and use config.txt dtparam options to enable SPI, it all works, both /dev/spidev* are there and they're working as well. I used raspi-config to make changes. Surprisingly, it didn't prompt me about "Enable SPI interface?", only about if module should be loaded on boot - which it does.

There's a difference in 'dmesg|grep spi', too - if SPI is working, there are three more lines, they say something like "** translation for device /soc/spi@hexhumbers **", and one line complaining about how there are no DMA ranges found. When SPI devices are not there, there are only 2 lines about SPI controller initialization. 

Thank you for your help in advance!

CRImier, 2015-04-27:
Forgot it - I use jessie release (installed just today). Everything worked fine just yesterday without DT on NOOBS Raspbian image upgraded to jessie. I do acknowledge it might be some kind of bug in raspi-config and friends that appeared just yesterday, but so far I haven't been able to find it at another bugtrackers.

diederikdehaas, 2015-05-05:
I'm not really clear on what the status of this issue is. Can you clarify that?
Note that raspi-config is a raspberry pi foundation (rpf) 'thing' and that it's not ready/updated for jessie (yet).

closed 2015-06-20 by CRImier