debian-pi/raspbian-ua-netinst #219: Can’t get Real-Time Clock to Work

suksit, 2015-05-04:
I installed Raspbian using ua-netinst successfully but when I followed the guide to add RTC to Raspberry Pi B+ (http://www.drewkeller.com/blog/adding-hardware-clock-raspberry-pi-ds3231), it didn’t work.

I got some error when running <code>sudo hwclock --debug</code> saying that it can’t find <code>/dev/rtc</code>.

So I tried to follow the same guide on the official Raspbian image and the RTC was installed successfully. Any ideas/suggestions?

Thanks!

diederikdehaas, 2015-05-04:
Maybe change `/etc/default/hwclock` and remove the line with `HWCLOCKACCESS=no` ?
This command is probably also needed `update-rc.d fake-hwclock disable`
You probably have to reboot afterwards.
I don't have a RTC device myself, so this is only educated guessing.

suksit, 2015-05-05:
Still didn’t work :/

Thanks for your suggestion, I’ll keep trying…

diederikdehaas, 2015-05-05:
Have you tried removing the `fake-hwclock` altogether?
`apt-get remove --purge fake-hwclock`

suksit, 2015-05-05:
Yep I removed <code>fake-hwclock</code> along with editing <code>/etc/default/hwclock</code>.

However, I think I know why the RTC device doesn’t show up. It’s because I don’t have <code>rtc-ds1307.ko</code> in <code>/lib/modules/...</code> directory.

I got the RTC working by running <code>rpi-update</code> to get the latest kernel and modules. Maybe not the correct way to go but it’s the easiest I guess :P

diederikdehaas, 2015-05-05:
and after running `rpi-update` you do have the `rtc-ds1307.ko` file?

diederikdehaas, 2015-05-05:
Confirmed:
From `/boot/config-3.18.0-trunk-rpi2`: 
`CONFIG_RTC_DRV_DS1307 is not set`
with kernel 3.18.12-v7+ `zgrep RTC_DRV_DS1307 /proc/config.gz` : 
`CONFIG_RTC_DRV_DS1307=m`

suksit, 2015-05-05:
Yes.

With the fresh install I got only like three <code>.ko</code> files in <code>/lib/module/\<version\>/kernel/drivers/rtc/</code>

After running <code>rpi-update</code>, there are like 10-20 <code>.ko</code> files in <code>/lib/module/\<newer-version\>/kernel/drivers/rtc/</code>

diederikdehaas, 2015-07-02:
I asked for support for the same RTC devices as the raspberrypi.org kernel supports: 
https://bugs.launchpad.net/raspbian/+bug/1470679

diederikdehaas, 2015-09-30:
The upstream issue as been resolved and (thus) so is this one.

@suksit Can you verify that that is indeed the case?

suksit, 2015-10-02:
Confirmed that the module is there after a fresh install and RTC clock works as expected.

Thanks for the follow-up :D

closed 2015-10-02 by suksit