debian-pi/raspbian-ua-netinst #225: `rpi-update` and `rpi-firmware`

Mausy5043, 2015-05-09:
I've occasionally seen references to `rpi-update` (https://github.com/Hexxeh) here on this repo.

Since neither Hexxeh's `rpi-update` nor `rpi-firmware` are installed by default I was wondering if you would see some added advantage to regularly running `rpi-update`?

Or do you consider these tools to be used when things are broken?

diederikdehaas, 2015-05-09:
You shouldn't use `rpi-update` unless you're told to (by a rpf member) or when you think it's fun to run the latest and (mostly) **untested** kernel and firmware.
Thankfully ShiftPlusOne (on the forum/irc) is now actively _discouraging_ ppl to use `rpi-update`.

When things are broken (and kernel related) like https://github.com/raspberrypi/linux/issues/914 then it makes sense to _also_ test whether the issue is present in rpf's very latest kernel, but by far most ppl should stay clear of `rpi-update`.

I've never heard of `rpi-firmware` though (`rpi-update` updates both kernel and firmware).
There are [nightly build firmware package](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=80023) which are properly packaged (as opposed to `rpi-update`) and thus manageable through APT

Mausy5043, 2015-05-09:
Thanks. :+1:

closed 2015-05-09 by Mausy5043