debian-pi/raspbian-ua-netinst #226: Can't change keyboard layout

darkbasic, 2015-05-15:
Hi,
I just tried your latest installer (1.0.7), configuring it to install Jessie instead of Wheezy.
Unfortunately changing keyboard layout doesn't work: dpkg-reconfigure keyboard-configuration lets me choose the italian layout but I still have the stock layout, even after a reboot.
dpkg-reconfigure console-data simply leads me to a "looking for keymap to install: none", while after installing console-tools it simply quits.

darkbasic, 2015-05-15:
P.S.
Server preset.

Mausy5043, 2015-05-15:
IIRC v1.0.7 of the installer is meant to install `wheezy`. Does it work with `wheezy`?

darkbasic, 2015-05-15:
AFAIK the installer is meant to install whatever you put in the release= string into installer-config.txt

diederikdehaas, 2015-05-15:
> AFAIK the installer is meant to install whatever you put in the release= string into installer-config.txt

Not 'whatever', but both `wheezy` and `jessie` are supported (`stretch` is not).

Can you share your `installer-config.txt`? And post the contents of `/etc/default/keyboard`?

diederikdehaas, 2015-05-15:
I have very little experience with this, but it looks like I can confirm your issue.
When the system is booted up and I see the console login prompt, I can type € and it is displayed.
When I log in and type € again, nothing is shown.

diederikdehaas, 2015-05-15:
@darkbasic check you have the following packages installed: `console-data`, `console-setup` and `keyboard-configuration`. It's probably also wise to do `dpkg-reconfigure <package>` where <package> is `locales`, and the previous mentioned packages.

darkbasic, 2015-05-15:
The problem was the lack of console-setup.

This is my installer-config.txt:
preset=server
release=jessie

Yes, I have aptitude and I also installed the base tasksel system with:
aptitude install ~pstandard ~pimportant ~prequired

I tought console-setup was meant to be part of tasksel's base system, my bad I didn't check.
Maybe you should add it to your presets.

diederikdehaas, 2015-05-15:
The list of packages that need to be installed by default does need some review.
If you don't specify any packages, ie use the defaults, various **required** packages don't get installed ...

diederikdehaas, 2015-05-15:
Note however that the aforementioned package probably won't be installed by default.
I normally connect to my Pi's through ssh and/or a serial connection and thus never encountered this issue.

Ok to close the issue?

darkbasic, 2015-05-16:
For me it's ok to close, but if you don't mention it somewhere (maybe in the readme?) someone else will surely re-file this bug in the future.

closed 2015-05-16 by darkbasic