debian-pi/raspbian-ua-netinst #227: Net install Jessie: mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO

phlegx, 2015-05-15:
Hi,

DMA cannot run if I use ```raspbian-ua-netinst``` version 1.0.7 and release Jessie. Here my ```dmesg``` log:

```log
[    1.378834] sdhci: Secure Digital Host Controller Interface driver
[    1.378856] sdhci: Copyright(c) Pierre Ossman
[    1.379088] mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO
...
[    1.438686] vc-sm: Videocore shared memory driver
[    1.438717] [vc_sm_connected_init]: start
[    1.439940] [vc_sm_connected_init]: end - returning 0
[    1.440724] /linux-3.18-3.18.5/drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
```

I get an additional problem with RTC: ```/drivers/rtc/hctosys.c: unable to open rtc device (rtc0)```

@pelwell wrotes: "I would guess that the DT blob (bcm27*.dtb) in your boot partition doesn't match the kernel. In our attempt to converge with the upstream kernel there will unfortunately be a few changes, like this one to the DMA driver, that require the kernel and DTB to be kept in sync."

**Can you please update bcm27*.dtb to match the kernel with Jessie?**

Mausy5043, 2015-05-15:
IIRC v1.0.7 is meant to install `wheezy`. Does it work when you install with `wheezy`?

phlegx, 2015-05-15:
No, I can see the same errors after rainbow screen.

diederikdehaas, 2015-05-15:
@phlegx your issue(s) is/are likely caused because devicetree is disabled by default on the Pi 1B/B+.
You can change that in `/boot/config.txt` by commenting out the `device_tree=` line in that file.
If you then reboot there's a reasonable chance that you get a rainbow screen and commenting out the `initramfs` line will probably 'solve' that problem.

The underlying issue is reported here: https://github.com/raspberrypi/linux/issues/914 
But I have no idea whether that will (ever?) be fixed since the raspberry pi foundation doesn't use an initramfs (normally) and thus there's little incentive to fix that issue.

phlegx, 2015-05-15:
I have comment out the ```device_tree=``` and ```initramfs``` line on Wheezy and Jessie using a Pi1. 

**Wheezy**
The system boots without problems but after the rainbow screen I see ```mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO``` and ```/linux-3.18-3.18.5/drivers/rtc/hctosys.c: unable to open rtc device (rtc0)``` in the first lines of output.

**Jessie**
The system boots without problems and the ```dmesg``` command has ```mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO``` and ```/linux-3.18-3.18.5/drivers/rtc/hctosys.c: unable to open rtc device (rtc0)``` in the output log.

diederikdehaas, 2015-05-15:
This very much looks like kernel stuff, so wheezy or jessie is very unlikely to make a difference (and both are supported btw).

It turns out that `CONFIG_DMA_BCM2708=y` isn't present/enabled in the raspbian.org kernel (which is different from the raspberrypi.org (RPF) kernel).
I'm not sure what the cause is for the `rtc` issue, but I'm seeing it too. I thought it wasn't an issue since I don't have a RTC device.

The solution consists of several step and basically entails switching over to the RPF kernel:
- install `raspberrypi-bootloader` package, this should uninstall several other packages, most notably `raspberrypi-bootloader-nokernel`
- comment out in `/boot/config.txt` the `kernel`, `initramfs` and `device_tree` lines. If those are the only once there you can also just delete `/boot/config.txt` (it will then use all the default values)
- reboot

Mausy5043, 2015-05-15:
Just checked the logs. I have the same messages on all my Pi's (1B's and 1B+'s). 
It has not been an issue for me as I don't have RTC-hardware.

phlegx, 2015-05-15:
@diederikdehaas this solves the problem! THX!

diederikdehaas, 2015-05-15:
It's best to report it in the raspbian.org bugtracker: https://bugs.launchpad.net/raspbian
That way it will get fixed for every one.

phlegx, 2015-05-15:
Bug reported: https://bugs.launchpad.net/raspbian/+bug/1455508

diederikdehaas, 2015-05-15:
Thanks :+1:

phlegx, 2015-05-15:
Which kernel file gets loaded if I remove ```kernel=...``` from ```/boot/config.txt```?

diederikdehaas, 2015-05-15:
The default, which is `kernel.img` on a Pi 1B/1B+ and `kernel7.img` on a Pi 2B

phlegx, 2015-05-15:
:+1:  Thx! :smile:

uecasm, 2015-05-20:
FWIW, I'm seeing this error on Pi2 running Wheezy with a custom kernel (based on a1c792ec9c8).  `CONFIG_DMA_BCM2708=y` **is** set in the kernel config (and this was the upstream default).

diederikdehaas, 2015-05-20:
Thanks for the info :+1: 

I compared the kernel configuration from `raspberrypi-bootloader` with the one from `linux-image-3.18.0-trunk-rpi_3.18.5-1~exp1+rpi16` on `BCM2708` and `CONFIG_DMA_BCM2708=y` was the only one relevant.
It would be interesting to know how yours compares to it/them.

uecasm, 2015-06-09:
Seems to be this: https://github.com/Hexxeh/rpi-firmware/issues/82.  I can confirm that making the `config.txt` change suggested in the comments makes the warning go away.  I don't know if it has any negative consequences but it still appears to be working as expected after this.

uecasm, 2015-06-10:
Note that this should be fixed in the latest Pi firmware; anyone still experiencing it, see if an `rpi-update` will do the trick.

mikrogroove, 2015-06-26:
I have updated to the latest firmware with `rpi-update` yet I'm still seeing 
`mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO`
and
`/linux-3.18-3.18.5/drivers/rtc/hctosys.c: unable to open rtc device (rtc0)`

`$ uname -a` outputs 
`Linux pint 3.18.0-trunk-rpi2 #1 SMP PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv7l GNU/Linux`

diederikdehaas, 2015-06-26:
For the msg to go away you need to boot into the kernel provided by `rpi-update` (it's use is deprecated btw) and to do that you need to modify `/boot/config.txt` and disable the `kernel` line. 
You then probably also want to disable the `initramfs` line

mikrogroove, 2015-06-26:
@diederikdehaas: If I comment out the `kernel` and `initramfs` lines in `config.txt` then the Pi refuses to boot.

diederikdehaas, 2015-06-26:
Then `rpi-update` installed a kernel which refuses to boot. Preferably you should report that on https://github.com/raspberrypi/linux/issues so they can (try to) solve that issue.
Now you know why `rpi-update` is deprecated :-P

You could try to fix it by installing the `raspberrypi-bootloader` package, which will ask you to remove the raspbian.org kernel package. That package contains the released (and thus tested) firmware and kernel from raspberrypi.org.

mikrogroove, 2015-06-26:
Ok, so I installed `aspberrypi-bootloader`, and after [a vast amount of overlay configuration malarkey](https://gist.github.com/mikrogroove/120a6dc72c3569f9311f) I again have a non-booting Pi. This is a headless machine so I'm not able to see what the boot screen says. Is there an official way to install a compatible firmware on the Pi2, or am I at the beginning of a long and painful journey? Maybe I'm better off accepting PIO mode?

diederikdehaas, 2015-06-26:
You need to figure out somehow (with monitor or a serial cable) what makes you Pi not boot.
The `raspberrypi-bootloader` package installs both firmware and a kernel, so they're definitely in sync.

Personally I'm not bothered by the message and/or the (apparent) use of PIO.

mikrogroove, 2015-06-26:
Ok, I'll just live with PIO mode for now then. Shame though, as I really could do with the performance boost of DMA. Should I uninstall the `raspberrypi-bootloader` package?

diederikdehaas, 2015-06-26:
> Should I uninstall the raspberrypi-bootloader package?

No, I don't think that's needed.

mikrogroove, 2015-06-27:
I gave `rpi-update` another go after dragging out a monitor and much to my surprise it now booted just fine - afaik I did nothing differently this time. I can confirm that I now see 

`mmc-bcm2835 3f300000.mmc: DMA channels allocated`

in dmesg, and that at least the start-up time appears to be significantly faster. rpi-update said that 

`This update bumps to rpi-4.0.y linux tree`

diederikdehaas, 2015-06-27:
:+1: 

I wouldn't be surprised if the jump to the 4.0 branch may (also) have a significant influence. 
I have the feeling that the 3.18 branch has various immature implementations. (But I'm not exactly an expert on this)

Phiber2000, 2015-07-28:
While the kernel package isn't updated, you DMA and power led can be activated as follows:
```apt-get install -y raspberrypi-bootloader
mv /boot/config.txt /boot/config_install.txt
touch /boot/config_install.txt```

diederikdehaas, 2015-07-28:
That procedure would install the raspberrypi.org kernel and boot into that :wink:

Phiber2000, 2015-07-28:
Yes. And unfortunately that's not the debian way - one of the big advantages of your netinst-project which is based on the "linux-image-rpi-rpfv" or "linux-image-rpi2-rpfv" packages of raspbian.
I don't like this workaround - but it works and in my opinion it's better than the rpi-update way.

mikrogroove, 2015-07-29:
@Phiber2000: In what way, specifically, is your method better than using `rpi-update`?

diederikdehaas, 2015-07-29:
`rpi-update` doesn't integrated with debian/raspbian package management and thus there is no way to remove it like you normally can with `apt-get remove <package>`.

diederikdehaas, 2015-09-30:
@phlegx @Phiber2000 @uecasm @mikrogroove 
Can you test whether this issue is still present? 
I've done various test just now (kernel = 3.18.5-1~exp1+rpi19) and I don't see this issue/warning any more. Furthermore `CONFIG_DMA_BCM2708` is now enabled in the kernel.
(And I have reported that too on the [raspbian bug #14555508](https://bugs.launchpad.net/raspbian/+bug/1455508))

diederikdehaas, 2015-10-12:
No response in almost 2 weeks, thus assuming the issue is fixed (as is the upstream issue).

closed 2015-10-12 by diederikdehaas

IDLabsgate, 2016-10-12:
I had two messages 
`mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO` , this one is gone now but still 

`mmc0: Unable to initialise DMA channels. Falling back to PIO 10` still exist , any advice?

Mausy5043, 2016-10-12:
Which version of the installer are you using?
Which kernel version are you using. There are known issues with previous kernels (https://github.com/Hexxeh/rpi-firmware/issues/82).

This also has a hint of a solution: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=114235

And [here](https://github.com/raspberrypi/linux/issues/682) it is suggested that a mismatch between the kernel and the DTB might have something to do with it.