debian-pi/raspbian-ua-netinst #228: check for required utils

goranche, 2015-05-17:
The scripts don't check if required utils (zip for example) are available on the build machine.

Adding a code section (or separate script to be included) to do so should be quite straightforward :smile: 
In case a needed util doesn't exist, a message should be displayed to the user, ideally with instructions on how to install it.

diederikdehaas, 2015-05-17:
> should be quite straightforward

I'm not so sure about that, when you think of all the different systems this can be used on.
Patches welcome, but it's unlikely I will spend time on this.
What's wrong with `BUILD.md`?

goranche, 2015-05-17:
> when you think of all the different systems

I'm aware of this... that's why I added the "should be" :wink: 

I actually added this as a placeholder for some code I will be working on (and adding as a PR)...

diederikdehaas, 2015-05-17:
Ok :+1:

goranche, 2015-05-21:
huh... curl wasn't found on one of my servers :smile:

diederikdehaas, 2016-02-21:
Ping?

Mausy5043, 2016-03-17:
Closing. 
Feel free to re-open if required.

closed 2016-03-17 by Mausy5043