debian-pi/raspbian-ua-netinst #229: Installer uses pre-defined NTP servers rather than using an NTP pool address

DougieLawson, 2015-05-17:
../scripts/etc/init.d/rcS has hard coded NTP server addresses.
```
    timeservers="$timeservers nisttime.carsoncity.k12.mi.us nist1-lnk.binary.net"
    timeservers="$timeservers ntp-nist.ldsbc.edu utcnist2.colorado.edu"
```

It should use dynamically assigned servers from the pool as per http://www.pool.ntp.org/en/

Use 0.pool.ntp.org & 1.pool.ntp.org for that in the default timeservers lists.

```
    timeservers="$timeservers 0.pool.ntp.org"
    timeservers="$timeservers 1.pool.ntp.org"
```

Right now that gets
```

pi@apollo:~$ nslookup 0.pool.ntp.org
Server:         10.1.1.11
Address:        10.1.1.11#53

Non-authoritative answer:
Name:   0.pool.ntp.org
Address: 195.186.1.100
Name:   0.pool.ntp.org
Address: 82.219.4.30
Name:   0.pool.ntp.org
Address: 91.121.169.20
Name:   0.pool.ntp.org
Address: 91.218.89.74

pi@apollo:~$ nslookup 1.pool.ntp.org
Server:         10.1.1.11
Address:        10.1.1.11#53

Non-authoritative answer:
Name:   1.pool.ntp.org
Address: 213.209.109.45
Name:   1.pool.ntp.org
Address: 212.18.3.19
Name:   1.pool.ntp.org
Address: 212.59.0.1
Name:   1.pool.ntp.org
Address: 213.199.225.40

pi@apollo:~$
```
and the addresses change on a round-robin basis.

diederikdehaas, 2015-05-17:
We're not using NTP, but rdate.
The reason for the hardcoded list is because not every time server supports rdate, but the ones in the list do.
When we switch to using NTP, we'll get rid of that list and use a NTP pool address.

DougieLawson, 2015-05-17:
That's even worse. 

Why not pick a web server that MUST reachable for netinst to work and use the date from the headers.

```
curl -s -D - -o /dev/null https://archive.raspbian.org | awk '/Date:/ {print $2" "$3" "$4" "$5" "$6" "$7;}'
```

diederikdehaas, 2015-05-17:
> That's even worse.

Welcome to the limited world of busybox :wink:  (where your command won't work for several reasons)
Anyway, the switch to NTP is in the works and likely will be in v1.0.8.

diederikdehaas, 2015-06-29:
This issue is now fixed with the merge of https://github.com/debian-pi/raspbian-ua-netinst/pull/261

closed 2015-06-29 by diederikdehaas