debian-pi/raspbian-ua-netinst #232: Failed to set time using timeserver

fduraibi, 2015-05-19:
Using the latest release 1.0.7 on RPi B (512MB) it fails to use any of the time servers, even though it got an IP from DHCP and I can ping it.
```
Network configuration:
  ip_addr = dhcp
  online_config = 

Waiting for eth0... Configuring eth0 with DHCP... 192.168.0.24
Failed to set time using timeserver 'time.nist.gov'.
.....
```

I read that the plan is to switch to ntp instead of rdate, but is there a workaround? since right now it always fails at the GPG verification step.

diederikdehaas, 2015-05-19:
My guess is that port 37 is blocked (and you should have [seen a msg about that](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L257)).

I'm not aware of a workaround you could use if you can't open port 37.

fduraibi, 2015-05-20:
I know my router's firewall is not blocking it, however, you are right, after further testing I believe it is the university's firewall that is blocking it. (I did the test from my android phone since it has busybox with rdate, it worked from home but not from the school's network)

Is there a way to manually set the time/date?

fduraibi, 2015-05-20:
I solved it by installing xinetd on my local machine and enabled the time service, then changed my local domain and machine name to 'time.nist.gov' so the RPi connected to my machine instead.

However, a manual way to enter the time  in case rdate fails will be great.

closed 2015-05-20 by fduraibi

diederikdehaas, 2015-05-20:
My compliments for your ingenuity :+1: