debian-pi/raspbian-ua-netinst #240: U-Boot support

bbinet, 2015-05-22:
Any plans to add U-Boot support?
http://elinux.org/RPi_U-Boot

I'd like to use U-Boot as secondary bootloader so that we can install two raspbian installs on different partitions: the second partition would be used as a fallback rescue mode when the first (main) install fails to boot.

It may also allow network boot via TFTP.

goranche, 2015-06-30:
I don't really see a point in adding this as a install option, as it would still require a lot of work on the users part (installing a separate system on a different partition, creating said partitions, ...)
installing u-boot seems like the easiest (and arguably smallest) step

@diederikdehaas any thoughts?

diederikdehaas, 2015-06-30:
There is a [Debian package](https://packages.debian.org/jessie/u-boot) and 'thus' also a raspbian package, which is pretty much a requirement to consider this in the first place.

But I have never used U-Boot myself and I can't say it's near the top of my wish/TODO list.
So, afaic, this issue can stay open but don't expect it to be fixed in the near future.

diederikdehaas, 2015-06-30:
I have just read some more on U-Boot and it very much seems like it replaces the kernel.
And since the jessie version of U-Boot is 2014.10 it looks very much like RPi2 would not be supported.
If that is true, that would be a NO-GO for me.

goranche, 2015-06-30:
> seems like it replaces the kernel

what do you mean by this? yes, it gets loaded instead of the kernel by the firmware, but then loads a real kernel, depending on various settings...

diederikdehaas, 2015-06-30:
> what do you mean by this?

That I probably should've read the whole article before commenting? :angel:

goranche, 2015-06-30:
:D

Mausy5043, 2016-02-22:
The added value of using/supporting U-Boot for (dual-)booting the raspbian-ua-netinst installer completely escapes me. The RPi comes with it's own bootloader. (ref: [this stackexchange article](https://raspberrypi.stackexchange.com/questions/10489/how-does-raspberry-pi-boot)) which is great.

Choosing to support U-Boot will open up the doors for all other bootladers (grub, lilo, BerryBoot, NOOBs, [etc](http://www.simtec.co.uk/products/SWLINUX/files/booting_article.html)).

goranche, 2016-02-22:
> for (dual-)booting the raspbian-ua-netinst installer

it's not meant to boot the installer... it's meant to be installed on the target system as the secondary bootloader...

but even with that, I fail to see the benefit of including this into the code base... it's for a very specific use case that a very small number of people will use... and even in those few cases the specifics will be wildly different, so ...

I don't really see this as a viable option

goranche, 2016-02-23:
it seems this isn't something anyone wants to spend time on, some (including myself) even think it's not really possible, at least not in a useful / user friendly way...

therefor, I'm closing the issue... should anyone feel that this was premature or would like to give it a shot, please, reopen the issue, and we'll see from there  :+1:

closed 2016-02-23 by goranche