debian-pi/raspbian-ua-netinst #241: raspbian-ua-netinst image RPI won't boot

typhoon71, 2015-05-23:
I tried using raspbian-ua-netinst on a RPI (256MB), both 1.06 and 1.07, but while they both seem to manage to create a SD card with the usual 2 partitions, when the process is finished the SD card won't boot.
Basically I format the SD card to FAT32, copy the files in the archive, plug the RPI; it boots, starts doing its stuff, then after some time it stops with only the red led on.
The card appears to be properly partitioned when read on another linux system.
Last raspian (5/15) works fine, so it should not be the RPI setup or the SD.
I'd like some help to find what's happening as I'd like to use raspbian-ua-netinst instead of cutting out packages from a raspian desktop image.
Thanks a lot.

bmedicke, 2015-05-23:
256MB means it is it a model A or first revision B. According to the readme it works with Pi Model 1B, 1B+ and 2B. I don't have a model A but don't they lack ethernet?

typhoon71, 2015-05-23:
It's a model B, 256MB, with ETH.
It's one of the first RPI (1) made, back in the china days of the project.
So it should be supported.
Personally I think something is failing, but I don't know what I need to check.

bmedicke, 2015-05-23:
Can you take a look at /var/log/raspbian-ua-netinst.log?

typhoon71, 2015-05-23:
Did the 1.07 image change recently? because it appearts to be working now...
Last time I tried (probably with a different image) I couldn't even ping the thing. Strange.
Now I'm installing stuff. I'm both happy (WORKS) and sad (let's just close the issue).
Thanks anyway, I tried again because someone actually answered, if I did get no answer I wouldn't try again).

closed 2015-05-23 by typhoon71

goranche, 2015-05-25:
> Did the 1.07 image change recently

no, the image itself didn't... something might have changed in the debian repositories, though