debian-pi/raspbian-ua-netinst #242: Mountpoint are strange/redundant

typhoon71, 2015-05-25:
I used df -h to check the free space on the SD card; this is the output:

[image]

It seems to me that mounting /run only would be enough, and also / gets mounted 2 times.
Standard raspbian doesn't have the issue.

Can this be fixed? I'm asking both for future installations and for this one. Thanks.

goranche, 2015-05-25:
and what is your issue with this?

typhoon71, 2015-05-25:
Having a clean and tidy minimal Raspbian install, without cluttering orunwanted packages is the main reason one uses raspbian-ua-netinst (I think).

Mounting stuff that's not needed like in /run/* case is not what I'd call clean.
Now I don't remember / being mounted twice in Raspbian stock image (or at least being visible), but if I get some time (later) I can check.

Btw, your "What is rootfs?" copy paste doesn't answer my question, nor explains where the difference between Raspbian stock image and raspbian-ua-netinst originates from (which should've been a more sensible approach).

goranche, 2015-05-25:
> doesn't answer my question

it wasn't meant to answer it, it was just an explanation on why it's "mounted twice"...

the difference between the stock raspbian image and raspbian-ua-netinst is that one is a custom image, while the other is a standard Debian install meant for specific hardware, and here lies your answer...
the mounts are a result of the standard Debian bootup, so ... it'll probably stay like that

goranche, 2015-05-25:
and [another link](http://askubuntu.com/questions/169495/what-are-run-lock-and-run-shm-used-for) to a nice explanation on /run/shm and /run/lock...

the reason for not just using /run? probably because it's a good idea to keep shm and lock on separate filesystems, as they do have different mount parameters, some of which are security related...

so, to get back to the original "issue", I don't agree that the mounts are strange and/or redundant :wink: 
if you agree, I'd close this issue

typhoon71, 2015-05-25:
OK, I'll ask somwhere else.

closed 2015-05-25 by typhoon71