debian-pi/raspbian-ua-netinst #246: Installation of regular Debian Jessie?

Zeerix, 2015-05-28:
What needs to be changed to install a regular Debian Jessie on a rpi2 just like in this link, but with this installer?

http://sjoerd.luon.net/posts/2015/02/debian-jessie-on-rpi2/

goranche, 2015-05-29:
haven't really gone into details of the link, but this installer will pretty much install a "regular" (whatever that means) Debian system onto the RPi2...

so, I guess my question would be: what are you missing in this installer (or rather the result of it), that the image in the link does provide?

Zeerix, 2015-05-29:
This installer installs packages from the Raspbian repository, which are recompiled for the Raspberry Pi 1, which only has a ARMv6 based CPU. The regular Debian packages are compiled for ARMv7-A or newer, so they would not work on the first RPI (see https://www.raspbian.org/RaspbianFAQ#What_is_Raspbian.3F).

The Raspberry Pi 2 has an ARMv7-A, so Debian (i.e. non-Raspbian) binaries could be faster on it, because they can make better use of the CPUs features.

goranche, 2015-05-29:
> This installer installs packages from the Raspbian repository

that is true, now I see what you're asking... I doubt this installer will ever do that, and also, I think a lot of changes should be performed to support it... :thought_balloon:

goranche, 2015-05-29:
@Zeerix have you done a comparison of the two systems? if so, is there a significant difference in performance?

Zeerix, 2015-05-29:
I have not installed it yet. I think I will try it on the weekend.

goranche, 2015-05-29:
> will try it

:+1: it'd be great if you could do some tests and let us know...
maybe it might be worth investigating further

Zeerix, 2015-05-31:
Ok, I tried a netinstalled Raspbian Jessie (A) and the Debian Jessie image from the link above (B).

First test: `sysbench --test=cpu --num-threads=4 run`
* A1) `total time: 109.9807s`
* A2) `total time: 109.9565s`
* B1) `total time: 88.0600s`
* B2) `total time: 87.9953s`

Second test: compiled both `libx264` and `ffmpeg` (see http://www.jeffreythompson.org/blog/2014/11/13/installing-ffmpeg-for-raspberry-pi/), then downloaded a 12 minutes video from youtube and re-encoded it:
`time ffmpeg -y -i input.mp4 -c:v libx264 -preset faster -crf 18 -c:a copy -pix_fmt yuv420p output.mkv`
* A) `real 97m10.573s   user 367m29.000s   sys 1m6.156s`
* B) `real 73m22.552s   user 276m5.616s   sys 0m53.024s`

This looks like an almost 25% improvement.

diederikdehaas, 2015-05-31:
Thanks :+1:

goranche, 2015-06-01:
hm... not bad, actually... :thought_balloon: 
@Zeerix  thanks

SandPox, 2015-08-01:
You can install real debian on Pi2, I'm not test on Pi1

Download this first: https://github.com/debian-pi/raspbian-ua-netinst/releases/download/v1.0.7/raspbian-ua-netinst-v1.0.7.zip
after download, extract it to sdcard

Then download this (debian netinstaller): http://ftp.debian.org/debian/dists/stable/main/installer-armhf/current/images/netboot/initrd.gz
extract it 1 time, you will get file "initrd", rename it to installer-rpi2.cpio and move it to sdcard (overwrite on original raspbian netinstaller installer-rpi2.cpio)
now boot it up, install debian like you do on PC, if it ask something like cannot detect kernel module.. etc just select no or skip it. After install complete, your Pi can't boot (that's normal because the installer format kernel partition), get the sdcard out and plug it to PC, copy the whole stuff from original raspbian netinstaller archive to sdcard again, delete file "config.txt" "installer-rpi1.cpio" "installer-rpi2.cpio", edit file cmdline.txt add "root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait"
it will look like:

consoleblank=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait

now plug it to rPi, it's ready to use, but you gonna need to install rpi-update to upgrade to latest kernel and install some kernel module

Phygon, 2015-10-08:
I think that #271 might have been the reason for the bad performance.
If that is the case, I don't see a reason to have regular Debian Jessie anymore.

Zeerix, 2015-10-08:
(I should use the correct account to comment!)
I think that #271 might have been the reason for the bad performance.
If that is the case, I don't see a reason to have regular Debian Jessie anymore.

diederikdehaas, 2015-10-08:
It could indeed have been caused by that. 
But I would still like to have a go at this even if it's only to satisfy my own curiosity :wink:

herrernst, 2015-10-28:
I also like to run a real Debian on my Raspberry Pi 2. I've used the instructions from https://github.com/debian-pi/raspbian-ua-netinst/issues/246#issuecomment-126926687. It took a long time. I didn't delete the boot partition, so I didn't have to copy the installer back. Additionally, I also had to rename the installer kernel to kernel.img. Now I just have to figure out how to install new kernels and update Raspberry Pi firmware with apt.

SandPox, 2015-10-29:
@herrernst you'll need mount mmcblk0p1 as vfat in /boot and then install rpi-update and upgrade kernel normally

raidensnake, 2017-06-08:
I've discovered it's also possible to install with expert mode. This is how I managed to get it running.

when booting from the netinstaller files mentioned by ghost edit file cmdline.txt and add the following to it:
priority=low vga=788

This enables Expert mode upon boot.

I hope this proves useful to people.

Mausy5043, 2017-10-17:
I'm not expecting an implementation of this feature.
Closing the issue. Please feel free to re-open if you think the closure was in error.

closed 2017-10-17 by Mausy5043