debian-pi/raspbian-ua-netinst #250: Installation of f2fs fails.

Dennis-Janssen, 2015-06-10:
I'm trying to install on a Pi 2B. I've downloaded raspbian-ua-netinst-v1.0.7.img.bz2 and copied the image to the sdcard. Created a file installation-config.txt with these contents:

```
preset=server
packages= # comma separated list of extra packages
mirror=http://mirrordirector.raspbian.org/raspbian/
release=jessie
hostname=pi
domainname=
rootpw=xxxxxxxx
cdebootstrap_cmdline=
bootsize=+128M # /boot partition size in megabytes, provide it in the form '+<number>M' (without quotes)
rootsize=     # / partition size in megabytes, provide it in the form '+<number>M' (without quotes), leave empty to use all free space
timeserver=time.nist.gov
ip_addr=dhcp
ip_netmask=0.0.0.0
ip_broadcast=0.0.0.0
ip_gateway=0.0.0.0
ip_nameservers=
online_config= # URL to extra config that will be executed after installer-config.txt
usbroot= # set to 1 to install to first USB disk
cmdline="dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline"
rootfstype=f2fs
rootfs_mkfs_options=
rootfs_install_mount_options='noatime,data=writeback,nobarrier,noinit_itable'
rootfs_mount_options='errors=remount-ro,noatime'
```

Put the sdcard in my raspi and booted. After a while the red led comes on. Then I find the following logfile on the sd card:

```
================================================
raspbian-ua-netinst
=================================================
Revision git~f18c193
Built on Tue May  5 20:24:19 CEST 2015
Running on Raspberry Pi version 2
=================================================
https://github.com/debian-pi/raspbian-ua-netinst/
=================================================
Copying boot files... OK
=================================================
=== Start executing installer-config.txt. ===
=== Finished executing installer-config.txt. ===
=================================================

Network configuration:
  ip_addr = dhcp
  online_config = 

Waiting for eth0... Configuring eth0 with DHCP... 192.168.1.250
Time set using timeserver 'time.nist.gov' to... Wed Jun 10 16:54:14 UTC 2015

Installation started at Wed Jun 10 16:54:04 UTC 2015.

Installer configuration:
  preset = server
  packages = 
  mirror = http://mirrordirector.raspbian.org/raspbian/
  release = jessie
  hostname = pi
  domainname = 
  rootpw = xxxxxxxx
  cdebootstrap_cmdline = --flavour=minimal --include=kmod,fake-hwclock,ifupdown,net-tools,ntp,openssh-server,isc-dhcp-client,linux-image-rpi2-rpfv,vim-tiny,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,less,man-db
  bootsize = +128M
  rootsize = 
  timeserver = time.nist.gov
  cmdline = dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline
  usbroot = 
  rootdev = /dev/mmcblk0
  rootpartition = /dev/mmcblk0p2
  rootfstype = f2fs
  rootfs_mkfs_options = 
  rootfs_install_mount_options = noatime,data=writeback,nobarrier,noinit_itable
  rootfs_mount_options = errors=remount-ro,noatime

Waiting for /dev/mmcblk0... OK
Applying new partition table... OK
Initializing /boot as vfat... OK
Loading f2fs module... OK
Initializing / as f2fs... OK
Mounting new filesystems... mount: mounting /dev/mmcblk0p2 on /rootfs failed: Invalid argument

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
 ```

What am I doing wrong? I'm assuming that either the rootfs_install_mount_options, or the mkfs_options is wrong, but I have no idea of what they should be.
Any pointers would be greatly appreciated.

Zeerix, 2015-06-10:
Leave them empty. The script will automatically select the correct mount options (which is just `noatime`).

Dennis-Janssen, 2015-06-10:
Yes, that worked! Thanks a lot.

I copied the installer-config.txt from the installation instructions so it might be helpful to others to change that. Also the installation instructions say the installer-config.txt should be in the same folder as the Readme, but there is no readme file.

closed 2015-06-10 by Dennis-Janssen

Mausy5043, 2015-06-11:
You changed the `rootfstype=ext4` to `rootfstype=f2fs`.
The other `rootfs_*` defaults are specific to `ext4` and should have been changed also or left blank when you decided to change the filesystem. :wink:
Please be advised that `installer_config.txt` needs only to contain the parameters you wish to change.

There IS a `README.md`. Please look again.

Mausy5043, 2015-06-11:
@diederikdehaas : I think more emphasis is needed in the manual on the fact that people don't need to put ALL the defaults in the `installer-config.txt`, only the ones they expressly want to change. 
The `rootfs_*` parameters are built on the fly depending on the fs selected with `rootfstype` and the default values are given in the manual as if they should be part of the `installer-config.txt`. But IMHO they rarely need to be modified. This all clearly needs some clarification in the manual. No concrete ideas at this point in time at my end though.

diederikdehaas, 2015-06-11:
Yep, especially that part of the manual/README needs a serious rewrite.