debian-pi/raspbian-ua-netinst #251: Created vagrant machine

botto, 2015-06-11:
Maybe overkill but makes it easier to just get started
Could not get loopback devices to work correctly in docker so I used vagrant in the end
https://github.com/botto/raspbian-ua-netinst-vagrant

goranche, 2015-06-12:
is there a particular issue you were trying to solve by using vagrant/docker?
given that there are very few dependencies when building the image, I have a feeling that using any of these technologies might be overkill (just installing docker and/or vagrant is a bigger dependency than what the scripts need by themselves)

goranche, 2015-06-12:
not that it isn't a cool thing to do, though :angel:

botto, 2015-06-12:
Mainly because I wanted to eventually have a ci server run builds for me and I want to containerize the environment running root.

Mausy5043, 2015-06-12:
Sounds like a job for `post-install.txt` :wink:

goranche, 2015-06-13:
> have a ci server run builds

sounds cool :+1:
although I don't think this warrants inclusion in the installer itself

diederikdehaas, 2015-06-13:
I think it's cool what you've done and thanks for sharing it :+1: 
I hope to build a docker image for the installer, but just as a way to experiment with docker.

I agree with goranche that I don't see a reason to include it in the installer itself as it's likely to confuse ppl unfamiliar with vagrant (like me).
I would like to include a reference/small description in the (still to be created) wiki someday, but that'll probably take a while.

goranche, 2016-03-15:
I'm closing this, as we seem to agree upon this not being something that would be added into the installer... should things change in the future (or anyone disagrees with this decision), this can be reopened

closed 2016-03-15 by goranche