debian-pi/raspbian-ua-netinst #252: packages have been kept back

lars-daniel, 2015-06-14:
I've just install latest raspbian-ua-netinst on my RaspberryPi Modell B (not RPi 2).
When trying to apt-get upgrade I'm getting some packages, which are hold back:

```
root@pi:~# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages have been kept back:
  fake-hwclock libgcc1 libstdc++6
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

I don't have set up anything else than locales & tzdata. How can I help you to fix this bug?

diederikdehaas, 2015-06-14:
When you do `apt-get dist-upgrade` the packages will probably be upgraded.
My guess is that they're from the raspberrypi.org repository.

JedMeister, 2015-06-15:
I am 99.99% sure that this is not a bug and that It is normal, expected Debian behaviour. Packages will only be updated with `apt-get upgrade` if they are essentially updates to the same version. If the version bump is too high (i.e. actually a version change rather than the application of a security or bugfix patch) then apt will hold the packages back (because there is always a risk in things breaking).

I suspect that the risk of things breaking is low; but it still exists and as Debian (and Raspbian follows) errs on the side of caution you need to explicitly tell it to upgrade to new version packages. Try `apt-get dist-upgrade` like @diederikdehaas says! :smile:

lars-daniel, 2015-06-15:
"dist-upgrade" worked, thanks a lot. Also, gcc-4.8-base init-system-helpers get pulled now.
Perhaps you should run dist-upgrade at the end of raspbian-ua-netinst?

diederikdehaas, 2015-06-15:
> Perhaps you should run dist-upgrade at the end of raspbian-ua-netinst?

We'll take it under consideration, but I'm inclined to go with _no_.
I think a `dist-upgrade` should not be done unattended as it could cause undesired effects, and one should therefor evaluate the options and make a deliberate decision. 
On that same note I think the `-y` parameter in `apt-get upgrade -y` is unwise in day to day operations; in scripts you have to though.

If someone wants gcc-4.8-base from raspbian.org and not raspberrypi.org, (s)he should make that decision themselves.

diederikdehaas, 2015-10-02:
@lars-daniel If you want it you can do it in `post-install.txt`, but as I said before _I_ think it should be something done manually.

I discussed it with my co-maintainer and we both agree that this should _not_ be integrated in raspbian-ua-netinst, thus I'm closing this issue.

closed 2015-10-02 by diederikdehaas