debian-pi/raspbian-ua-netinst #253: Base preset bricks kernel installation

Saij, 2015-06-17:
Hi,

when you choose the base preset to install a Raspberry PI 2 you'll always get a RSOD (Rainbow Screen Of Death) because it'll not install the module ${kernel_meta_package}.
In line 304 of /scripts/etc/init.d/rcS the variable cdebootstrap_cmdline will be completly overwritten.

Using the server preset and everything will be fine.

Greetings
Saij

diederikdehaas, 2015-06-17:
Thanks for your report :+1: 

Note to self: also check other presets when testing :wink:

Saij, 2015-06-17:
Little Quick Fix:

in installer-config.txt just put following line
packages=${kernel_meta_package}

closed 2015-09-27 by diederikdehaas

diederikdehaas, 2015-12-01:
The fix is included with v1.0.8 of the installer, so it should now just work.

PR #281 (2015-09-27): Fix presets
This PR re-implements the presets configuration so that a kernel is always installed as well as `cpufrequtils` to fully utilize the CPU.

This fixes #253, #277, #279 and #271.
commit 59af50b6 (2015-09-27): Re-implemented the presets so the kernel package is always installed.

This fixes #253, fixes #277 and fixes #279.