debian-pi/raspbian-ua-netinst #267: Raspi Firmware - RPI-Update - Possible?

Zuendapp, 2015-06-30:
Hello,

is there any need to install the github software rpi-update?
Does is work?

Thank you very much!
Best regards
Stefan

diederikdehaas, 2015-06-30:
> is there any need to install the github software rpi-update?

No. The current advice is to actually **NOT** use it.
(unless specifically asked to do so and/or you want to write code for the firmware and/or kernel)

If you really do want it, it's available in the raspberrypi.org repository.

Zuendapp, 2015-06-30:
Hi,

ok thank you!
How can I install the newest firmware from raspbberypi.org repository?

Thank you very much!

Stefan

diederikdehaas, 2015-06-30:
Install the `raspberrypi-bootloader` package. This package contains both the raspberrypi.org kernel and firmware and will uninstall some other packages.
After that comment out or delete the `kernel` and `initramfs` lines from `/boot/config.txt` and reboot

closed 2015-06-30 by diederikdehaas

jedie, 2015-08-04:
> Install the raspberrypi-bootloader package. This package contains both the raspberrypi.org kernel and firmware and will uninstall some other packages.
After that comment out or delete the kernel and initramfs lines from /boot/config.txt and reboot

Please add this to README!