debian-pi/raspbian-ua-netinst #269: Custom install: static IP

Marbaf, 2015-07-07:
Hi,

First sorry because following is a question and not an identified issue. Feel free to remove if inappropriate.

I tried to install master branch with static IP declaration inside installer-config.txt on PI B rev2.
Each time it failed whereas installation works fine with default dhcp settings.
Could the reason be that I let parameter ip_nameservers empty ?

Kind regards

goranche, 2015-07-07:
> Could the reason be that I let parameter ip_nameservers empty

most probably :wink: 
if you don't specify any name server (DNS server), the installer will not be able to resolve the addresses of the repositories it needs for the packages it needs to install

diederikdehaas, 2015-07-07:
> Could the reason be that I let parameter ip_nameservers empty ?

That is VERY likely. If you don't use dhcp, you need to specify all things needed to make a network connection manually.

Marbaf, 2015-07-07:
Thank you guys.
Indeed when I change an already running pi from DHCP to static IP, system already knows about nameservers.
Apologize for inconvenience.

closed 2015-07-07 by Marbaf