debian-pi/raspbian-ua-netinst #270: Failed to change mirror in installer-config.txt

yangxuan8282, 2015-07-12:
Hi,I am trying to do a fresh install on my RPI2 use **raspbian-ua-netinst-v1.0.7**
but since there is just very limited Internet bandwidth between my country(China) with other country
so if I use the default mirror,the installation will always failed 
so I changed the mirror in *installer-config.txt*

    mirror=http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian

and just get some errors 

    === Start executing installer-config.txt ===
    /etc/init.d/rcS: /bootfs/installer-config.txt: line 1: mirror=http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian: not found
    === Finished executing installer-config.txt ===

I have tried many other mirror from China in this page :http://www.raspbian.org/RaspbianMirrors
but get same errors
Any idea?

goranche, 2015-07-12:
I don't think you should be using raspbian mirrors, but the debian mirrors instead.

try using [http://ftp.cn.debian.org/debian/](http://ftp.cn.debian.org/debian/) for the mirror instead

yangxuan8282, 2015-07-12:
@goranche ok,I'll try
one more question,can I just add this line in *installer-config.txt*
   
    rootfstype=f2fs

to directly install system in f2fs,instead of migrating to f2fs after installation?

goranche, 2015-07-12:
yes, you can... take a look at #250, just to be safe

yangxuan8282, 2015-07-13:
@goranche ,thanks for reply,I just have tried it,but still failed
I guess maybe the format is incorrect,or the installer can't read the *installer-config.txt* in right way? 

here is my *installer-config.txt*
     
    mirror=http://ftp.cn.debian.org/debian/

same error:

    === Start executing installer-config.txt ===
    /etc/init.d/rcS: /bootfs/installer-config.txt: line 1:  mirror=http://ftp.cn.debian.org/debian/: not found
    === Finished executing installer-config.txt ===

diederikdehaas, 2015-07-13:
> you shouldn't be using raspbian mirrors, but a debian mirror instead

Huh, why?

goranche, 2015-07-13:
> Huh, why?

hm... did I mess up? :thought_balloon:

diederikdehaas, 2015-07-13:
I think you did. 
Installing Debian packages on a RPi2 should be possible (with some caveats), that's not what this issue is about.

@yangxuan8282 Do you have file named raspbian-ua-netinst-<datetimestamp>.log in `/boot/` and if so, could you share those?

yangxuan8282, 2015-07-13:
can't uploaded file,for it says *Unfortunately,don't support that file type*
so paste below:

    
    =================================================
    raspbian-ua-netinst
    =================================================
    Revision git~f18c193
    Built on Tue May  5 20:24:19 CEST 2015
    Running on Raspberry Pi version 2
    =================================================
    https://github.com/debian-pi/raspbian-ua-netinst/
    =================================================
    Copying boot files... OK
    =================================================
    === Start executing installer-config.txt. ===
    /etc/init.d/rcS: /bootfs/installer-config.txt: line 1: ﻿mirror=http://ftp.cn.debian.org/debian/: not found
    === Finished executing installer-config.txt. ===
    =================================================
    
    Network configuration:
      ip_addr = dhcp
      online_config = 
    
    Waiting for eth0... Configuring eth0 with DHCP... 192.168.99.171
    Time set using timeserver 'time.nist.gov' to... Mon Jul 13 02:17:40 UTC 2015
    
    Installation started at Mon Jul 13 02:17:30 UTC 2015.
    
    
    Installer configuration:
      preset = server
      packages = 
      mirror = http://mirrordirector.raspbian.org/raspbian/
      release = wheezy
      hostname = pi
      domainname = 
      rootpw = raspbian
      cdebootstrap_cmdline = --flavour=minimal --include=kmod,fake-hwclock,ifupdown,net-tools,ntp,openssh-server,isc-dhcp-client,linux-image-rpi2-rpfv,vim-tiny,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,less,man-db
      bootsize = +128M
      rootsize = 
      timeserver = time.nist.gov
      cmdline = dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline
      usbroot = 
      rootdev = /dev/mmcblk0
      rootpartition = /dev/mmcblk0p2
      rootfstype = f2fs
      rootfs_mkfs_options = 
      rootfs_install_mount_options = noatime
      rootfs_mount_options = noatime
    
    Waiting for /dev/mmcblk0... OK
    Applying new partition table... OK
    Initializing /boot as vfat... OK
    Loading f2fs module... OK
    Initializing / as f2fs... OK
    Mounting new filesystems... OK
    Copying /boot files in... OK
    Starting install process...
    P: Retrieving Release
    P: Retrieving Release.gpg
    P: Validating Release
    I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
    P: Parsing Release
    P: Retrieving Packages.gz
    P: Validating Packages.gz
    P: Parsing Packages
    P: Retrieving libc-bin
    P: Validating libc-bin
    P: Retrieving gcc-4.7-base
    P: Validating gcc-4.7-base
    P: Retrieving libgcc1
    P: Validating libgcc1
    P: Retrieving libc6
    P: Validating libc6
    P: Retrieving multiarch-support
    P: Validating multiarch-support
    P: Retrieving libbz2-1.0
    P: Validating libbz2-1.0
    P: Retrieving libstdc++6
    P: Validating libstdc++6
    P: Retrieving zlib1g
    P: Validating zlib1g
    P: Retrieving libapt-pkg4.12
    P: Validating libapt-pkg4.12
    P: Retrieving liblzma5
    P: Validating liblzma5
    P: Retrieving libselinux1
    P: Validating libselinux1
    P: Retrieving tar
    P: Validating tar
    P: Retrieving dpkg
    P: Validating dpkg
    P: Retrieving readline-common
    P: Validating readline-common
    P: Retrieving libtinfo5
    P: Validating libtinfo5
    P: Retrieving libreadline6
    P: Validating libreadline6
    P: Retrieving libusb-0.1-4
    P: Validating libusb-0.1-4
    P: Retrieving gpgv
    P: Validating gpgv
    P: Retrieving gnupg
    P: Validating gnupg
    P: Retrieving raspbian-archive-keyring
    P: Validating raspbian-archive-keyring
    P: Retrieving apt
    P: Validating apt
    P: Retrieving mawk
    P: Validating mawk
    P: Retrieving base-files
    P: Validating base-files
    P: Retrieving base-passwd
    P: Validating base-passwd
    P: Retrieving sensible-utils
    P: Validating sensible-utils
    P: Retrieving debianutils
    P: Validating debianutils
    P: Retrieving dash
    P: Validating dash
    P: Retrieving bash
    P: Validating bash
    E: Couldn't download pool/main/b/bash/bash_4.2+dfsg-0.1+deb7u3_armhf.deb!
    
    Oh noes, something went wrong!
    You have 10 seconds to hit ENTER to get a shell...

yangxuan8282, 2015-07-13:
@diederikdehaas so,do u mean that both raspbian mirror and debian mirror will be work for installer?
and i post the log in /boot/ on last reply

diederikdehaas, 2015-07-13:
Do you have a line-feed ('Enter') at the end of your `installer-config.txt`?

Could you try a different **raspbian** mirror, from http://raspbian.org/RaspbianMirrors ?
So, don't use a debian mirror as goranche suggested and also not the one you listed in your first post, since it is "Unreachable as of 15-may-2015" according to the RaspbianMirrors page.
Does it fail in the same place or at some different one?

diederikdehaas, 2015-07-13:
> do u mean that both raspbian mirror and debian mirror will be work for installer?

No. At least not OOTB.
The RPi 2 does support armhf as used by Debian and as such the Debian packages could be used. But you would need (at the very minimum) the kernel from raspbian.org/raspberrypi.org.
So while using mostly Debian packages should _theoretically_ be possible, it is neither tested nor supported.

diederikdehaas, 2015-07-13:
@yangxuan8282 Do you have another Debian (-based) system in your Local Network?
If that's the case, you could consider installing `apt-cacher-ng` on it and adjust `installer-config.txt` mirror to something like this:  
`mirror=http://<machine-with-apt-cacher-ng>:3142/mirrordirector.raspbian.org/raspbian/`

yangxuan8282, 2015-07-13:
@diederikdehaas just have checked with the *installer-config.txt*,no line-feed 
and I have already tried every mirror from China in  http://raspbian.org/RaspbianMirrors
but without luck,in most case just failed to change mirror,with same error,something like:

    /etc/init.d/rcS: /bootfs/installer-config.txt: line 1: ﻿mirror=http://mirrors.neusoft.edu.cn/raspbian/raspbian/: not found

diederikdehaas, 2015-07-13:
Try **with** line-feed and see whether that makes a difference.

What I gathered from the log you posted, it failed to set the mirror and therefor used the default one, every single time.

yangxuan8282, 2015-07-13:
@diederikdehaas it works! thx a lot
strange,but it seems like an **Enter** is a MUST
otherwise,the installer won't recognize the new mirror

and since this issues have been solved,you can close it now
I will wait the installation complete,and report the result here

goranche, 2015-07-13:
sorry for the confusion on my part :angel: 

and regarding the enter, you should **always** put a newline at the end of any text file (at least on *nix systems)

yangxuan8282, 2015-07-13:
@diederikdehaas installation finished in 48 minutes,and after reboot it ask login now
can I connect it via ssh now?I don't have a keyboard for pi

@goranche I see,thx for friendly reminding,I am still newbies

diederikdehaas, 2015-07-13:
@yangxuan8282 Yes you can connect through ssh (`root`/`raspbian`)

yangxuan8282, 2015-07-13:
finally changed the mirror in *installer-config-txt* with

    mirror=http://mirrors.aliyun.com/raspbian/raspbian/

and add a newline at the end of it
this mirror is much faster for user in China than

    http://mirrordirector.raspbian.org/raspbian/

what's more,it's up to date and stable
the connection with mirror in low speed or unstable(in my home,the official mirror speed is 20-40k per second,the aliyun mirror is 300-700k per second) will cause several issues
in this case is 

    E: Couldn't download pool/main/b/bash/bash_4.2+dfsg-0.1+deb7u3_armhf.deb!

and in other case,such as do a **sudo apt-get update** in **raspbian**,will result in **hash sum mismatch** or **size mismatch**,however this kind of hint won't tell user the problem is caused by mirror connection(especially for newbies)

hope that raspbian could put this aliyun mirror in http://raspbian.org/RaspbianMirrors for user in China,make it easy for people to find a more suitable mirror,maybe the speed is differ from one city to another,but I have also saw some other blogs report it's better than other 

post this in case of somebody face same problem

closed 2015-07-13 by yangxuan8282

diederikdehaas, 2015-07-13:
> finally changed the mirror in installer-config-txt with 
mirror=http://mirrors.aliyun.com/raspbian/raspbian/
and add a newline at the end of it
this mirror is much faster for user in China than
http://mirrordirector.raspbian.org/raspbian/
what's more,it's up to date and stable

The best way for that to happen is when the people behind that mirror contact raspbian as explained on http://raspbian.org/RaspbianMirrors

yangxuan8282, 2015-07-14:
yeah,I guess so