debian-pi/raspbian-ua-netinst #272: Pi1: release=jessie ->  No space left on device

jedie, 2015-08-04:
I used v1.0.7 git~f18c193 and changed only a few things in my own **installer‑config.txt**:
```
release=jessie
packages=mc,htop,sudo,raspberrypi-bootloader
hostname=raspberrypi
```
This results in:
```
...
P: Unpacking package vim-common
P: Unpacking package vim-tiny
P: Unpacking package wget
dpkg: error processing /var/cache/bootstrap/less_444-4_armhf.deb (--unpack):
 cannot copy extracted data for './bin/less' to '/bin/less.dpkg-new': failed to write (No space left on device)
dpkg: error processing /var/cache/bootstrap/locales_2.13-38+rpi2+deb7u8_all.deb (--unpack):
 cannot copy extracted data for './usr/share/bug/locales/presubj' to '/usr/share/bug/locales/presubj.dpkg-new': failed to write (No space left on device)
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
tar: 
./preinst: Wrote only 3584 of 10240 bytes

tar: 
./postinst: Cannot write
: No space left on device
...
```

Another boring thing: The install failed and reboot and restart the install... it fail, reboot and restart... endless loop :(

diederikdehaas, 2015-08-04:
Can you post your whole `installer-config.txt`?

jedie, 2015-08-04:
That was the whole config ;) Or do you mean the log files?

diederikdehaas, 2015-08-04:
In that case, yes, please post the log files.

jedie, 2015-08-04:
It is only a 512MB card, too small?

Complete log: https://gist.github.com/jedie/119d9fc17dfcd4f64396

diederikdehaas, 2015-08-04:
Very likely. If you have an USB stick/drive to put `/` on (with `usbroot=1`) then it should work.

jedie, 2015-08-04:
I will try the same config with a 4GB SD...

Maybe after the "change /boot to 128MB" the README is outdated and should e.g. say min. 1Gb SD card?

diederikdehaas, 2015-08-04:
It has been updated and has now:
> - can fit on 512MB SD card, but 1GB is more reasonable
- SD card of at least 640MB or at least 128MB for USB root install (without customization)

It's probably not 'true' 100% of the time, but otherwise I'd have to go into so much detail it would become un-readable.
My guess is that you could make it work on a 512MB card, but you'd have to do some (serious) customizing.

jedie, 2015-08-04:
4GB SD install is done and worked!
```
root@raspberrypi:~# uname -a
Linux raspberrypi 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv6l GNU/Linux
root@raspberrypi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  3.5G  383M  3.0G  12% /
udev             10M     0   10M   0% /dev
tmpfs            37M  4.2M   33M  12% /run
tmpfs            92M     0   92M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            92M     0   92M   0% /sys/fs/cgroup
/dev/mmcblk0p1  122M   41M   82M  34% /boot
```

So for install a >512MB card is needed, but in the end is can fit into 512MB...

closed 2015-08-04 by jedie

diederikdehaas, 2015-08-04:
Good to know, thanks for reporting back :+1: