debian-pi/raspbian-ua-netinst #273: Packages added to installer-config.txt not avilable?

starksm64, 2015-08-05:
I'm trying to add some additional packages to the netinstall image, and I have tried using the installer-config.txt packages setting, as well as just updating the packages list in the update.sh script. 

When I use an installer-config.txt like:
```shell
starksm@debian-ua-netinst:~/Dev/raspbian-ua-netinst$ cat installer-config.txt 
packages=bluez,gcc-4.8,g++-4.8,build-essential,oracle-java8-jdk

P: Parsing Packages
W: Can't find package gcc-4.8 for inclusion
W: Can't find package g++-4.8 for inclusion
W: Can't find package oracle-java8-jdk for inclusion
P: Retrieving libc-bin
...
```

When I log into the system, I am able to apt-get these packages just fine:
```shell
root@pi:~# apt-get install gcc-4.8 g++-4.8 oracle-java8-jdk
Reading package lists... Done
Building dependency tree      
Reading state information... Done
The following extra packages will be installed:
  binutils cpp-4.8 fontconfig fontconfig-config gcc-4.8-base gir1.2-glib-2.0
  hicolor-icon-theme libasan0 libasound2 libatk1.0-0 libatk1.0-data libatomic1
  libavahi-client3 libavahi-common-data libavahi-common3 libcairo2
  libcloog-isl4 libcups2 libdatrie1 libfontconfig1 libfreetype6 libgcc-4.8-dev
  libgcc1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgirepository-1.0-1
  libglib2.0-0 libgomp1 libgraphite2-2.0.0 libgtk2.0-0 libgtk2.0-bin
  libgtk2.0-common libharfbuzz0a libisl10 libjasper1 libjbig0 libjpeg8 libmpc3
  libpango-1.0-0 libpango1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0
  libpangox-1.0-0 libpangoxft-1.0-0 libpixman-1-0 libpng12-0 libstdc++-4.8-dev
  libstdc++6 libthai-data libthai0 libtiff4 libx11-6 libx11-data libxau6
  libxcb-render0 libxcb-shm0 libxcb1 libxcomposite1 libxcursor1 libxdamage1
  libxdmcp6 libxext6 libxfixes3 libxft2 libxi6 libxinerama1 libxml2 libxrandr2
  libxrender1 libxtst6 python-gi sgml-base shared-mime-info ttf-dejavu-core
  ucf x11-common xml-core
Suggested packages:
  binutils-doc gcc-4.8-locales gcc-4.8-doc libstdc++6-4.8-dbg libgcc1-dbg
  libgomp1-dbg libitm1-dbg libatomic1-dbg libasan0-dbg libtsan0-dbg
  libbacktrace1-dbg libquadmath-dbg libasound2-plugins cups-common
  librsvg2-common gvfs libjasper-runtime ttf-baekmuk ttf-arphic-gbsn00lp
  ttf-arphic-bsmi00lp ttf-arphic-gkai00mp ttf-arphic-bkai00mp
  libstdc++-4.8-doc python-gi-cairo sgml-base-doc debhelper
Recommended packages:
  libglib2.0-data
The following NEW packages will be installed:
  cpp-4.8 fontconfig fontconfig-config g++-4.8 gcc-4.8 gcc-4.8-base
  hicolor-icon-theme libasan0 libasound2 libatk1.0-0 libatk1.0-data libatomic1
  libavahi-client3 libavahi-common-data libavahi-common3 libcairo2
  libcloog-isl4 libcups2 libdatrie1 libfontconfig1 libfreetype6 libgcc-4.8-dev
  libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgraphite2-2.0.0 libgtk2.0-0
  libgtk2.0-bin libgtk2.0-common libharfbuzz0a libisl10 libjasper1 libjbig0
  libjpeg8 libmpc3 libpango-1.0-0 libpango1.0-0 libpangocairo-1.0-0
  libpangoft2-1.0-0 libpangox-1.0-0 libpangoxft-1.0-0 libpixman-1-0 libpng12-0
  libstdc++-4.8-dev libthai-data libthai0 libtiff4 libx11-6 libx11-data
  libxau6 libxcb-render0 libxcb-shm0 libxcb1 libxcomposite1 libxcursor1
  libxdamage1 libxdmcp6 libxext6 libxfixes3 libxft2 libxi6 libxinerama1
  libxml2 libxrandr2 libxrender1 libxtst6 oracle-java8-jdk sgml-base
  shared-mime-info ttf-dejavu-core ucf x11-common xml-core
The following packages will be upgraded:
  binutils gir1.2-glib-2.0 libgcc1 libgirepository-1.0-1 libglib2.0-0 libgomp1
  libstdc++6 python-gi
8 upgraded, 72 newly installed, 0 to remove and 3 not upgraded.
Need to get 134 MB of archives.
After this operation, 305 MB of additional disk space will be used.
Do you want to continue [Y/n]? y
Get:1 http://archive.raspberrypi.org/debian/ wheezy/main gcc-4.8-base armhf 4.8.2-21~rpi3rpi1 [154 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libxml2 armhf 2.8.0+dfsg1-7+wheezy4 [824 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libasound2 armhf 1.0.25-4 [410 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libavahi-common-data armhf 0.6.31-2 [137 kB]
Get:5 http://archive.raspberrypi.org/debian/ wheezy/main libgcc1 armhf 1:4.8.2-21~rpi3rpi1 [54.2 kB]
Get:6 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libavahi-common3 armhf 0.6.31-2 [52.6 kB]
Get:7 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libavahi-client3 armhf 0.6.31-2 [55.4 kB]
...
```

Is there a way to make these packages visible during the netinstall?

diederikdehaas, 2015-08-05:
It looks like you're using the installer incorrectly and some things are not (yet) possible, at least not OOTB.
`update.sh` is used to build the installer itself and has no effect on the installation being done on the RPi. Unless you want to contribute to the installer itself, it should not be modified.
What you should do, is write a released installer to the sd card, and then add your `installer-config.txt` to the sd card before you put it in the RPi and boot up. This is all described in the README.md and if things there are not clear, let us know how we can improve it and/or submit your improvements through a pull request.

As for the packages you've listed in your `installer-config.txt`, some can be installed and some can not. The criteria is whether it's available in Debian/raspbian main archive.
`bluez` and `build-essential` can be installed
`gcc-4.8` and `g++-4.8` can be installed, but only when `release=jessie`
`oracle-java8-jdk` can not be installed since it's not available in the Debian/raspbian archive ... at all.
https://www.debian.org/distrib/packages#search_packages should help you find whether a certain package is available in debian/raspbian.

That you are able to install `gcc-4.8`, `g++-4.8` and `oracle-java8-jdk` after the installation completed is because they are part of the raspberrypi.org archives and those can not be using during the installation OOTB. This may change in the future though. 
A 'workaround' is installing those packages through `post-install.txt`, like this:
```
chroot /rootfs /usr/bin/apt-get -y install gcc-4.8 g++-4.8 oracle-java8-jdk &>/dev/null
```

RichardX720, 2015-08-12:
Question, running chroot /rootfs /usr/bin/wget url... that would still be the busybox wget? I cannot seem to get wget to work with a username/password on busybox. I read in some other posts that people have had it work but that fix eludes me! 

Thanks

diederikdehaas, 2015-08-12:
Probably, unless you install the wget package first :wink:

RichardX720, 2015-08-12:
perfect point :) will give it a try!

RichardX720, 2015-08-13:
Another question, I've been reading up on chroot, using it is new to me.. I cant seem to grasp one thing. If I run an apt-get install of, for example, MySQL can I also pass commands to configure MySQL in the same manner? or is use of apt-get simply putting things in place for the reboot?

RichardX720, 2015-08-13:
worked like a charm!

diederikdehaas, 2015-10-02:
@starksm64 What's the status on this issue?

diederikdehaas, 2015-10-08:
No response, thus closing.

closed 2015-10-08 by diederikdehaas