debian-pi/raspbian-ua-netinst #274: Feature request: Way to specify /etc/rc.local script in post install image

starksm64, 2015-08-05:
Would it be possible to have the build.sh script include an rc.local script if seen (similar to how post-install.txt is tested for) incorporated into the post net install image?

What I would like is to have an ability to run a script that runs to post configure the rebooted raspbian image. I'm still looking into how to do this, but perhaps it is easy or not viable. Looking for feeback.

Mausy5043, 2015-08-05:
I use the `post-install.txt` to install a script that runs after the re-boot using a cron job with the `@reboot` directive. Is that what you're looking for?

diederikdehaas, 2015-08-05:
> I use the post-install.txt to install a script that runs after the re-boot

That's indeed how `post-install.txt` can and should be used :+1: 
(besides doing things before the reboot)

RichardX720, 2015-08-12:
Great Hub guys! Has been very helpful. would you be so kind to give me an example of how you use @reboot in cron to run a script after install? Is it possible to only have that run once? Thanks

bmedicke, 2015-08-13:
_postinstall.txt_:

    chroot /rootfs /bin/cp /boot/cronjob-mail /etc/cron.d/mail-py

_cronjob-mail_:

    @reboot root /boot/mail.py

When setting up my SD card I put the `cronjob-mail` file in the `/boot` directory. You could just `echo` that line directly. To have it run only once your script could delete its cronjob after it's done.

RichardX720, 2015-08-13:
This may be slightly off topic but a little insight will help me alot!

If I run from post-install: 

chroot /rootfs ./myscript.sh 

the script runs and installs a bunch of stuff including sudo (/etc/sudoer).. However I have this script also adding a user which it does just fine BUT..

right after the new user is created it runs:
echo 'thenewuser ALL=(ALL:ALL) ALL' >> /etc/sudoers

It errors here and stops the script that it cannot find this yet it is there after the install of the package.. Am I incorrectly using the path?

Thanks

Richard

diederikdehaas, 2015-08-13:
I don't know much about sudo, but I think it's a permission issue:
```
$ ls -l /etc/sudoers
-r--r----- 1 root root 669 Feb 23 17:05 /etc/sudoers
```
As you can see, there is now 'w' in there, thus you aren't allowed to write into it.
What happens if you change
`echo 'thenewuser ALL=(ALL:ALL) ALL' >> /etc/sudoers`
to
`echo 'thenewuser ALL=(ALL:ALL) ALL' >> /etc/sudoers.d/thenewuser.sudo.conf`
?
See `/etc/sudoers.d/README` for details

goranche, 2015-08-13:
you shouldn't be writing to the sudoers file anyway... create a file in ```/etc/sudoers.d``` (as @diederikdehaas suggested)

goranche, 2015-08-13:
as an example, this is part of one of my ```post-install.txt``` scripts:
```
for command in /sbin/reboot /sbin/halt; do
	echo "zurkon ALL=(ALL) NOPASSWD: ${command}" >> /rootfs/etc/sudoers.d/zurkon
done
chmod 0440 /rootfs/etc/sudoers.d/zurkon
```
(zurkon is a chat bot I've written, which runs on the raspberry and is obviously able to restart and shut down the raspberry... it runs as user zurkon)

goranche, 2015-08-13:
@RichardX720 also, why do you need to add that to ```/etc/sudoers```? check the content of the existing sudoers file, I believe you will find a line line this:
```
%sudo	ALL=(ALL:ALL) ALL
```

this means that any system user, belonging to the sudo group, will be able to run commands using ```sudo```... so you should add the user to the sudo group when you create it (and leave ```/etc/sudoers``` as is)... something like this (from the same zurkon script as above):
```
chroot /rootfs useradd -s /bin/bash -p "sa5snuFEEEqVg" -m -G sudo goranb
```

you can leave out the ```-p "..."``` part to not set a default password for the user, if you prefer...
(you can generate your own by using this: ```perl -e 'print crypt("somepassword", "salt-value"),"\n"'```, there are of course other ways as well)

diederikdehaas, 2015-10-02:
@starksm64 And what's the status on this issue?

diederikdehaas, 2015-10-08:
No response, thus closing

closed 2015-10-08 by diederikdehaas