debian-pi/raspbian-ua-netinst #275: couldn't download pool/main/ncurses libtinfo 5_5

Bobbaone, 2015-08-16:
At trying to install the img at the raspberry, a error appears:

couldn't download pool/main/ncurses/libtinfo5_5.9-10_armhf.deb

wget ERROR 404

goranche, 2015-08-17:
is this happening every time? say today? :smile: 

I think this might be a transient issue at the packages repository, over which we don't have any control

Bobbaone, 2015-08-17:
Hm, i dont use it anymore, i use dietpi...

goranche, 2015-08-17:
ok, in that case I guess I can close this issue...

if you feel it's still needed, you can of course reopen it

closed 2015-08-17 by goranche

diederikdehaas, 2015-08-17:
> couldn't download pool/main/ncurses/libtinfo5_5.9-10_armhf.deb

JFTR: that path is incorrect, it should've been `pool/main/n/ncurses/libtinfo5_5.9-10_armhf.deb`