debian-pi/raspbian-ua-netinst #276: wrong apt repository after Jessie install

oupala, 2015-09-03:
After installing Jessie on a raspberry 1, I wanted to install apache2.

Running `apt-cache show apache2`, I saw that apache 2.2 was about to be installed while Jessie is now offering apache 2.4. I was wondering why I would get stuck to apache 2.2 and I had a look at `/etc/apt/sources.list`.

I was surprised to discover that the second apt repository was mentionning **Wheezy** release, not Jessie. The first apt repository was set on Jessie.

Any clue about that ?

JustBlackBird, 2015-09-03:
Wheezy release was hardcoded in 3e5323a3726f1b94a75499de139bd65562f2c8ea. Are you sure you've installed Jessie? How did you do that?

oupala, 2015-09-03:
Thanks @JustBlackBird for this fast answer. It does at least explain why I had such entry in my sources file.

I installed jessie using the [customization](https://github.com/debian-pi/raspbian-ua-netinst#installer-customization) capability of the installer. I created an `installer‑config.txt` file where I set `release=jessie`.

The raspberrypi archives [now support wheezy **and jessie**](http://archive.raspberrypi.org/debian/dists/): I think it is time to revert 3e5323a3726f1b94a75499de139bd65562f2c8ea.

JustBlackBird, 2015-09-03:
As I see, `release=jessie` option is ignored for the latest version of the installer and `wheezy` is used instead of it.

As for reverting, there was a discussion about that in #233 some time ago.

diederikdehaas, 2015-09-03:
`release=jessie` is only ignored for the raspberrypi.org repositories.
Unfortunately their repositories are still not ready for `jessie`, even today.

JustBlackBird, 2015-09-03:
@diederikdehaas, could you point another repository that can be used with the installer? And how one can use them?

diederikdehaas, 2015-09-03:
I don't understand what you mean.
There are 2 repositories configured with the installer. By far the most packages come from the _raspbian.org_ repositories and for the `raspbian.org` repositories the `release=jessie` setting **does** work.
It would be very strange if you would get apache 2.2 while on jessie.
Could you post the output of `apt-cache policy apache2`?

oupala, 2015-09-03:
To be clear, here is my apt sources.list after installing raspbian with an `installer‑config.txt` file where I set `release=jessie`.

```bash
>more /etc/apt/sources.list
deb http://mirrordirector.raspbian.org/raspbian jessie main firmware
deb http://archive.raspberrypi.org/debian wheezy main
```

oupala, 2015-09-03:
@diederikdehaas What do you mean by:

> Unfortunately their repositories are still not ready for `jessie`, even today.

Here is the result of the apt policy command:
̀
```bash
>apt-cache policy apache2
apache2:
  Installé : 2.4.10-10+deb8u1
  Candidat : 2.4.10-10+deb8u1
 Table de version :
 *** 2.4.10-10+deb8u1 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
̀```

apache 2.2 is not present.

I had maybe dreamt yesterday. I'll try to reinstall from scratch to be sure...

But anyway, this leads to the point: is it really usefull to leave the raspberrypi.org line in apt source file? If they do not support jessie, we should remove the line when installing jessie! Is it usefull to keep a line that refers to another major release of debian?

diederikdehaas, 2015-09-03:
> this leads to the point: is it really usefull to leave the raspberrypi.org line in apt source file?

I think so as they provide packages which are useful/relevant for the RPi. It's unfortunate that they still haven't updated it all to jessie, but I'm not aware that their packages conflict with package in raspbian.org jessie

oupala, 2015-09-04:
Do you know why are these packages not merged into the raspbian repo?

Anyway, please leave this issue open as I will test from scratch in the next days. I'll close the issue if I can't reproduce the strange behaviour I had with apache2.

diederikdehaas, 2015-09-04:
I haven't tried it myself, but I'm quite sure that `release=scratch` won't work as `cdebootstrap` doesn't support it (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=783672)

Mausy5043, 2015-09-05:
>> I will test from scratch in the next days
 > `release=scratch` won't work

LOL :smile:

diederikdehaas, 2015-09-05:
LOL, `release=stretch` won't work either :stuck_out_tongue_winking_eye: ... 
but I now see that oupala didn't mention trying with the `stretch` release. Brainfart on my part.

oupala, 2015-09-05:
This was a funny word mistake by the way! ;-)

I tried again from a fresh install (*fresh* is less confusin than *scratch*) and I was wrong: with the default `sources.list`, raspbian proposes me to install apache 2.4, not 2.2.

I still don't understand why raspberrypi maintains a separate repo from raspbian, but everything seems to work well with the default **Jessie** install of **raspbian-ua-netinst**. Thanks!

Feel free to close the issue if you think everything is ok for you.

diederikdehaas, 2015-09-05:
> Feel free to close the issue if you think everything is ok for you.

Good, will do just that.
On the bright side, I heard that progress is being made on the raspberrypi.org jessie repo so there's a reasonable chance that commit https://github.com/debian-pi/raspbian-ua-netinst/commit/3e5323a3726f1b94a75499de139bd65562f2c8ea will be reverted for the next release of the installer (but I don't know when that will happen).

closed 2015-09-05 by diederikdehaas

diederikdehaas, 2015-09-10:
JFTR: the change has been reverted: 1dfa5f2b338e14df7d84b19bbcf278d0be1b48bc