debian-pi/raspbian-ua-netinst #277: Hangs when rebooting after a 'base' install

mike632t, 2015-09-10:
The default installation options work fine - however in order to really minimize the disk space and allow me to use a serial console used I created an 'installer-config.txt'  containing the following.  

```preset=base
hostname=pi-serv01
rootpw=*********
bootsize=+128M
rootsize=+1918M
cmdline="dwc_otg.lpm_enable=0 console=ttyAMA0,9600 kgdboc=ttyAMA0,9600 console=ttyAMA0 elevator=deadline"
```

The installation appears to work (on a normal Pi with HDMI display) but the system does not come back up when rebooted, there seems to be something missing from the boot partition.  After the installation has completed if I delete all the files in the boot partition apart from  'installer-config.txt' and 'cmdline.txt'.  Then copy in all the files from the boot partition created using the default options the PI them boots normally.

diederikdehaas, 2015-09-10:
```
cmdline="dwc_otg.lpm_enable=0 console=ttyAMA0,9600 kgdboc=ttyAMA0,9600 console=ttyAMA0 elevator=deadline"
```
It looks to me that that line is wrong. I'm pretty sure that your console part should be `console=ttyAMA0,115200` and your 2nd `console` is wrong (AFAIK a serial console should have a speed) and you should just delete it. .
If you want both a 'normal' console (`tty1`) and a serial console which shows all boot messages, do something like this:
`cmdline="dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200`

Do you have a `/var/log/raspbian-ua-netinst.log` file? If so, could you post it?

mike632t, 2015-09-10:
I think the bit you want is this 

```
Installing bootloader package (=firmware)... OK
Installing libraspberrypi-bin package (=vcgencmd)... OK
Configuring bootloader to start the installed system...
Kernel version: ''
Copying kernel to /boot/kernel.img... cp: can't stat '/rootfs/boot/vmlinuz-': No such file or directory
FAILED !
Creating default cmdline.txt... OK
Saving current time for fake-hwclock...
Installation finished at Thu Sep 10 16:46:35 UTC 2015 and took 8 min 18 sec (498 seconds)
Unmounting filesystems... 
```

I can post the rest if you want

diederikdehaas, 2015-09-10:
That is indeed the part I needed. Something went wrong with the installation of the kernel and then the system can indeed not boot. Will look into that.
Do you have a Pi 1 or Pi 2?

mike632t, 2015-09-10:
Removed duplicate ```console=ttyAMA0``` makes not difference to running system - will try with an install shortly

However the option to sepecify the baud rate in cmdline.txt (and inittab) would be great as not all serial interfaces support 115200 (I use 9K6 so the modified line speed is correct)

mike632t, 2015-09-10:
Pi B - can try on a P2 as well using the same settings

diederikdehaas, 2015-09-10:
> However the option to sepecify the baud rate in cmdline.txt (and inittab) would be great as not all serial interfaces support 115200 (I use 9K6 so the modified line speed is correct)

Ok, I'll keep that in mind for a next version (1.1.x branch)

diederikdehaas, 2015-09-11:
Can you also post the section that starts with `Installer configuration:` ?

commit 59af50b6 (2015-09-27): Re-implemented the presets so the kernel package is always installed.

This fixes #253, fixes #277 and fixes #279.
closed 2015-09-27 by diederikdehaas