debian-pi/raspbian-ua-netinst #279: Base install using jessie fails when allowing root to login using password

mike632t, 2015-09-11:
Last entries in raspbian-ua-netinst-<timestamp>.log
```
Configuring installed system:
  Setting root password... OK
  Configuring /etc/fstab... OK
  Configuring tmpfs to use RAM... OK
  Configuring hostname... OK
  Configuring network settings... OK
  Enabling serial console... OK
  Allowing root to login with password... 
```
Fail routine catches error (didn't see message on screen untill just too late) and then system reboots and install process starts again.

'installer-config.txt' contains 
```
release=jessie
preset=base
hostname=unknown
rootpw=W1nt3r04**
bootsize=+128M
rootsize=+1918M
cmdline="dwc_otg.lpm_enable=0 console=ttyAMA0,9600 kgdboc=ttyAMA0,9600 elevator=deadline"
```

diederikdehaas, 2015-09-11:
Sounds like https://github.com/debian-pi/raspbian-ua-netinst/issues/253

mike632t, 2015-09-11:
Thanks.

I have the installer running with serial console this time so won't miss anything (deleted ```console=tty1``` in cmdline.txt) will let you know if it works this time.

Couldn't find anywhere to just leave a coment - but this is an awsome installer!

diederikdehaas, 2015-09-11:
> Couldn't find anywhere to just leave a coment - but this is an awsome installer!

Thanks :+1:

mike632t, 2015-09-11:
Got a bit more from the console output this time - following should help.

```
  Allowing root to login with password... sed: /rootfs/etc/ssh/sshd_config: No such file or directory

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
swapoff: /etc/fstab: No such file or directory
The system is going down NOW!
```

diederikdehaas, 2015-09-11:
Yeah, it's essentially the same as #253. `preset=base` and likely `preset=minimal` are broken (because I 'never' test them).

mike632t, 2015-09-11:
No issues with server (as you expected) - assume that to use the build instrustions (https://github.com/debian-pi/raspbian-ua-netinst/blob/master/BUILD.md)  you need to use a Pi ?

diederikdehaas, 2015-09-11:
> assume that to use the build instrustions (https://github.com/debian-pi/raspbian-ua-netinst/blob/master/BUILD.md) you need to run the build script on a Pi ?

No. I have never done that. I build the installer on my home PC running Debian Sid on amd64.

commit 59af50b6 (2015-09-27): Re-implemented the presets so the kernel package is always installed.

This fixes #253, fixes #277 and fixes #279.
closed 2015-09-27 by diederikdehaas