debian-pi/raspbian-ua-netinst #280: Kernel stays at 3.18

devlg, 2015-09-20:
After installing jessie which worked fine, i installed rpi-update to update my kernel and firmware. It updated, says its up2date but still uses older 3.18 kernel:

root@beta:~# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date

root@beta:~# uname -a
Linux beta.xxx 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv6l GNU/Linux

Are there more steps to perform?

kpfleming, 2015-09-20:
Installing a new kernel package does not change the running kernel; to start the new kernel, you'll need to reboot.

devlg, 2015-09-20:
I know that :-) But there is no new kernel loaded up.

diederikdehaas, 2015-09-20:
Remove the `kernel` and `initramfs` lines from `/boot/config.txt` and reboot.

devlg, 2015-09-21:
Worked, thanks!

closed 2015-09-21 by diederikdehaas