debian-pi/raspbian-ua-netinst #282: Raspberry Pi display not supported

MichealMi, 2015-09-23:
Hi,
I'm actually a Linux noob, so I can't tell if this is a real "bug/issue". 

Since I discovered this installer I don't use any other image, so thanks for the great work! However, I can't get the Official Raspberry Pi 7" Display to work. Is this an issue or is this intended? do I need to install drivers or a special package?

Thanks

goranche, 2015-09-23:
hi, @MichealMi 

yes, there is an issue with the display, I got mine a few days ago, but haven't found the time to debug the issue and find a workaround... however, knowing that I'm not the only one with the issue will make me try harder :wink: 

expect additional info soon :+1:

MichealMi, 2015-09-23:
Thanks for the Info and thanks for contributing to the installer. It's a great little piece of software which made my life easier :)

goranche, 2015-09-23:
well, got it to work, a.k.a. to boot with the LCD on and console output on the LCD, didn't try the touchscreen yet...

nothing serious, the firmware needs to be updated:

```sudo apt-get install raspberrypi-bootloader```

@diederikdehaas says he'll file an issue with a request for an updated firmware, so this should be a temporary workaround :+1: 

if needed, I'll prepare a snippet that can be placed in post-install.txt, which will install the new firmware during the installation...

goranche, 2015-09-23:
come to think of it, just adding ```raspberrypi-bootloader``` to ```packages=``` in ```installer-config.txt``` should be enough :thought_balloon:

diederikdehaas, 2015-09-23:
raspbian bug: https://bugs.launchpad.net/raspbian/+bug/1499046

goranche, 2015-09-23:
ok, so adding the package to installer-config.txt doesn't work, but you can add the following to your ```post-install.txt``` file, and after install the LCD works just fine... (adjust the ```gpu_mem``` to your liking)

```
# We want some video memory
echo "gpu_mem=64" >> /rootfs/boot/config.txt

# Install the newest firmware so the LCD works again
chroot /rootfs apt-get -y install raspberrypi-bootloader
```

MichealMi, 2015-09-23:
Thanks for the hint :+1: Hope to see a real solution soon, because I can't see the install progress at the moment.

goranche, 2015-09-23:
> I can't see the install progress at the moment

yeah, I know, that's why I included the ```post-install.txt``` snippet (you can find some info on post-install.txt in the readme file)...

also, you can still use a HDMI cable to output video to a normal screen... AFAIK it should even be possible to use both outputs at the same time (with different content, of course)

diederikdehaas, 2015-10-14:
@LukasMa A test version of updated firmware files was provided to me and based on that I've created a new installer. Can you test whether that fixes the issue?

The installer files can be found at http://cknow.org/rpi/ and are named `raspbian-ua-netinst-20151014-rpi-display-test` + the usual extensions (`.img.bz2`/`.img.xz`/`.zip`).
When you've written the installer to the sd-card you also have to add the following to `post-install.txt` (create that file if it doesn't exists):
```
echo -n "Copying firmware deb files to system... "
cp /rootfs/boot/config/apt/raspberrypi-bootloader-nokernel_1.20150923-1~nokernel1_armhf.deb /rootfs/root/
cp /rootfs/boot/config/apt/libraspberrypi0_1.20150923-1~nokernel1_armhf.deb /rootfs/root/
cp /rootfs/boot/config/apt/libraspberrypi-bin_1.20150923-1~nokernel1_armhf.deb /rootfs/root/
echo "OK"

echo -n "Installing updated raspberrypi-bootloader-nokernel package... "
chroot /rootfs usr/bin/dpkg -i /root/raspberrypi-bootloader-nokernel_1.20150923-1~nokernel1_armhf.deb
echo "OK"

echo -n "Installing updated libraspberrypi0 package... "
chroot /rootfs usr/bin/dpkg -i /root/libraspberrypi0_1.20150923-1~nokernel1_armhf.deb
echo "OK"

echo -n "Installing updated libraspberrypi-bin package... "
chroot /rootfs usr/bin/dpkg -i /root/libraspberrypi-bin_1.20150923-1~nokernel1_armhf.deb
echo "OK"
```

LukasMa, 2015-10-14:
@diederikdehaas: Wow, thanks for you commitment :) I'm busy today and tomorrow. I will give you some feedback on friday. Thanks!

diederikdehaas, 2015-10-17:
@LukasMa friendly ping? :wink:

LukasMa, 2015-10-17:
@diederikdehaas: Yes, ist works :) Sorry for the late response. Thanks! Great! :)

diederikdehaas, 2015-10-17:
Thanks :+1:

mhusari, 2015-11-04:
When can we expect actual release that contains this fix?

diederikdehaas, 2015-11-04:
On Wednesday 04 November 2015 00:38:10 Mikko Husari wrote:
> When can we expect actual release that contains this fix?

It depends on when a new version of the firmware package is released and I 
don't know when that will be.

mhusari, 2015-11-04:
Regular rasbian already has "up to date" firmware package, can't we use that?

diederikdehaas, 2015-11-04:
No, not really. The image from raspberrypi.org has a package which combines the kernel and firmware and if we want to use that, we'd have to rewrite a (large?) part of the installer and I don't think it's worth it.

mhusari, 2015-11-04:
Ok. Understood, thanks for quick and clear reply. Now, for me all things seem to work nice except the touch. Has anyone else had any issues using the touch interface? Maybe I am just missing some configuration/modules?

mhusari, 2015-11-05:
Okay, maybe touch is not working because kernel lacks support for touchscreen. root@pi:~# grep -i touch /boot/config-3.18.0-trunk-rpi2
CONFIG_INPUT_TOUCHSCREEN is not set

also, should probably have CONFIG_TOUCHSCREEN_RPI_FT5406 as a module.

diederikdehaas, 2015-11-05:
Now I'm confused because I got the impression that only updating the firmware fixed the issue.

@LukasMa and @goranche can you comment on this?

diederikdehaas, 2015-11-05:
@mhusari where did you find the CONFIG_TOUCHSCREEN_RPI_FT5406 setting? Because `grep -i touch /boot/config-3.18.0-trunk-rpi2` did not return that one.

diederikdehaas, 2015-11-05:
I updated the [raspbian bug](https://bugs.launchpad.net/raspbian/+bug/1499046) with your findings @mhusari. Thanks for tracking that down :+1:

mhusari, 2015-11-06:
I just checked what modules current raspbian loaded, then tried to load those as well. When it failed I checked if kernel had the module at all. I am not sure if _only_ those additions fix the issue, but I am quite sure that those _at least_ are required with this display.

diederikdehaas, 2015-11-06:
Though so. Thanks for clarifying.

diederikdehaas, 2016-08-09:
Can someone test whether the RPi display works with the v1.1.x branch?
That branch already has a newer kernel and firmware which has the mentioned kernel config options enabled or build as a module.

pgrandin, 2017-05-19:
Tested with a RPI 2B, can confirm that it works too!

Mausy5043, 2017-05-19:
Display seems to work now. Closing

closed 2017-05-19 by Mausy5043

pgrandin, 2017-05-19:
Do you guys have a plan to release an image for the 1.1 branch? wifi and raspberry displays are really useful out of the box features.