debian-pi/raspbian-ua-netinst #283: Sandisk Ultra 64GB Micro SDXC (Class 10) doesnt work

devlg, 2015-09-25:
I tested it with 2 cards, both doesnt work with your image. Working flawless with the latest Raspian Image. What might cause this trouble? Missing FW?

diederikdehaas, 2015-09-25:
Could be or newer firmware.
What you could try is putting the files from https://github.com/raspberrypi/firmware/tree/master/boot onto the SD card and thereby updating the firmware files.

devlg, 2015-09-26:
Tested with both SD-Cards, doesnt change anything at all unfortunatly.

goranche, 2015-09-26:
what happens when you boot it up?

devlg, 2015-09-26:
Nothing, screen stays black. Not even the activity led blinks. Tested with 2 sd-cards (as writte above) and 2 raspberry pi's. Again, working flawless with raspbian.

goranche, 2015-09-26:
hm, weird... I can't seem to find any local supplier who'd have these cards in stock, and I'm guessing the price would be quite high... I also don't really have a need for 64GB SD cards :smile: 

@diederikdehaas could it be an partitioning issue due to the size? not completely sure myself how, though  :thought_balloon:

goranche, 2015-09-26:
huh... interesting comment on a forum with a list of cards that work

> _June 9, 2012_
> 16GB Sandisk Extreme 45MB/s SDCard works with Pi but only after applying the Debian update of kernel.img and start.elf to the current April 2012 Debian release. And it flies. Boot time from cold to login is 17 seconds.

goranche, 2015-09-26:
@devlg are you maybe overclocking the raspberry? did you change the config.txt file?

also, what kind of power supply are you using and what hardware do you have connected? power fluctuation might be an issue (although I'm not sure how to explain the working system image)...
still, might help

diederikdehaas, 2015-09-27:
My next guess would be that the kernel is the culprit. What happens if you take the kernel(s) from the raspberry pi image, place them next to the installer files and specify those kernel files in `config.txt`?

devlg, 2015-09-27:
Gonna test this. I can (if you want me to) send you one of these cards for free via postal service. Shouldnt be that much expensive.

goranche, 2015-09-27:
> also using the same for my other raspis which work fine

yeah, that's the part that I was wondering about... like @diederikdehaas mentioned, the kernel might be the problem, although in that case you should at least see the "rainbow box" at boot :thought_balloon: 

let's try the kernel first, and if we don't get to it, we can discuss sending the card or something... :wink:

Zeerix, 2015-10-09:
With the recent switch to Jessie (#289) and the kernel upgrade, this might work now, am I right?

diederikdehaas, 2015-10-09:
@Zeerix You were having this issue too then?

If you now build the installer you'd get a somewhat newer kernel (+rpi19), but it's still on 3.18 (and not 4.1 like the raspberrypi jessie image)

petrkalos, 2015-10-22:
I've experience the same issue with a 32GB SanDisk.
I've found a workaround to the problem by setting the rootsize=+5000M in the installer-config.txt

diederikdehaas, 2015-10-22:
@devlg does the installer not start at all or does the installed system not boot up?

devlg, 2015-10-22:
Nothing, screen stays black. Not even the activity led blinks :-) Gonna test this, thanks @petrkalos

diederikdehaas, 2015-10-22:
@petrkalos response indicates that the installation process started for him, but that the installed system failed to boot.
@devlg Your response didn't tell me whether the installation never started or that the installed system failed to boot and I'd like to know that.

@petrkalos can you try with different values for `rootsize`? I wonder whether/where there is a 'tipping point' from which the system starts to fail, for example around 16GB.

petrkalos, 2015-10-22:
@diederikdehaas I think something went wrong in my partitioning even for the 5GB rootsize...
Take a look here...
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  2.7G  769M  1.8G  30% /

Also the system is veeeeeeeeery slow when it comes to sdcard IOs although dd reports about 5MB/s

I will run some tests with different rootsize tomorrow and I will let you know

diederikdehaas, 2015-10-22:
If you have `/var/log/raspbian-ua-netinst.log` and could share that, that would be great :-)
The output of `dmesg` would help too.

Thanks in advance :+1:

devlg, 2015-10-22:
My installation @diederikdehaas does not even start. My screen stays black as if there is no sd-card in at all.

diederikdehaas, 2015-10-22:
Ah ok, thanks!

diederikdehaas, 2015-11-11:
@devlg How did you put the installer onto the sd card? And if you're on Windows, can you check what the filesystem type is? It should be FAT32, but Windows would default to exFAT for a 64GB card.

diederikdehaas, 2015-11-19:
Ping?

diederikdehaas, 2016-03-02:
A change has been merged into the v1.1.x branch and it would be interesting to know whether that fixes this issue.
If I get no response in a week or so, I'll close it any way as the previous _Ping_ is now more then 3 months old.

devlg, 2016-03-03:
Thanks for your replay. I've switched to Raspbian Lite in the meantime 
which works as expected. Thanks!

-- 
Florian Strankowski
Mitglied Deutscher Verband der Pressejournalisten (DVPJ)

Core Solutions
Adolph Schönfelder Straße 68
22083 Hamburg, Germany

phone +49 173 754 2502
web https://coresec.de

This e-mail may contain confidential and/or privileged information. If
you are not the intended recipient (or have received this e-mail in
error) please notify the sender immediately and destroy this e-mail. Any
unauthorised copying, disclosure or distribution of the material in this
e-mail is strictly forbidden and may be unlawful.

goranche, 2016-03-03:
in that case I recommend closing this issue

diederikdehaas, 2016-03-03:
Lets give @petrkalos a chance to respond ...

petrkalos, 2016-03-03:
I apologise for my inactivity.
I am currently running an installation using the 1.1.x branch as you suggested with the 8k offset.
Looks like the step "Initializing / as ext4..." takes very long time and after a while it is failing (timeout?)
Saying that "cannot mount /, invalid arguments" etc.. but I guess that's because the actuall formating is failing.
Is there any way that I can get the full logs from the first boot process?

diederikdehaas, 2016-03-03:
> I dare to say that my SDCard is very slow and the mkfs is timing out OR my RPi have some hardware issues.

That does indeed sound like the most plausible explanation. 

Another thing you could do is remove the `&>/dev/null` from https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L655 (and then rebuild the installer) as then you'd get all the output of `mkfs.ext4`.
Running `mkfs.ext4` in verbose mode should also help shed some light on the issue.
Just in case: can you post your whole `installer-config.txt`? You can remove/replace any passwords or other personal info to make it 'anonymous'.

TIA :+1:

petrkalos, 2016-03-05:
> Just in case: can you post your whole installer-config.txt? You can remove/replace any passwords or other personal info to make it 'anonymous'.

 It contains only the rootsize and the rootfstype, other than that it's empty.

> That does indeed sound like the most plausible explanation.

Although that one doesn't explain why it's OK with btrfs. Note that btrfs formatting is very fast (2-3 seconds)

I'll try to see what error the mkfs.ext4 reports.
Is there any way to check the dmesg output after the failure and before the reboot?

diederikdehaas, 2016-03-05:
> Note that btrfs formatting is very fast (2-3 seconds)

I _think_ that the creation of the btrfs filesystem is fundamentally different from ext* (but I do not know). That it is finished in 2-3 seconds seems to indicate that. And that short time also makes it not time out and thus things work as you'd expect.

> Is there any way to check the dmesg output after the failure and before the reboot?

If it fails you get 10 seconds to press a key which will then start up a terminal in which you can execute `dmesg`. You should be able to do that with a normal keyboard attached to the Pi or if you have a serial cable, you can do it through that too.
The easiest way is probably to remove the `&>/dev/null` in the script and re-build the installer.
I'll make one for that shortly and update this post with the url for it.

petrkalos, 2016-03-05:
No need I've already build it without the redirection. I'll try it tomorrow and I'll post the logs here.

diederikdehaas, 2016-03-05:
Cheers :+1:

petrkalos, 2016-03-05:
OK I have some findings...
First I've noticed that the mkfs.ext4 doesn't report any errors, the failure was coming from the mounting of the rootfs, the kernel reports this...

```
[  774.122398] JBD2: no valid journal superblock found
[  774.122402] EXT4-fs (sdb2): error loading journal
```

Thus I thought that there is something wrong with the mkfs shipped with the armhf debian but I replicate the exact same behaviour on my Ubuntu 14.04 amd64 using the same SDCard.
So I tried to create the partition without journal (**mkfs.ext4 -O ^has_journal**) instead of the writeback and it worked.

petrkalos, 2016-03-05:
Posting here the working installer-config.txt...
```
rootfstype=ext4
rootfs_mkfs_options='-O ^has_journal'
rootfs_install_mount_options='noatime,nobarrier,noinit_itable'
rootfs_mount_options='errors=remount-ro,noatime'
```

diederikdehaas, 2016-03-05:
I appreciate your extensive testing :+1: 

I'm not sure what to do about this though. It sounds like a defect with that card and if that is indeed the case, that's not something we can fix. 
OTOH, it's probably not so smart on our side to redirect all output to `/dev/null` ... :thought_balloon:

goranche, 2016-03-05:
> EXT4-fs (sdb2): error loading journal

:thought_balloon: the `sdb2` confuses me a bit... :confused:

diederikdehaas, 2016-03-05:
Could be he copied the error message from his Ubuntu machine ...

goranche, 2016-03-05:
could be...

petrkalos, 2016-03-05:
Yes I've copied the message from the Ubuntu, as I was getting exactly the same behaviour.
I think my SDCard is somehow defective and yes there is nothing that you can do about this.

What I would suggest is to add a configuration flag for turning on/off the journal in ext4 for 2 reasons
1. I've seen posts in stack0verflow with people having the same problem with me
2. Journaling obviously reduces the already short lifetime of an sdcard

I guess you could now close the issue

goranche, 2016-03-05:
there already is an option that allows you to do this, `rootfs_mkfs_options`, I wouldn't add additional options just for turning on particular flags... :thought_balloon:

diederikdehaas, 2016-03-05:
> I wouldn't add additional options just for turning on particular flags...

:+1: 
IMO the chances for errors is WAY too high as we would have to take into account every permutation of file system parameters and I think that's next to impossible and should be left to the user.

goranche, 2016-03-05:
adding it to the documentation might be a good idea, though... :thought_balloon: 

I don't particularly care, since virtually all of my (non development) systems are read-only... 
(and of course, I know about the possible negative effects :innocent: )

diederikdehaas, 2016-03-05:
The documentation could indeed use some lovin' :stuck_out_tongue_winking_eye:

PR #371 (2016-03-02): Changed default bootoffset to 8192.
This fixes #299 and possibly fixes #283.
commit 1dd80861 (2016-03-02): Changed default bootoffset to 8192.

This fixes #299 and possibly fixes #283.
closed 2016-03-05 by diederikdehaas