debian-pi/raspbian-ua-netinst #284: Default boot partition seems too small

FuNK3Y, 2015-09-27:
Yesterday I tried to deploy netinst fresh on my raspberry, but it never succeeded (endless reboot, it never get passed the deployment). I hooked it on a TV and I saw some error message related to lack of free space and initrd.

I did increase the default boot partition size from 128 to 256 and everything went through perfectly.

goranche, 2015-09-27:
sorry to hear that... did you use the release build from github, or did you build the installer yourself?

diederikdehaas, 2015-09-27:
@FuNK3Y can you post the output of `ls -l /boot/`? 128MB not being enough sounds odd to me.

FuNK3Y, 2015-09-27:
du -h /boot
8.0K    /boot/System Volume Information
68K     /boot/overlays
40M     /boot

ls -lh /boot
-rwxr-xr-x 1 root root 4.3K Sep 27 12:14 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root 4.6K Sep 27 12:14 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root 5.5K Sep 27 12:14 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root  18K Sep 27 12:14 bootcode.bin
-rwxr-xr-x 1 root root  142 Sep 27 12:15 cmdline.txt
-rwxr-xr-x 1 root root 113K Sep 27 12:08 config-3.18.0-trunk-rpi
-rwxr-xr-x 1 root root  147 Sep 27 11:59 config-reinstall.txt
-rwxr-xr-x 1 root root  209 Sep 27 12:48 config.txt
-rwxr-xr-x 1 root root 2.3K Sep 27 12:14 fixup_cd.dat
-rwxr-xr-x 1 root root 6.0K Sep 27 12:14 fixup.dat
-rwxr-xr-x 1 root root 9.0K Sep 27 12:14 fixup_x.dat
-rwxr-xr-x 1 root root 4.9M Sep 27 12:11 initrd.img-3.18.0-trunk-rpi
-rwxr-xr-x 1 root root  183 Sep 27 11:59 installer-config.txt
-rwxr-xr-x 1 root root 8.1M Sep 27 11:59 installer-rpi1.cpio
-rwxr-xr-x 1 root root 8.0M Sep 27 11:59 installer-rpi2.cpio
-rwxr-xr-x 1 root root 2.6M Sep 27 12:15 kernel.img
-rwxr-xr-x 1 root root 2.6M Sep 27 11:59 kernel-rpi1_install.img
-rwxr-xr-x 1 root root 2.7M Sep 27 11:59 kernel-rpi2_install.img
-rwxr-xr-x 1 root root 1.5K Sep 27 12:14 LICENCE.broadcom
drwxr-xr-x 2 root root 4.0K Sep 27 12:15 overlays
-rwxr-xr-x 1 root root 542K Sep 27 12:14 start_cd.elf
-rwxr-xr-x 1 root root 2.6M Sep 27 12:14 start.elf
-rwxr-xr-x 1 root root 3.5M Sep 27 12:14 start_x.elf
-rwxr-xr-x 1 root root 1.9M Sep 27 12:08 System.map-3.18.0-trunk-rpi
drwxr-xr-x 2 root root 4.0K Sep 27 11:59 System Volume Information
-rwxr-xr-x 1 root root 2.6M Sep 27 12:08 vmlinuz-3.18.0-trunk-rpi

I took the release from github. I did a straightforward deployment, with only hostname & ip as customization.

goranche, 2015-09-27:
did it reboot right after booting, or was it later in the installation process? could it be that the problem is in the size of the root partition? what size of SD card are you using?

not sure how you could print out the partition table content on windows :thought_balloon:

diederikdehaas, 2015-09-27:
I have no idea why 128MB would not be enough. The directory listing looks perfectly normal to me. 
Unfortunately there seem to be no log files of the failure(s).

FuNK3Y, 2015-09-27:
It did reboot later in the installation process, it was somewhere near the building of the initrd image as far as I understood.

I am using a 8Gb microSD card.

Here is how I built the microSD card under windows
- Diskpart clean
- Create 1 partition of 8Gb formated FAT32
- Copy the content of the zip file to the SD card
- Creation of installer‑config.txt file

goranche, 2015-09-27:
could you do a "Diskpart detail disk" (I'm writing this from the documentation, I don't have a windows machine to try it, so I'm hoping you're better at this than I am) after the install...

I'd really like to see what partition scheme the installer managed to come up with :smile: 
(it should be the whole disk, but you never know)

FuNK3Y, 2015-09-27:
Here are the information from the raspberry itself:

df -h
Filesystem      Size  Used Avail Use% Mounted on
rootfs          7.0G  476M  6.1G   8% /
udev             10M     0   10M   0% /dev
tmpfs            44M  184K   44M   1% /run
/dev/mmcblk0p2  7.0G  476M  6.1G   8% /
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            88M     0   88M   0% /run/shm
/dev/mmcblk0p1  244M   40M  205M  17% /boot
tmpfs            88M   11M   77M  13% /tmp

root@satip:/# fdisk -l

Disk /dev/mmcblk0: 7948 MB, 7948206080 bytes
4 heads, 16 sectors/track, 242560 cylinders, total 15523840 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x00000000

        Device Boot      Start         End      Blocks   Id  System
/dev/mmcblk0p1              16      500095      250040    b  W95 FAT32
/dev/mmcblk0p2          500096    15523839     7511872   83  Linux

As you can see, everything is correct. I have a 256 boot requested it in the installer‑config.txt

IMHO the space problem is not caused by a wrong partitioning, as it works when I specify 256mb. I would rather suspect some cleanup not done properly after inflating files, or inflating too much files, ...

goranche, 2015-09-27:
no, it doesn't seem to be a problem, I agree... still had to check :innocent:

goranche, 2015-09-27:
> from the raspberry itself

oh, right... forgot about that, it actually installed with the larger root... sorry about that :blush:

goranche, 2015-09-27:
do you have anything special in installer-config.txt

diederikdehaas, 2015-09-27:
Better yet, post your whole `installer-config.txt`

FuNK3Y, 2015-09-27:
Here you go:
bootsize=+256M
hostname=satip
domainname=mysweethome.local
ip_addr=192.168.1.22
ip_netmask=255.255.255.0
ip_broadcast=192.168.1.255
ip_gateway=192.168.1.254
ip_nameservers=192.168.1.1

The faulty one was the exact same, without the bootsize

goranche, 2015-09-27:
I like your domainname :+1: :wink: 

well, nothing left to do, but to download the image, copy it ... oh, wait, you copied the files to the card, right... ok, I'll try that

diederikdehaas, 2015-09-27:
@FuNK3Y Do you have a serial console (cable) such as this one: https://www.adafruit.com/products/954 ?

goranche, 2015-09-27:
couldn't reproduce, sorry... I've tried both on the RPi 2 (my "default" device) and the RPi B+, and all works without any problem...

I don't have an RPi B anymore, but don't really see how the result could be different :thought_balloon: 

I copied the files to the FAT partition, the only difference is I created the partition on a Mac...

FuNK3Y, 2015-09-27:
@diederikdehaas sadly no, I have no such cables
@goranche Thank you very much for testing, the outcome is very interesting!

I'll try to find some SD card and reproduce, but it might take me some more time

diederikdehaas, 2015-09-27:
Ok, was worth a shot :wink: 
When the installer fails it tries to write the log file to the boot partition, which of course won't succeed if the problem is the free space on that partition. A serial cable is another way to monitor the installation progress and you could've copied the log file/info from there ... if you had one.

Sadly I have no idea as to why it's failing for you. Logically it doesn't make any sense and if it was a general problem we would be able to reproduce it and there would likely be many reports and/or +1's on this one.

FuNK3Y, 2015-10-01:
Humm my raspberry was not in a great shape, it died yesterday; so I have no possibility to test until I get a replacement.

I'll close the ticket then; and reopen in case I am able to reproduce.

Thank you everyone for your support !

diederikdehaas, 2015-10-01:
> I'll close the ticket then; and reopen in case I am able to reproduce.

Excellent and good luck :+1:

FuNK3Y, 2015-10-11:
New raspberry, same problem.

This time I took a picture of the error on the screen
[image]

diederikdehaas, 2015-10-11:
That looks like a problem with the SD card itself. The error msg now is not that the partition is full (which you can/could verify with `df -h`), but that the disk itself was set to readonly ... and that's a problem if it wants to write something.

FuNK3Y, 2015-10-11:
I did wipe entirely my SD card to redeploy from scratch and it worked. Previously the SD card was in my faulty raspberry, it might have corrupt something.

Let's close this for good => HW problem.

closed 2015-10-11 by FuNK3Y