debian-pi/raspbian-ua-netinst #286: installer-config.txt not working

TimVNL, 2015-09-30:
Hi,
I have a installer-config.txt on the root of my SD card but it doesn't seen to work any suggestions? 

I am using version 1.0.7

diederikdehaas, 2015-09-30:
Can you post your `installer-config.txt` and do you have an `/var/log/raspbian-ua-netinst.log` file? If so, can you share that as well?

hidden-refuge, 2015-10-01:
I've had the same issue. I also used 1.0.7 and the installer-config.txt was totally ignored by the installer. It didn't even exist after I extracted the files and moved them to the SD. So I made it myself but everything inside was just ignored.

goranche, 2015-10-01:
what system are you using? is it possible that installer-config.txt got created with a slightly different filename? (upper / lower case, underscore instead of hyphen, ...)

hidden-refuge, 2015-10-01:
Windows 7. I created the file with Notepad++ and copy the content of the example file on the readme page of this project plus included my additions like jessie instead of wheezy and some packages I needed. I named it as it was supposed to be named "installer-config.txt". Encoding was UTF8. 1.0.6 has a installer-config.txt inside the Windows package but 1.0.7 lacks it totally.

goranche, 2015-10-01:
[guidelines for contributing](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/CONTRIBUTING.md):

> Furthermore, it helps if you provide the configuration files you used, such as `installer-config.txt`. Make sure to remove/replace sensitive information, if present. The installer also creates a log file, which is either located in `/boot/` or when the installation finished, in `/var/log/`. If you can provide those, that helps too.

TimVNL, 2015-10-01:
after some tests it work file does like empty lines in my case

closed 2015-10-01 by TimVNL