debian-pi/raspbian-ua-netinst #287: rpi-update confusion config.txt I have a question

sirmonkeyblur, 2015-09-30:
may I ask why the kernel version is in the config.txt file?
(rpi-update just downloaded and writes kernel7.img)

I love your tool! and used it to build a pi.
but I'm using the 7" official raspberry display and need the latest firmware/kernel to make it work. 

so I had a hard time, till I figured out it was the 3 lines in the config.txt that specified the kernel to boot. i deleted that and now everything works.  (i do like your idea of having the kernel version specified)

diederikdehaas, 2015-09-30:
> may I ask why the kernel version is in the config.txt file?

Because if you don't specify the kernel to use, it will use the default, which is kernel.img on a Pi 1 and kernel7.img on a Pi 2.
See also https://github.com/debian-pi/raspbian-ua-netinst/issues/282

sirmonkeyblur, 2015-09-30:
ok, I can appreciate haveing a _real_ versioned kernel. Makes many things much easier.

so what would you recommend to update the kernel? 
the raspberrypi-bootloader package has the correct kernel and firm where. but it makes a kernel7.img file.

Thanks

diederikdehaas, 2015-09-30:
Installing the raspberrypi-bootloader package seems like the best option.
And if you _really_ want a versioned kernel for that too, you can copy kernel7.img and rename the copy and specify that in the `kernel=` line in `/boot/config.txt`. But it seems a bit pointless to me and you should update it again when a new version of raspberrypi-bootloader is made available.

sirmonkeyblur, 2015-09-30:
ooo ok ok. so I get the bug report now. there are different packages and one of them isn't getting the new firmware or kernel stuff.  (need to ping a package maintainer )

could I build another 'preset' like "server-alt_kernel_package" ? how would you recommend I do this ?
or perhaps have a post install script that switches it?

diederikdehaas, 2015-09-30:
https://github.com/debian-pi/raspbian-ua-netinst/issues/282#issuecomment-142705091

sirmonkeyblur, 2015-10-01:
ok cool, thanks for all the help :-) :-) :-)
should anyone find this via the googles and yahoos

i downloaded the https://github.com/debian-pi/raspbian-ua-netinst/releases/latest extracted and such then added a "post-install.txt"  and then put below in that file

no it became a bigger adventure. I don't mean to point fingers but it looks like kernel and firmware are changing faster than the package maintainers can keep up? maybe?

so after i booted into my system (thanks to this awesome projects install wizardy! )
I took some notes form this
https://0xstubs.org/stock-debian-jessie-on-the-raspberry-pi-2/
made the preferences file and added to the /etc/apt/source.list file
apt-get update

well here is a snippet from a script
apt-get -y install raspberrypi-bootloader

#remove kernel package used 
#this part is going to need to be updated as the version number changes. or learn howto bash script better
dpkg -r linux-image-rpi2-rpfv
dpkg -r linux-image-3.18.0-trunk-rpi2

#remove raspberrypi-bootloader-nokernel?

#install other kernel package
#install firmware goodies
apt-get -y install --reinstall raspberrypi-bootloader
apt-get -y install libraspberrypi0 libraspberrypi-dev libraspberrypi-bin libraspberrypi-doc

Offical 7inch display working like a champ... not to play with Qt5 and/or Kivy :-)

thanks again raspbian netinstall team! :-) :-) :-)
good luck getting those kernel packages updated and stuff

diederikdehaas, 2015-10-02:
@sirmonkeyblur Does that mean that the issue is hereby resolved?

closed 2015-10-08 by diederikdehaas