debian-pi/raspbian-ua-netinst #288: Jessie upgrade?

hidden-refuge, 2015-10-01:
So Raspberry Pi Foundation has updated Raspbian officially to Jessie with 4.1 kernel and all the never packages. Will this be updated to Jessie, too? I tried changing the mirror used in this to "jessie" instead of "wheezy" but it still installed wheezy for some unknown reason. Actually everything inside the installer-config.txt was just ignored.

diederikdehaas, 2015-10-01:
The raspbian.org kernel, which is used by this installer is still 3.18 on both wheezy and jessie. And I don't know when this will be updated. BTW Debian Jessie uses 3.16.

> Actually everything inside the installer-config.txt was just ignored. 

Why do you think that? It's all working fine here ...

diederikdehaas, 2015-10-01:
Also, from [guidelines for contributing](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/CONTRIBUTING.md):

> Furthermore, it helps if you provide the configuration files you used, such as `installer-config.txt`. Make sure to remove/replace sensitive information, if present. The installer also creates a log file, which is either located in `/boot/` or when the installation finished, in `/var/log/`. If you can provide those, that helps too.

We like to help, but we really need more info then "it doesn't work" in order to do so.

hidden-refuge, 2015-10-01:
Raspbian based on Debian Jessie by Raspberry Pi Foundation is using kernel 4.1.

See yourself: https://www.raspberrypi.org/downloads/raspbian/ (RASPBIAN JESSIE)

goranche, 2015-10-01:
> Raspbian based on

is not the same as Debian Jessie  :wink:

hidden-refuge, 2015-10-01:
So wait. I thought that this Raspbian is what Raspberry Pi Foundation has built but without all the preinstalled stuff and etc. Freedom of choice. But you are building it off the real Debian ARM version?

goranche, 2015-10-01:
this is debian for the raspberry, so to speak, yes...

diederikdehaas, 2015-10-01:
The difference regarding the kernel between this installer and what you get with the raspberrypi.org image is the kernel package used. 
This installer uses the `linux-image-rpi2-rpfv` (Pi 2) or `linux-image-rpi2-rpfv` (Pi 1) package and that (currently) installs a 3.18 kernel. I call this the raspbian.org kernel.
On the raspberrypi.org image they use the `raspberrypi-bootloader` package and that installs the 4.1 kernel from raspberrypi.org. And I call this the raspberrypi.org kernel.

We use/prefer the raspbian.org kernel because it's packaged properly according to debian standards (including a kernel-headers package) and the raspberrypi.org kernel is not.

hidden-refuge, 2015-10-01:
Thanks for the explanation guys. I did a dist upgrade to Jessie and it worked fine.

Close this issue.

closed 2015-10-01 by hidden-refuge