debian-pi/raspbian-ua-netinst #293: Not installing debian Jessie after placing installer‑config.txt with 'release=jessie'

kdeenkhoorn, 2015-10-10:
Dear All,
I've tried to install debian Jessie with raspbian-ua-netinst version 1.0.7.
I've placed installer-config.txt next to cmdline.txt on the SD card with contents of ' release=jessie' and a 'hostname=myname' .
But during the installation the release=wheezy is used.
And the hostname Pi is used.

Am i doing something wrong ?
I am installing a raspberry pi 2B.

Thanks in advance,

Kl@@s

kdeenkhoorn, 2015-10-10:
Dear All,
My error, i've mounted the image after xzcat .... under ubuntu and added the file  installer-config.txt.
So far so good but after installation and reboot the file was called 'installer???config.txt' so that's the reason why it was not sourced.

Must be something to do with mount options.

goranche, 2015-10-10:
> Must be something to do with mount options.

might be :thought_balloon:  it might also be, that you used a dash instead of a hyphen (or the other way around, can never really remember which is which)... happens :wink: 

I'm assuming everything was ok after fixing the filename?  in that case, feel free to close the issue :smile:

kdeenkhoorn, 2015-10-10:
I'm running jessie 8.0 now is that right ?

goranche, 2015-10-10:
if all went right then it's Debian 8, which is codenamed Jessie, yes

you can check by doing ```cat /etc/issue```

kdeenkhoorn, 2015-10-10:
Ok, then i'm a happy camper!

Thanks for the opportunity to only install a server image for the raspberry pi 2b.

closed 2015-10-10 by kdeenkhoorn

goranche, 2015-10-10:
> then i'm a happy camper

:+1: glad to hear that :wink:   and thank you for using the project