debian-pi/raspbian-ua-netinst #295: Raspbian jessie

kruks23, 2015-10-14:
Hello,

I have download the latest version of raspbian-ua-netinst (1.0.7) and I have created a file named installer-config.txt with this line:

release: jessie

But something was wrong and the installation doesn't start.

If I dont create the installer-config.txt the installation works fine but the system installed is raspbian wheezy.

I would like to install jessie, could you help me please?

Thank you.

diederikdehaas, 2015-10-14:
> But something was wrong and the installation doesn't work.

You'd have to give more details in order for us to help you.
Are there files written to the sd-card that end with `.log`? When the installation fails, the installer tries to write the logfile back to the sd-card (next to the installer files) and that can help us see where/why things went wrong.
Did you add a line-feed (<Enter>) after `release=jessie`?

kruks23, 2015-10-15:
There is no log, I'm using a raspberry pi model B.

With my raspberry pi 2, the installation works well, but I think that the system installed is not the official raspbian jessie. Is a mix of two repositories? maybe a mirror for debian armhf?

Maybe that's why it doent work on my raspberry pi model B?

Thank you.

diederikdehaas, 2015-10-15:
All the source code is available right here, so you can check that your assumption/accusation is false.
The installer works here on my Pi 1B, 1B+ and 2B.

kruks23, 2015-10-15:
I'm sorry, I didnt want to make any accusation. I will try again and check what is going wrong.

Thank you.

closed 2015-10-15 by kruks23

diederikdehaas, 2015-10-15:
Feel free to reopen this issue when you do have more info :-) but right now there's too little info to work with.